espressif/arduino-esp32 #1051: Preferences putBytes error

DeltaCore, 2018-01-30:
Hello.

I found a strange error in the Preferences lib.

first of all, my code:

```c++
#include <Arduino.h>
#include <Preferences.h>

#define nvs_size sizeof(uint8_t) * 2048
#define nvs_init_tag_value 0x0111

const char *nvs_init_tag = "init";
const char *nvs_size_tag = "size";
const char *nvs_data_tag = "data";

void setup()
{
	Serial.begin(921600);
	delay(2000);
	Serial.println("Hallo github");
	Preferences *nvs = new Preferences();
	nvs->begin("data");
	uint8_t *nvs_data_buffer = (uint8_t *)malloc(nvs_size);
	memset(nvs_data_buffer, 0, nvs_size);
	if (nvs->getUInt(nvs_init_tag) != nvs_init_tag_value)
	{
		// init everything
		nvs->putUInt(nvs_init_tag, 0);
		nvs->putULong(nvs_size_tag, nvs_size);
		nvs->putBytes(nvs_data_tag, nvs_data_buffer, nvs_size);
		nvs->putUInt(nvs_init_tag, nvs_init_tag_value);
	}
	nvs->getBytes(nvs_data_tag, nvs_data_buffer, nvs_size);
	Serial.println("hmmm");
}

void loop()
{
}
```
Console output:
```
pio device monitor --baud 921600  --rts 0 --dtr 0 --filter direct
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/cu.SLAB_USBtoUART  921600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Hallo github
[E][Preferences.cpp:354] getUInt(): nvs_get_u32 fail: init NOT_FOUND
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400014fd  PS      : 0x00060730  A0      : 0x800d4450  A1      : 0x3ffca220
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x3f400f01  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x20000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400014fd:0x3ffca220 0x400d444d:0x3ffca230 0x400e608e:0x3ffca540 0x400e60c6:0x3ffca5d0 0x400e0c36:0x3ffca610 0x400e0ee3:0x3ffca670 0x400d09b7:0x3ffca690 0x400e0ccf:0x3ffca6c0

Rebooting...
```

Stacktrace:

```
esp32dev-stacktrace 0x400014fd:0x3ffca220 0x400d444d:0x3ffca230 0x400e608e:0x3ffca540 0x400e60c6:0x3ffca5d0 0x400e0c36:0x3ffca610 0x400e0ee3:0x3ffca670 0x400d09b7:0x3ffca690 0x400e0ccf:0x3ffca6c0

0x400014fd: ?? ??:0
0x400d444d: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529
0x400e608e: _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
0x400e60c6: vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41
0x400e0c36: log_printf at /Users/marcel/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:401
0x400e0ee3: Preferences::putBytes(char const*, void const*, unsigned int) at /Users/marcel/.platformio/packages/framework-arduinoespressif32/libraries/Preferences/src/Preferences.cpp:401
0x400d09b7: setup() at /Users/marcel/dev/me/pref_test/src/main.cpp:25
0x400e0ccf: loopTask(void*) at /Users/marcel/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:15
```

I tinkered a bit with that, added debug in the Pref. lib and so on. This problem only occurs only when the `nvs_size` define bigger than 1024 is, like there 2048.
I could not find a max size for an nvs tag (here data). Is there a limit am not aware of?

Best regards, 
Marcel

### Hardware:
Board:							m5stack
Core Installation/update date:			(where do i get that? latest from platformio as of today)
IDE name:							Platform.io
Flash Frequency:					40Mhz
Upload Speed:						2000000

everslick, 2018-04-19:
1984 bytes, AFAIK.

beegee-tokyo, 2018-04-20:
@everslick @DeltaCore 
nvs.h says in line 127:
```
 * @param[in]  value   The value to set.
 *                     For strings, the maximum length (including null character) is
 *                     1984 bytes.

```
Added to the [Wiki](https://desire.giesecke.tk/index.php/2018/04/20/preferences-library-max-tag-length/)

DeltaCore, 2018-09-03:
Wow, i thought i closed this issue.
Thanks @beegee-tokyo for pointing that out. I missed that.

- DeltaCore

closed 2018-09-03 by DeltaCore