espressif/arduino-esp32 #1053: How to disable Bluetooth Classic and enable only BLE?

shahabmusic, 2018-01-31:
I currently have TLS OTA + TLS MQTT.
When I try to scan for available wifi networks or when I try to scan for BLE devices, my ESP32 wroom crashes and restarts or sometimes returns no BLE device (despite having a few around the house). Or sometimes my secure TLS disconnects when I try to BLE scan. I googled and found that apparently I run out of heap because of TLS... 
I also read somewhere that in ESP IDF, there is a function (bluetooth_mode_ble() I guess) that disables the bluetooth classic and therefore releases more heap.
Is that right? how can I achieve having TLS MQTT and BLE simultaneously in Arduino?

erikwilson, 2018-01-31:
I haven't tried it myself but my guess is that you will either need to reduce the mqtt key size or disable/destroy bluetooth altogether when communicating with mqtt.
Also see: https://github.com/espressif/esp-idf/issues/346

beegee-tokyo, 2018-02-05:
I haven't played around with BLE client much, but I have a BLE server, MQTT client, UDP broadcast listener and some sensorik stuff running stable on my [ESP32 weather station](http://desire.giesecke.tk/index.php/2018/01/29/esp32-weatherstation/). To get this to work, I had to complete eliminate my SPIFFS and EEPROM partitions to get enough space.

I tried as well a BLE client implementation and experienced a loss of WiFi connection everytime I started a BLE scan. I think there is a lot of improvement in the BLE libraries and esp-idf required to run both BLE and WiFi in parallel.
Not sure however why the BLE server works fine in my apparently and the BLE client causes problems.

closed 2019-08-15 by stale[bot]