espressif/arduino-esp32 #1055: Webserver Broken

shaddow501, 2018-01-31:
Hello

It seems that in the last release something broke the Webserver.
I am trying to load index.htm from SPIFFS and I see in the terminal that the the request for the file was given but in the web page the page stay without any response, I mean I see that there is a request but eventually no page is loaded.

response from the serial terminal - handleFileRead: /index.htm
The file exist in the spiffs.
This is list of the file in the spiffs:

isting directory: /
  FILE: /auto-scale-axis.js  SIZE: 2479
  FILE: /chartist.css.map  SIZE: 15734
  FILE: /bower.json  SIZE: 826
  FILE: /chartist.js  SIZE: 172296
  FILE: /jquery.min.js  SIZE: 92633
  FILE: /line.js  SIZE: 17617
  FILE: /step-axis.js  SIZE: 1634
  FILE: /chartist.min.css  SIZE: 10878
  FILE: /chartist.min.js  SIZE: 38150
  FILE: /svg-path.js  SIZE: 15771
  FILE: /chartist.min.js.map  SIZE: 232141
  FILE: /svg.js  SIZE: 25199
  FILE: /WiFiManager.template.html  SIZE: 3758
  FILE: /chartist.scss  SIZE: 6857
  FILE: /class.js  SIZE: 3671
  FILE: /core.js  SIZE: 38357
  FILE: /edit.htm.gz  SIZE: 4116
  FILE: /event.js  SIZE: 2542
  FILE: /favicon.ico  SIZE: 1150
  FILE: /fixed-scale-axis.js  SIZE: 2406
  FILE: /graphs.js.gz  SIZE: 1971
  FILE: /index.htm  SIZE: 20170
   FILE: /interpolation.js  SIZE: 15612
  FILE: /axis.js  SIZE: 4571
  FILE: /base.js  SIZE: 8187

The older lib had different issue that caused the module to reset, but in there the webpage did open without changing anything in my code. 
so I have moved to the new lib.

The older lib date of the files that I have download were from 05/11/2017. (Webserver worked there, pages did load)

What broke it? where should I look?

copercini, 2018-01-31:
Looks like related with: https://github.com/espressif/arduino-esp32/issues/1022

shaddow501, 2018-01-31:
Could be, When can it be fixed or what should I do to fix it?

JohnRabbit54, 2018-02-02:
You can find some more issues here dealing with this problem. It has to do that the arduino component was compiled with SPIFFS_META_LENGTH=4. This implementation is incompatible with previous implementation with SPIFFS_META_LENGTH=0 (or no META info). If you are using Arduino as ESP-IDF component you are in luck. In menuconfig you can set SPIFFS_META_LENGTH to zero. However if you are using this in the Arduino IDE I don't know what you can do.

smacyas, 2018-02-04:
You have to change file from tools/sdk/lib/libspiffs.a to the same file from 70d0d464 commit. Thanks for this problem solving to @vindolin

shaddow501, 2018-02-07:
Hello smacyas

I cannot find the file in the link you have provided...

smacyas, 2018-02-07:
You have to click "Browse files" on the right top of page. Direct link: https://github.com/espressif/arduino-esp32/tree/70d0d4648766cea047613062ed50a9ecfc9de31f/tools/sdk/lib

shaddow501, 2018-02-07:
Ok, Thanks smacyas.

usmanshahid001, 2018-04-02:
does i need to use different plugin as well because it is not working by just replacing the spiffs.a

copercini, 2018-06-27:
Please update to last code, it's now supporting webserver and spiffs was fixed =)
https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer

closed 2018-06-27 by copercini