espressif/arduino-esp32 #1060: WiFi.status() return 1 -> AP not found while ap is present

sonastyle, 2018-02-01:
### Hardware:
Board:							       NODE32S 4MB Flash
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE / Visual Studio (vMicro)
Flash Frequency:					        40Mhz
Upload Speed:						921600

### Description:
First time when the code starts I call the wifi connect function and it connects to the specified AP and works well. If I call the WiFi.disconnect(true) method it always return no AP not found as WiFi.status() = 1 while that same AP is visible to all other device as well as its showing on return of scan_hotspot() method. This thing sometimes happen after a soft restart also and only can be retrieved via hard restart. Please let me know the fault points.

### Sketch:

#include <Arduino.h>
#include <WiFi.h>

const char* esid = "MCL";                     // hotspot ssid
const char* epass = "MCL@1234";       // ssid password

void webService(){
server.on("/", HTTP_GET, rootGet);
}
void rootGet(AsyncWebServerRequest *request){
     request->send(200, "text/html", "This is index page");
}
// wifi reconnect
void wifi_reconnect(int count) {
Serial.println(String(F("Wifi status before try: ")) + WiFi.status() + F(". SSID: ") + esid + F(". Pw: ") + epass);
int wifi_retry = 0;
while ((WiFi.status() != WL_CONNECTED) && (wifi_retry < count) && WiFi.getMode() == 
WIFI_MODE_APSTA) {
Serial.println(String(F("Ssid: ")) + esid + F(". Pw: ") + epass + F(". Mode: ") + WiFi.getMode());
WiFi.begin(esid, epass);
int concode = WiFi.waitForConnectResult();
wifi_retry++;
Serial.print(String(F("Retried: ")) + wifi_retry + F(" times.\t"));
Serial.println(String(F("Wifi status trying to connect: ")) + concode);
}
}
Serial.println(WiFi.status()); 
}
// hotspot scan
void scan_hotspot() {
	Serial.println("Hotspot scanning called");
	send_feedback(msg_scan_sotspot);
	int n = WiFi.scanNetworks();
	Serial.println(F("scan done"));
	String _st="";
	for (int i = 0; i < n; ++i)
	{
		// Print SSID and RSSI for each network found
		_st += "<option value='";
		_st += WiFi.SSID(i);
		_st += "'>";
		_st += WiFi.SSID(i);
		_st += " ( ";
		_st += 2 * (WiFi.RSSI(i) + 100);
		_st += "&#37; )"; // % symbol in html >> &#37;
		_st += (WiFi.encryptionType(i) == WIFI_AUTH_OPEN) ? "<span class='unlock'> Open</span>" : "<span class='lock'> Secure</span>";
		_st += "</option>";
	}
	Serial.println(_st);
}

void setup() {
 Serial.begin(115200);
WiFi.mode(WIFI_AP_STA);
WiFi.softAP(ssid, passphrase);
Serial.println(F("Soft AP started"));
IPAddress softapip = WiFi.softAPIP();
String lapip = String(softapip[0]) + '.' + String(softapip[1]) + '.' + String(softapip[2]) + '.' + String(softapip[3]);
Serial.println(String(F("AP Ip: ")) + lapip);
  webService();										   // webserver routing method
  server.begin(); 									   // webserver begin
  Serial.println(F("web service called"));
wifi_reconnect(10); // Try 10 times
}

void loop() {
if(WiFi.status() != 3){
wifi_reconnect();
}
if(Serial.available() >0){
String a = Serial.readStringUntil('\n');
if(a== "discon"){
WiFi.disconnect(true);   // on demand disconnect wifi
}else if(a== "scan"){
scan_hotspot();             // on demand scan hotspot
}
}
}

### Debug Messages: Serial output
```
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 1 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 2 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 3 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 4 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 5 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 6 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 7 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 8 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 9 times.	Wifi status trying to connect: 1
Ssid: MCL. Pw: Mcl@1234. Mode: 3
Retried: 10 times.	Wifi status trying to connect: 1

Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
code does not crashed, just can not reconnect wifi on loop

tablatronix, 2018-02-07:
reduce your test code

sonastyle, 2018-02-12:
The above code is working now with addition of **100mS delay before WiFi.begin(esid, epass)** and the scenario is reproducible. Please let me know the technical significance of this.

tablatronix, 2018-03-05:
Not sure, but I have a 200 ms delay after enableSta to allow for it to start, because I had the same problems

lbernstone, 2018-06-06:
If you have solved your issue, please close it.

closed 2018-07-20 by sonastyle