espressif/arduino-esp32 #1062: Can't connect to WiFiServer when in AP mode

learmond, 2018-02-02:
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation/update date:		v1.3.2 from package.json
IDE name:				                Platform.io - via VSCode
Flash Frequency:					40Mhz
Upload Speed:					default

### Description:
I'm wanting to run a WiFiServer instance on the ESP32 and make a client connection from some test client. To do this, i'm testing it in WiFi client mode, as well as WiFi AP mode. Once a TCP client connects to the server on the ESP32, we the write some string data to the client and close the connection.

1. As a client, I'm testing using my phone's hotspot mechanism. It successfully connects to my phone, and using the netcat utility within the phones terminal program (Termux), I can connect to and receive data from the WiFiServer.

2. **THE PROBLEM** As an Access Point, I can successfully connect my phone to the WiFi AP (with default and custom AP config), however netcat cannot establish a connection with the WiFiServer! It just times-out.

### Sketch as WiFiClient:
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiAP.h>

WiFiServer      server(42000);
const char*     ssid = "testAP";
const char*     pwd = "default";

void setup()
{
    Serial.begin(38400);
    while( !Serial );

    // WiFi Client code
    Serial.printf("Connecting to [%s] ", ssid);
    WiFi.begin(ssid, pwd);
    while( WiFi.status() != WL_CONNECTED )
    {
      delay(500);
      Serial.print(".");
    }

    server.begin();
}

void loop()
{
    static char buff[256];
    WiFiClient client = server.available();
    if( client )
    {
        sprintf(buff, "Hello you, from [%s]\n", client.remoteIP().toString().c_str() );
        client.write((const char*)buff);
        delay(1000);
        client.stop();
    }
}
```
## Serial output
```
Connecting to [testAP] .............
WIFI Successfully connected to AP: [testAP]. IP: [192.168.43.136]
```

## netcat output from phone
```
$> netcat 192.168.43.136 42000
Hello you, from [192.168.43.1]            <------------ WORKS!
```

### Changes for WiFi AP mode
```cpp
// added event handler
void wifi_event(WiFiEvent_t event)
{
    switch(event)
    {
        case SYSTEM_EVENT_AP_START:
            Serial.printf("AP [%s] started. IP: [%s]\n", ssid, WiFi.softAPIP().toString().c_str() );
            break;
        case SYSTEM_EVENT_AP_STOP:
            Serial.println("AP Stopped");
            break;
        case SYSTEM_EVENT_AP_STACONNECTED:
            Serial.println("WiFi Client Connected");
            break;
        case SYSTEM_EVENT_AP_STADISCONNECTED:
            Serial.println("WiFi Client Disconnected\n");
            break;
        default:
            Serial.printf("Unhandled WiFi Event: %d\n", event );
            break;
    }
}

void setup()
{    
    Serial.begin(38400);
    while( !Serial );

    WiFi.softAP("esp32_test", "default");
    WiFi.onEvent(wifi_event);
    server.begin();
}
```

## Serial output
```
AP [esp32_test] started. IP: [192.168.4.1]
WiFi Client Connected            <------- event, when my phone's wifi successfully connects.
```

## netcat output from phone
```
$> netcat 192.168.4.1 42000
Ncat: Connection timed out.     <--------- after about 10 seconds of waiting.
```

### Debug Messages:
... not sure how to do this... I added the following build flag to my platformio.ini, however nothing else was printed:
`build_flags = -DLOG_LOCAL_LEVEL=ESP_LOG_DEBUG`

Has anyone seen this before? Any ideas?
Thanks so much for any help!

beegee-tokyo, 2018-02-02:
To activate debug output the correct line is
`build_flags = -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG`
Source [ESP32 Wiki](http://desire.giesecke.tk/index.php/2018/01/30/enable-log-levels-in-platformio/)

learmond, 2018-02-02:
Thanks @beegee-tokyo 
Started up in AP mode, connected my phone's wifi, and attempted a `netcat` connection to `192.168.4.1 port 42000`, and got the following output:
```
[D][WiFiGeneric.cpp:270] _eventCallback(): Event: 13 - AP_STOP
AP [esp32_test] started. IP: [192.168.4.1]
Server listening on port: [42000]
[D][WiFiGeneric.cpp:270] _eventCallback(): Event: 15 - AP_STADISCONNECTED   <--- appeared when phone wifi connected
WiFi Client Connected
```
Looking at `WiFiGeneric.cpp:270`, it looks like the `system_event_names[]` label array is missing an enum label. As we can see in the output, `AP_STADISCONNECTED` is printed out, however, event 15 corresponds to `system_event_id_t::SYSTEM_EVENT_AP_STACONNECTED` in `esp_event.h`. The missing enum label is for `system_event_id_t::SYSTEM_EVENT_STA_LOST_IP`.

Still, my problem persists.

learmond, 2018-02-02:
So, partial success, but I'm rather baffled.
I connected a windows 10 laptop to the esp32's wifi and used `ncat.exe` to run the same test:
```
c:\...\ncat.exe -vv 192.168.4.1 42000
Ncat: Version 7.60 ( https://nmap.org/ncat )
NCAT DEBUG: Using trusted CA certificates from c:\Program Files (x86)\Nmap\ca-bundle.crt.
libnsock nsock_iod_new2(): nsock_iod_new (IOD #1)
libnsock nsock_connect_tcp(): TCP connection requested to 192.168.4.1:42000 (IOD #1) EID 8
libnsock nsock_trace_handler_callback(): Callback: CONNECT SUCCESS for EID 8 [192.168.4.1:42000]
Ncat: Connected to 192.168.4.1:42000.         <------- Successfully connected
libnsock nsock_iod_new2(): nsock_iod_new (IOD #2)
libnsock nsock_read(): Read request from IOD #1 [192.168.4.1:42000] (timeout: -1ms) EID 18
libnsock nsock_readbytes(): Read request for 0 bytes from IOD #2 [peer unspecified] EID 26
libnsock select_loop(): nsock_loop error 10038: An operation was attempted on something that is not a socket.         <------ Got Error
Ncat: 0 bytes sent, 0 bytes received in 0.12 seconds.
libnsock nsock_trace_handler_callback(): Callback: READ KILL for EID 18 [192.168.4.1:42000]
libnsock nsock_trace_handler_callback(): Callback: READ KILL for EID 26 [peer unspecified]
libnsock nsock_iod_delete(): nsock_iod_delete (IOD #1)
libnsock nsock_iod_delete(): nsock_iod_delete (IOD #2)
```

Then after looking up the error, I tried some other options, specifically the _hidden_ `--nsock-engine` option:
```
c:\...\ncat.exe -vv --nsock-engine poll 192.168.4.1 42000
Ncat: Version 7.60 ( https://nmap.org/ncat )
NCAT DEBUG: Using trusted CA certificates from c:\Program Files (x86)\Nmap\ca-bundle.crt.
libnsock nsock_iod_new2(): nsock_iod_new (IOD #1)
libnsock nsock_connect_tcp(): TCP connection requested to 192.168.4.1:42000 (IOD #1) EID 8
libnsock nsock_trace_handler_callback(): Callback: CONNECT SUCCESS for EID 8 [192.168.4.1:42000]
Ncat: Connected to 192.168.4.1:42000.         <----- Successfully connected
libnsock nsock_iod_new2(): nsock_iod_new (IOD #2)
libnsock nsock_read(): Read request from IOD #1 [192.168.4.1:42000] (timeout: -1ms) EID 18
libnsock nsock_readbytes(): Read request for 0 bytes from IOD #2 [peer unspecified] EID 26
libnsock nsock_trace_handler_callback(): Callback: READ SUCCESS for EID 18 [192.168.4.1:42000] (30 bytes): Hello, you from [192.168.4.3].
Hello, you from [192.168.4.3]                         <-------- SUCCESS ! Got the reply !!??
libnsock nsock_readbytes(): Read request for 0 bytes from IOD #1 [192.168.4.1:42000] EID 34
libnsock nsock_trace_handler_callback(): Callback: READ EOF for EID 34 [192.168.4.1:42000]
close: Result too large
```

I tried the `--nsock-engine poll` option from my phone's `netcat` and it didn't complain about an unknown option, but it didn't work, and still timed out.

d0ct0rd-dalimo, 2019-01-09:
I have the some different problem: when I compile and flash WiFi AP code, it works normally, but if I reconnect to this AP, it gives some data then drop the connection and goes down. Then it refuses further connections to this AP

closed 2019-08-15 by stale[bot]