espressif/arduino-esp32 #1063: Spiffs mount failed in sparkfun things esp32

Akskaypohankar, 2018-02-02:
Please fill the info fields, it helps to get you faster support ;)

I tried to upload the spiffs_test example to my Sparkfun esp32 things and it gives me following error please help me with this. 

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:10212
entry 0x40078a00
E (22) SPIFFS: mount failed, -10025
SPIFFS Mount Failed

https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?sparkfun esp32 things ?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

Akskaypohankar, 2018-02-02:
it looks like the problem is solved for me.
I found this on espressif idf and with some modifications it looks like it worked of my sparkfiun esp 32 
anyways the solution is sufficient for me. posting this for anyone who needs it. 
the main source of solution.

https://github.com/espressif/esp-idf/blob/master/examples/storage/spiffs/main/spiffs_example_main.c 

modifications for Arduino ide 
#include <stdio.h>
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_spiffs.h"

static const char *TAG = "example";

void setup()
{Serial.begin (115200);
    ESP_LOGI(TAG, "Initializing SPIFFS");
    Serial.println("Initializing SPIFFS");
    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = true
    };
    
    // Use settings defined above to initialize and mount SPIFFS filesystem.
    // Note: esp_vfs_spiffs_register is an all-in-one convenience function.
    esp_err_t ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
            Serial.print("Failed to mount or format filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAG, "Failed to find SPIFFS partition");
            Serial.print("Failed to find SPIFFS partition");
        } else {
            ESP_LOGE(TAG, "Failed to initialize SPIFFS (%d)", ret);
            Serial.printf("Failed to initialize SPIFFS (%d)", ret);
        }
        return;
    }
    
    size_t total = 0, used = 0;
    ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to get SPIFFS partition information");
        Serial.println("Failed to get SPIFFS partition information");
    } else {
        ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
        Serial.printf("Partition size: total: %d, used: %d \n ", total, used);
    }

    // Use POSIX and C standard library functions to work with files.
    // First create a file.
    ESP_LOGI(TAG, "Opening file");
    Serial.print("Opening file");
    FILE* f = fopen("/spiffs/hello.txt", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        Serial.print("Failed to open file for writing");
        return;
    }
    fprintf(f, "Hello World!\n");
    fclose(f);
    ESP_LOGI(TAG, "File written");
    Serial.print("File written");

    // Check if destination file exists before renaming
    struct stat st;
    if (stat("/spiffs/foo.txt", &st) == 0) {
        // Delete it if it exists
        unlink("/spiffs/foo.txt");
    }

    // Rename original file
    ESP_LOGI(TAG, "Renaming file");
    if (rename("/spiffs/hello.txt", "/spiffs/foo.txt") != 0) {
        ESP_LOGE(TAG, "Rename failed");
        Serial.print("Rename failed");
        return;
    }

    // Open renamed file for reading
    ESP_LOGI(TAG, "Reading file");
    f = fopen("/spiffs/foo.txt", "r");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for reading");
        Serial.print("Failed to open file for reading");
        return;
    }
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);
    // strip newline
    char* pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);
    Serial.printf("Read from file: '%s'", line);

    // All done, unmount partition and disable SPIFFS
    esp_vfs_spiffs_unregister(NULL);
    ESP_LOGI(TAG, "SPIFFS unmounted");
    Serial.print("SPIFFS unmounted");
}
void loop ()
{
  }

serial monitor output 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:10212
entry 0x40078a00
Initializing SPIFFSPartition size: total: 1374476, used: 502 
 Opening fileFile writtenRead from file: 'Hello World!'SPIFFS unmounted

lbernstone, 2018-02-02:
If you do not have a SPIFFS partition already formatted, you need to initalize it.  Your example does that with the .format_if_mount_failed = true config, but you can do this with a simple `SPIFFS.begin(true)`

Akskaypohankar, 2018-02-02:
@lbernstone 
well, I tried your suggestions and it looks like it worked you saved me lots of troubles can this modification added to main Arduino examples library for future peoples.

schweini, 2018-04-15:
I can confirm that SPIFFS.begin(true) fixes the issue in the included SPIFFS_Test example sketch,.

copercini, 2018-08-05:
It's easier to format now thanks a @lbernstone commit https://github.com/espressif/arduino-esp32/commit/17065dfd3a55f1e22cb770514a623057d4831724

will close as solved =)

closed 2018-08-05 by copercini

playground, 2021-02-07:
Trying on Atom Smart Speaker development kits, SPIFFS.begin() is throwing an exception but works on ESP32. `Guru Meditation Error: Core 1 panic'ed (Unhandled debug exception) Debug exception reason: Stack canary watchpoint triggered (loopTask) Core 1 register dump: PC : 0x4008c1b7 PS : 0x00060436 A0 : 0x8008c2ac A1 : 0x3ffb0070 A2 : 0x3ffcc7f4 A3 : 0xbaad5678 A4 : 0x3ffcc7e8 A5 : 0x00000001 A6 : 0x00060a23 A7 : 0x00000000 A8 : 0x3ffcc844 A9 : 0x3ffcc7ec A10 : 0x00000002 A11 : 0x3ffcc7e8 A12 : 0x00012508 A13 : 0x00012494 A14 : 0x3ffcc8c0 A15 : 0x3ffcc848 SAR : 0x00000010 EXCCAUSE: 0x00000001 EXCVADDR: 0x00000000 LBEG : 0x400014fd LEND : 0x4000150d LCOUNT : 0xffffffff`