espressif/arduino-esp32 #1064: Does the SHT10 sensor work with ESP32 ?

JacoFourie, 2018-02-02:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?DOIT ESP32 Devkit v 1
Core Installation/update date:			?Current one
IDE name:							?Arduino IDE
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I can not get the SHT10 to work with the ESP32. It works with the ESP8266 and all the Arduinos

### Sketch:
```#include <SHT1x.h>
 
// Specify data and clock connections and instantiate SHT1x object
#define dataPin  2
#define clockPin 15
SHT1x sht1x(dataPin, clockPin);
 
void setup()
{
   Serial.begin(115200); // Open serial connection to report values to host
   Serial.println("Starting up");
}
 
void loop()
{
  float temp_c;  
  float humidity;
 
  // Read values from the sensor
  temp_c = sht1x.readTemperatureC(); 
  humidity = sht1x.readHumidity();
 
  // Print the values to the serial port
  Serial.print("Temperature: ");
  Serial.print(temp_c, DEC);
  Serial.print("C / ");  
  Serial.print(humidity);
  Serial.println("%");
 
  delay(2000);
}
```

### Debug Messages:
```
The output shows a comm problem
Temperature: -40.0000000000C / -4.40%

```

JacoFourie, 2018-02-02:
I changed the pins 

````
#define dataPin  23
#define clockPin 27
SHT1x sht1x(dataPin, clockPin);
````
Now the output is 

````
Temperature: 615.3499755859C / -6114.33%
````

jaco7, 2018-03-06:
i have the same error, did u choose it?

JacoFourie, 2018-03-06:
Nope I have not yet got it working. Still need to see if I can get it working. But i do have the new I2C sensor on the way as it is replacing this one. But I still have a number of them and would love to use it on the ESP32.

beegee-tokyo, 2018-03-06:
I would look into the library. I had big problems with a DHT11 sensor until I changed the library. What was needed was to disable task switching while communicating with the sensor. I added
```
  // ESP32 is a multi core / multi processing chip
  // It is necessary to disable task switches during the readings
  portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
  portENTER_CRITICAL(&mux);
```
before the communication with sensor started and
`  portEXIT_CRITICAL(&mux);`
after the communication ended.
Check [void DHTesp::readSensor()](https://github.com/beegee-tokyo/DHTesp/blob/master/DHTesp.cpp) in my DHTesp library.

jaco7, 2018-03-07:
i ve solved the problem of -40C, the problem was that the sensor cable(2m long) was going with an ADSL(internet) line cable together,in the same tonnel.i put the sensor cable in another way and everething working good!!! i think the problem is a magnetic fields around the cables, and the most magnetic field cables are 220/110V electricity cables.

JacoFourie, 2018-03-07:
@jaco7 please share the library and code you used to get it working. I have tried 2 diffrent boards and different probes all still on a breadboard but I cant get it to work. I assume you are here in South Africa. Can I give you a call ?

jaco7, 2018-03-07:
https://github.com/practicalarduino/SHT1x   that is library i used. in my code im using uploading sketches by WEB(OTA), and using telnet monitor instead of serial monitor(this everything doing through the wifi),and send temp and hum to the web server.so its too difficult will be for u to understand.try this lib i gave u, its 100% working, if not - problem in your code or connections.of course u can call me, no problem, but im a bit far away from Africa,- Ukraine :)

JacoFourie, 2018-03-07:
OK my name is also Jaco. But I am from South Africa. Thought the name was only here. I am using the same library and as you can see it is a simple test app. But it wont work.

jaco7, 2018-03-07:
sht1x lib is the same for esp8266 and esp32.simple test example must work anyway. so the problem is from your side - connections! when my sensor is working it shows right temp/hum,but when i unplug it from my board in the monitor i see only -40c.so in my opinion u have bad connections on your board.look better

JacoFourie, 2018-03-07:
I have a Arduino Mega a ESP8266 and 2 diffrent ESP32 on the bench. I will take the probe and connect it to the Arduino. Then run the sample sketch and all works. then I will move it to the ESP8266 and all works fine. Then I move it to the ESP32 and then it wont work. So the same connections and sensor will work on the other 2 but not any of my ESP32's. What brand of board do you have ?

jaco7, 2018-03-07:
mine is WEMOS D1, its the same as esp8266.it is sends all meteo data to the web page, https://narodmon.ru/6051  here its my sensor plased and temp/hum/pressure datas for now,and updates every 5min.

JacoFourie, 2018-03-07:
OK but the D1 is an ESP8266 not a ESP32. So you are using a ESP8266 not a ESP32 ?
As I said it works on the ESP8266. I have a ESP8266 in the field testing https://demo.thingsboard.io/dashboards/fba6c7c0-f75d-11e7-abe9-1d8d2edf4f93?publicId=62079880-f6e8-11e7-abe9-1d8d2edf4f93

But now I want to use the ESP32. I use these 2 types of boards 

https://www.aliexpress.com/item/ESP32-Development-Board-Serial-Port-WiFi-Bluetooth-Ethernet-Internet-of-Things-Wireless-Transmission-Module-Control-Board/32848324784.html?ws_ab_test=searchweb0_0,searchweb201602_1_10152_10151_10065_10344_10068_10342_10343_10340_10341_10084_10083_10618_10307_5711211_10313_10059_10534_100031_10103_10627_10626_10624_10623_10622_5722411_10621_10620_5711311,searchweb201603_25,ppcSwitch_5&algo_expid=5cc7a5a8-17c2-478d-8080-3e654bc67465-24&algo_pvid=5cc7a5a8-17c2-478d-8080-3e654bc67465&transAbTest=ae803_5&priceBeautifyAB=0

and 

https://www.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Core-ESP-32-ESP-32S-ESP/32815035373.html?s=p&ws_ab_test=searchweb0_0,searchweb201602_1_10152_10151_10065_10344_10068_10342_10343_10340_10341_10084_10083_10618_10307_5711211_10313_10059_10534_100031_10103_10627_10626_10624_10623_10622_5722411_10621_10620_5711311,searchweb201603_25,ppcSwitch_5&algo_expid=5cc7a5a8-17c2-478d-8080-3e654bc67465-2&algo_pvid=5cc7a5a8-17c2-478d-8080-3e654bc67465&transAbTest=ae803_5&priceBeautifyAB=0

JacoFourie, 2018-03-12:
@jaco7 am I correct that you are not using an ESP32 but you are using a ESP8266 ?

JacoFourie, 2018-05-09:
OK so I got it working with this sensor. I needed to add external pull up resistors for the serial and data lines.

closed 2018-05-09 by JacoFourie

DSysoletin, 2019-04-02:
Hello!
I had same problem with SHT10 (I've used library practicalarduino/SHT1x) and ESP32.
I've found working solution - to make it work, we need to put SHT10-related code to core0. Probably there are some timing issues if we run it on core1 (all system routines is also ran on core1).

Code example:

TaskHandle_t TaskA;
//sht10 data
float sht10_temp=0;
float sht10_hum=0;

void setup() {
...
//SHT10 loop on core 0
   xTaskCreatePinnedToCore(
   sht10_loop,                  /* pvTaskCode */
   "Workload1",            /* pcName */
   1000,                   /* usStackDepth */
   NULL,                   /* pvParameters */
   1,                      /* uxPriority */
   &TaskA,                 /* pxCreatedTask */
   0);                     /* xCoreID */
...
}

void sht10_loop(void * parameter)
{
  while(1)
  {
    //portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
    //portENTER_CRITICAL(&mux);
    sht10_temp = sht1x.readTemperatureC();
    sht10_hum = sht1x.readHumidity();
    //portEXIT_CRITICAL(&mux);
    
    
    delay(5000);
  }
}

And, in main code. we just read values of sht10_temp and sht10_hum.
This works good for me.

beegee-tokyo, 2019-11-28:
Finally I had to use a SHT10 sensor in one of my projects.
The existing libraries did not work, regardless which core I used to run the readings.

Examining the existing libraries showed that they are written in a way that would never run reliable on an ESP32. 
E.g. Waiting for an ACK after sending the command to the sensor was done in a loop without using timers, just relying on the fact that a looping 100 times would result in a specific wait time.
Valid way to program if you are on AVR or other slower processors, but this will fail on an ESP32.

I updated the most recent library ([SHT1x](https://github.com/Vincent-Pang/SHT1x) from Vincent Pang) and made it ESP32 compatible.

I added the new library to both ArduinoIDE and PlatformIO library managers, but it might take some time before it will be available.

In the meantime you can install it manually from [SHT1x-ESP](https://github.com/beegee-tokyo/SHT1x-ESP)

JacoFourie, 2019-12-01:
Great. Thanks. Will test it some time.

lilechao, 2019-12-03:
Hello, I used this esp8266-12e, and then used D1, D2 IO port to connect SHT10, but also - 40. Which library do you use

JacoFourie, 2019-12-03:
I got it working using the normal Arduino one but I did need to add a resistor to fix the problem.

lilechao, 2019-12-04:
> I got it working using the normal Arduino one but I did need to add a resistor to fix the problem.

ok,thank you

jucgarciar, 2023-05-01:
buenas noches comunidad, estoy  utilizando el sensor FS200-SHT10 con mi esp32 pero me arrojan valore negativos: 3:23:01.490 -> Temperature: -39.6599998474C / -39.4199981689F. Humidity: -2.69%
23:23:03.494 -> Temperature: -39.6599998474C / -39.4199981689F. Humidity: -2.69%
23:23:05.515 -> Temperature: -39.6599998474C / -39.4199981689F. Humidity: -2.69%
23:23:07.528 -> Temperature: -39.6599998474C / -39.4199981689F. Humidity: -2.69%

el codigo que estoy empleando es el siguiente: 

/**
 * ReadSHT1xValues
 *
 * Read temperature and humidity values from an SHT1x-series (SHT10,
 * SHT11, SHT15) sensor.
 *
 * Copyright 2009 Jonathan Oxer <jon@oxer.com.au>
 * www.practicalarduino.com
 */

#include <Arduino.h>

#include <SHT1x-ESP.h>

// Specify data and clock connections and instantiate SHT1x object
#define dataPin  10
#define clockPin 13

// default to 5.0v boards, e.g. Arduino UNO
SHT1x sht1x(dataPin, clockPin);

// if 3.3v board is used
//SHT1x sht1x(dataPin, clockPin, SHT1x::Voltage::DC_3_3v);

void setup()
{
  Serial.begin(9600); // Open serial connection to report values to host
  Serial.println("Starting up");
}

void loop()
{
  float temp_c;
  float temp_f;
  float humidity;

  // Read values from the sensor
  temp_c = sht1x.readTemperatureC();
  temp_f = sht1x.readTemperatureF();
  humidity = sht1x.readHumidity();

  // Print the values to the serial port
  Serial.print("Temperature: ");
  Serial.print(temp_c, DEC);
  Serial.print("C / ");
  Serial.print(temp_f, DEC);
  Serial.print("F. Humidity: ");
  Serial.print(humidity);
  Serial.println("%");

  delay(2000);
}

ademas estoy emplenado la resitencia de 10K entre data y vcc

el sensor que tengo es el FS200-SHT10 de la version actualizada donde los pines son lo siguientes:
vcc=rojo
negro=GND
amarrillo=sck
verde=datos

ya he realizado muchas pruebas pero no encuentro el error, me podrian ayudar gracias