espressif/arduino-esp32 #1066: Support of unsigned long long in WString

JacekDob, 2018-02-02:
### Hardware:
IDE name:							Arduino IDE
### Description:
Please add support of unsigned long long in WString

helmar74, 2018-03-16:
Added WString.cpp and WString.h ton sonoff directory, got following compile error:

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, DOUT, 80MHz, 4MB (32Mb), 115200, None"

WString.cpp:1: error: 'String' does not name a type

 String::String(unsigned long long value, unsigned char base)

Anything I can try?

juanimoni, 2018-03-17:
Hi JacekDob, like helmar74, I got a compiler error too. 
........\Tasmota\webserver.ino: In function 'void HandleInformation()':

webserver:1934: error: call of overloaded 'String(uint64_t)' is ambiguous

   func += F("}1" D_ESP_CHIP_ID "}2"); func += String(ESP.getChipId());

any advice ?                                                                     ^

JacekDob, 2018-03-18:
To compile tasmota esp32 you need to edit and add in arudino-esp32 files:

In [your_dir]\Arduino\hardware\espressif\esp32\cores\esp32\WString.cpp:
```
String::String(unsigned long long value, unsigned char base)
{
    init();
    char buf[1 + 8 * sizeof(unsigned long long)];
    ultoa(value, buf, base);
    *this = buf;
}
```

In [your_dir]\Arduino\hardware\espressif\esp32\cores\esp32\WString.h:
```
explicit String(unsigned long long, unsigned char base = 10);
```

helmar74, 2018-03-18:
yeah! That fixed it, thank's a lot!

helmar74, 2018-03-18:
Next issue ;-)
I compiled it and send it to a ESP32 it's starting, but the ESP didn't connect to Accesspoint. 
The conifguration in user_config.h is fine, as I'm using the same for other tasmota projects.
Here's output of serial log:
0:00:00 APP: Baudrate eingestellt auf 115200

00:00:00 Projekt sonoffESP32 Sonoff (Topic sonoff, Fallback DVES_00000000, GroupTopic sonoffs) Version 5.12.0c-STAGED
00:00:00 WIF: verbinden mit AP1  in Modus 11
00:00:15 WIF: Verbindung fehlgeschlagen da der AP nicht antwortet (timeout)
00:00:15 WIF: verbinden mit AP1  in Modus 1

I also changed GroupTopic in user_config.h. it's ignored somehow

helmar74, 2018-03-18:
[https://github.com/espressif/arduino-esp32/issues/1234](url)

closed 2018-04-19 by everslick