espressif/arduino-esp32 #1068: SPIFFS reads junk data.

rfestag, 2018-02-02:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation/update date:			???
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Some changes after 12/19/2017 appear to have broken SPIFFS. See https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/6. I  just rolled back this library to the last commit on that day, and reads appear to be correct.

I don't know if this means arduino-esp32fs-plugin needs an update, or if something unexpected changed here.

### Sketch:
```cpp
#include <Arduino.h>
#include <SPIFFS.h>
#include <FS.h>

void setup() {
  Serial.begin(115200);
  delay(2000);      // sanity check delay

  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount SPIFFS");
  }
  Serial.printf("Listing directory: %s\n", "/");

  File root = SPIFFS.open("/");
  if(!root){
    Serial.println("Failed to open directory");
    return;
  }
  if(!root.isDirectory()){
    Serial.println("Not a directory");
    return;
  }
  File file = root.openNextFile();
  while(file){
    Serial.print("  FILE: ");
    Serial.print(file.name());
    Serial.print("  SIZE: ");
    Serial.println(file.size());    
    file = root.openNextFile();
    uint8_t x; while (file.read(&x, 1)) Serial.printf("%02X", x);
  }
  
  Serial.println("Done Setup");
}
void loop() {

}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

hpwit, 2018-02-04:
hello
I did reinstall everything according to the manual but still get the error
i am on mad os X
i have used the git clone command
how can you lake sure I have the right library ?
thank 
rgds
yves
BTW do you know what flash mode QIO DIO mean 'cause the version of the esp32fs that I have always uplaod in DIO
from which date is your esp32fs ?$thank you

lonerzzz, 2018-02-12:
@hpwit QIO and DIO are flash mode speeds which depend on the hardware you are using. You want to make sure that you use the same flash speed as your hardware prints out. In the Arduino IDE, this is shown under the Tools menu as Flash Mode.

closed 2019-08-15 by stale[bot]