espressif/arduino-esp32 #1069: Issue with delayMicroseconds inside ISR or Task

XerXeX, 2018-02-02:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		02/Jan/2018
IDE name:						Arduino IDE
Flash Frequency:					40Mhz and 80Mhz
Upload Speed:					921600

### Description:
I am having a issue when using `delayMicroseconds` inside a ISR or Task

Picture of a oscilloscope attached to the [interruptPin](https://debug.xecr.dk/interruptpin.png)

This is the first code i tried. It is very basic, and just dims a bulb to 30%.

```
void setup() {
  pinMode(12, INPUT_PULLUP);
  pinMode(18, OUTPUT);
  attachInterrupt(12, isr, RISING);
}

void loop() {

}

void isr() {
  delayMicroseconds(7000);
  digitalWrite(18, HIGH);
  delayMicroseconds(500);
  digitalWrite(18, LOW);
}
```
I would expect the output to look a lot like the interruptPin, but [this](https://debug.xecr.dk/isr.png) is what came out, with make no sense.
This method would work perfectly on a Arduino board.

I then tried to do it with a task.

```
void setup() {
  pinMode(12, INPUT_PULLUP);
  pinMode(18, OUTPUT);
  attachInterrupt(12, isr, RISING);
}

void loop() {
}

void isr() {
   xTaskCreate(
    manageDimmer,          /* Task function. */
    "DimmerOne",        /* String with name of task. */
    10000,            /* Stack size in words. */
    NULL,             /* Parameter passed as input of the task */
    10,                /* Priority of the task. */
    NULL);            /* Task handle. */
}

void manageDimmer( void * parameter) {
  delayMicroseconds(7000);
  digitalWrite(18, HIGH);
  delayMicroseconds(500);
  digitalWrite(18, LOW);
  vTaskDelete( NULL );
}
```

wich resulted in [this](https://debug.xecr.dk/task.png) output. Not very useful.

Then I tried to just raise a flag and do it in the main loop.

```
bool fire = false;
void setup() {
  pinMode(12, INPUT_PULLUP);
  pinMode(18, OUTPUT);
  attachInterrupt(12, isr, RISING);
}

void loop() {
  if(fire) {
    delayMicroseconds(7000);
    digitalWrite(18, HIGH);
    delayMicroseconds(500);
    digitalWrite(18, LOW);
    fire = false;
  }
}

void isr() {
   xTaskCreate(
    manageDimmer,          /* Task function. */
    "DimmerOne",        /* String with name of task. */
    10000,            /* Stack size in words. */
    NULL,             /* Parameter passed as input of the task */
    10,                /* Priority of the task. */
    NULL);            /* Task handle. */
}

void manageDimmer( void * parameter) {
  fire = true;
  vTaskDelete( NULL );
}
```

This time I got a perfect [output](https://debug.xecr.dk/loop.png)
Is this some bug with `delayMicroseconds` or am I just wrong by doing it this way?

XerXeX, 2018-02-08:
Really? Nothing?

stickbreaker, 2018-02-08:
@XerXeX The use of a delay inside a ISR is verboten. No one is supporting your design decision.  You need to use the hardware timers Review the [Timer API](https://esp-idf.readthedocs.io/en/v3.0-rc1/api-reference/peripherals/timer.html)

Chuck.

XerXeX, 2018-02-09:
But delay should work fine in a task running on a separate core, right? I'm only using ISR to start the task.

jasoroony, 2018-02-11:
I've just started using the ESP32 (so take any of my comments with a grain of salt) but I have suggested in this post:

https://github.com/espressif/arduino-esp32/issues/282#issuecomment-364723118

To not bother with the slow critical sections (that, incidentally, have an affect of slowing down the Arduino loop call).

I was unaware of any verboten uses of APIs on ESP that work on Arduino, though.  But much like the 4 input only pins (that are called IO pins) I guess there's a learning curve with most things.

I am also going to want to use a delay in an ISR, once I get my head around this chip's idiosyncrasies.  I would be interested in why you are seeing this problem, and avoid using up a valuable timer if it's not needed.

stickbreaker, 2018-02-11:
@jasoroony If you are going to use delays in ISR code Why don't you just use a polling loop in your main loop code? If you don't need to/care if the main loop is hung waiting for some event to happen, just code that feature into the main loop.  

ISR code should be quick, deterministic, and never call code that is not designed/validated to operate in an interrupt context.  

My guidance to use the hardware timer for your interrupt delay is based on prior experience.  If you need to trigger an event after a known interval, that is by definition what a timer is for. 

 You would have two ISR functions:
* One that triggered on your 'initial event'.
This function would load the correct interval (delay) into the pre-configured timer. Then return.
* The second ISR would be connected to the Timer.
When the timer expired it would be triggered.  Inside this timer ISR you would reset/disable the timer, then process your delayed action and return. 

 Using this sequence, no delay(spinning) is required, The two interrupts interfere with the main loop for a short, deterministic amount of time.    

Chuck.

jasoroony, 2018-02-12:
Hi Chuck, I understand and thanks for your comments, what you describe is what GRBL does on an Arduino currently when stepping the CNC motors, where the stepper driver's step pulse needs to be a longish ~12us, and it uses two timers...  But it's a too small delay for reliable deactivation in the main GRBL loop without risking it not pulsing properly in the next ISR stepper call.

However, their LPC1768/9 branch did away with all that because the required pulse delay is so short.  The delay start time is recorded at ISR entry, so the final delay request is much smaller after the ISR code has done its business and used up most of that required pulse time.  I'm happy to continue with their method in my conversion to ESP32 because they are still able to generate high step counts with what might be considered a slow ISR.  Yes, an ISR should be quick, but only as fast as it needs to be for the application - and the simplification in code is great plus the need for allowing further enhancements to the GRBL application is low.

Adding to that, I've seen odd things take place on these 32 bit chips that have more sophisticated devices running than the 8 bit Arduinos, in that ISRs can be delayed (not sure if interrupts are software prioritised in ESP32 like they can be in LPC?).  E.G USB-Serial on Arduino is a separate chip, but not so on all of the 32 bit ones.  I don't want to "reinvent the wheel" and have to fix strange issues when all that was required is a slightly slower ISR that already matches the application's requirements.

Given all that, if *all* the micros() and delay function needs to work is to read special register ccount (which I'll probably code myself because I don't care about overflows in the 17 minute range) , then as long as it's a reliable value when read within an ISR it should all be good.  That's what I was getting at when I said I'm interested in knowing why someone else is seeing this particular issue.  It doesn't make sense the ccount stops in an ISR as it needs to be reliable for what it's being used for here: micros() [that's one question], it might be that it can't be read reliably within an ISR [I would be surprised if that were the case].

I'm just not up to the point where I'm coding software to try this out.  I'm still building the board and doing my research, which I should have done for the 4 input only pins!  (My PCB design is at version 2.2 already!)

jasoroony, 2018-02-13:
https://github.com/espressif/arduino-esp32/issues/282#issuecomment-288687620

This comment says 17 minutes before an overflow, but in testing it's a bit over 17 seconds (240,000,000 * 17 = 4,080,000,000 which is close to a uint32_t's maximum).

I've tried a number of different things, and there doesn't seem to be any problem with a delayMicroseconds call in an ISR.  The pictures in the opening post kind-of prove that, the times of the peaks look wrong, but the width doesn't look bad.

The ISR also doesn't have the IRAM_ATTR modifier, so it is in the slow flash, I'd start with that.

closed 2019-08-15 by stale[bot]