espressif/arduino-esp32 #1070: UART2 LoadProhobited exception

Elektrik1, 2018-02-03:
### Description:
After latest esp-idf and arduino-esp32 UART2 (pins 16,17) has stopped working. Exception on print/write routines. If I comment out .println (leaving only .begin part) no exception is thrown. Same code was working before updates.

### Sketch:
```cpp

#include <Arduino.h>

HardwareSerial Serial2(2);
#define SERIAL2_RXPIN 16
#define SERIAL2_TXPIN 17

void setup() {
      Serial2.begin(9600, SERIAL_8N1, SERIAL2_RXPIN, SERIAL2_TXPIN);
      setup_bt();
}

void loop() {

}

void setup_bt(){
      Serial2.println("HI");
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:00x3fff001c,len:5560
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13296
entry 0x40078f5c
I (30) boot: ESP-IDF v3.1-dev-325-g9ecd718f-dirty 2nd stage bootloader
I (30) boot: compile time 01:43:07
I (32) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label             Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (68) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (75) boot:  2 app0             OTA app          00 10 00010000 00140000
I (83) boot:  3 app1             OTA app          00 11 00150000 00140000
I (90) boot:  4 eeprom           Unknown data     01 99 00290000 00001000
I (97) boot:  5 spiffs           Unknown data     01 82 00291000 0016f000
I (105) boot: End of partition table
I (109) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3cd04 (24
9092) map
I (205) esp_image: segment 1: paddr=0x0004cd2c vaddr=0x3ffc0000 size=0x02070 (
8304) load
I (208) esp_image: segment 2: paddr=0x0004eda4 vaddr=0x40080000 size=0x00400 (
1024) load

Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400e3a8b  PS      : 0x00060830  A0      : 0x800e3ac0  A1      : 0x3f
fce0f0
0x400e3a8b: Print::write(char const*) at C:/Dropbox/Arduino/Sketches/Projects/ES
P32/A2DP/IDF/components/arduino/cores/esp32/Print.cpp:75
 (inlined by) Print::print(char const*) at C:/Dropbox/Arduino/Sketches/Projects/
ESP32/A2DP/IDF/components/arduino/cores/esp32/Print.cpp:84

A2      : 0x3ffc3300  A3      : 0x3ffdcc58  A4      : 0x00000000  A5      : 0x00
000010
A6      : 0x00000057  A7      : 0x00000037  A8      : 0x00000000  A9      : 0x3f
fce0e0
A10     : 0x0000000c  A11     : 0x3ffdcc64  A12     : 0x000000ff  A13     : 0x00
00ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000014  EXCCAUSE: 0x00
00001c
EXCVADDR: 0x0000000c  LBEG    :: 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xf
ffffffc

Backtrace: 0x400e3a8b:0x3ffce0f0 0x400e3abd:0x3ffce110 0x400eb657:0x3ffce130 0x4
00eb697:0x3ffce1a0 0x400e3107:0x3ffce1c0
0x400e3a8b: Print::write(char const*) at C:/Dropbox/Arduino/Sketches/Projects/ES
P32/A2DP/IDF/components/arduino/cores/esp32/Print.cpp:75
 (inlined by) Print::print(char const*) at C:/Dropbox/Arduino/Sketches/Projects/
ESP32/A2DP/IDF/components/arduino/cores/esp32/Print.cpp:84

0x400e3abd: Print::println(char const*) at C:/Dropbox/Arduino/Sketches/Projects/
ESP32/A2DP/IDF/components/arduino/cores/esp32/Print.cpp:75

0x400eb657: setup_bt at C:/Dropbox/Arduino/Sketches/Projects/ESP32/A2DP/IDF/main
/main.cpp:218

0x400eb697: setup() at C:/Dropbox/Arduino/Sketches/Projects/ESP32/A2DP/IDF/main/
main.cpp:86

0x400e3107: loopTask(void*) at C:/Dropbox/Arduino/Sketches/Projects/ESP32/A2DP/I
DF/components/arduino/cores/esp32/main.cpp:15

```

baggior, 2018-02-06:
maybe is related to #991 
The UART2 prints correctly only the first packet.. the followings are garbled

tablatronix, 2018-02-07:
I think exceptions were just added to compile, so maybe preexisting ?
https://github.com/espressif/arduino-esp32/commit/52d47054c2cbc4554cd8605235cf24540ce6d91c

closed 2018-09-27 by Elektrik1