espressif/arduino-esp32 #1071: i2C problems

dnienhis, 2018-02-03:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Espressif ESP32 ESP32-DEVKITC
Core Installation/update date:			25 Jan 18
IDE name:							Arduino IDE 1.8.5 - Windows 10
Flash Frequency:					80 Mhz
Upload Speed:						921600

### Description:

Trying to read time from the Adafruit DS1307 Board (https://www.adafruit.com/product/3296) - completely functional with ESP8266 & Arduino UNO over i2C - unable to read from ESP32.   This is the test/demo code supplied by Adafruit - also using their library - the only thing I changed was adding and removing the #include <WiFi.h> when compiling for ESP32.

On the ESP32 I connected the i2C signals as follows:

SCL - GPIO21
SDA - GPIO22

(the connections didn't match up with what i'd read online - i found these via Oscilliscope & trial & error while running demo code below).

I'm attaching Oscilliscope captures from each of the three test scenarios.

This is my first time to post so please 'be gentle' LOL !

[image]
[image]
[image]
[image]
[image]

### Sketch:
// Date and time functions using a DS1307 RTC connected via I2C and Wire lib
 
//#include <WiFi.h>
#include <Wire.h>
#include "RTClib.h"
 
#if defined(ARDUINO_ARCH_SAMD)
// for Zero, output on USB Serial console, remove line below if using programming port to program the Zero!
   #define Serial SerialUSB
#endif
 
RTC_DS1307 rtc;
 
char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};
 
void setup () {
 
#ifndef ESP8266
  while (!Serial); // for Leonardo/Micro/Zero
#endif
 
  Serial.begin(9600);
  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    while (1);
  }
 
  if (! rtc.isrunning()) {
    Serial.println("RTC is NOT running!");
    // following line sets the RTC to the date & time this sketch was compiled
    rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
    // This line sets the RTC with an explicit date & time, for example to set
    // January 21, 2014 at 3am you would call:
    // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
  }
}
 
void loop () {
    DateTime now = rtc.now();
   
    Serial.print(now.year(), DEC);
    Serial.print('/');
    Serial.print(now.month(), DEC);
    Serial.print('/');
    Serial.print(now.day(), DEC);
    Serial.print(" (");
    Serial.print(daysOfTheWeek[now.dayOfTheWeek()]);
    Serial.print(") ");
    Serial.print(now.hour(), DEC);
    Serial.print(':');
    Serial.print(now.minute(), DEC);
    Serial.print(':');
    Serial.print(now.second(), DEC);
    Serial.println();
   
    Serial.print(" since midnight 1/1/1970 = ");
    Serial.print(now.unixtime());
    Serial.print("s = ");
    Serial.print(now.unixtime() / 86400L);
    Serial.println("d");
   
    // calculate a date which is 7 days and 30 seconds into the future
    DateTime future (now + TimeSpan(7,12,30,6));
   
    Serial.print(" now + 7d + 30s: ");
    Serial.print(future.year(), DEC);
    Serial.print('/');
    Serial.print(future.month(), DEC);
    Serial.print('/');
    Serial.print(future.day(), DEC);
    Serial.print(' ');
    Serial.print(future.hour(), DEC);
    Serial.print(':');
    Serial.print(future.minute(), DEC);
    Serial.print(':');
    Serial.print(future.second(), DEC);
   Serial.println();
   
    Serial.println();
    delay(3000);
}
 
 

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

BlackTiger44, 2018-02-03:
https://github.com/espressif/arduino-esp32/issues/1020#issuecomment-362754979

stickbreaker, 2018-02-03:
@dnienhis try substituting my version of the I2C subsytem.  just grab my release 0.1.2 at [stickbreaker/arduino-esp32 release 0.1.2](https://github.com/stickbreaker/arduino-esp32/releases)

The current main stream esp32 library is unreliable my version works, but there are issues with merging it.

Chuck.

dnienhis, 2018-02-04:
Thanks Chuck - I updated with your files and still get nothing back from the DS1307.   Checked GPIO21 & GPIO22 with scope - no signal on either pin.   Then went easter egging with scope - found what looks like might be the SCL (and i'm speculating) on the GPIO1 pin - that's the only pin i see any changes on?

dnienhis, 2018-02-09:
I just downloaded a fresh copy of everything and all appears to be fine now.  Thx!

closed 2018-02-09 by dnienhis

68gt500, 2018-04-07:
Thank you Chuck!
 
For the very first time, since I bought the esp32, I have been able to reliably read the I2C bus - over longer timespans - without it hanging.
I must have spend hours if not days trying to figure out the cause and trying different approaches and workarrounds..

Your work ist truely appreciated. 

Now I can  give this thing a new chance - was about to throw it in the garbage.

stickbreaker, 2018-04-07:
@68gt500 good car, but i lusted after a Mach 1

Now if i can just get Slave and Multi-master.

Chuck

68gt500, 2018-04-07:
Hello Chuck,

thanks, we are a Ford crazed family..

I actually do own a 68 GT500, my son in law has a 69 Mach 1.

Keep up the good work!

Greetings

Michael

<mention@noreply.github.com>
Datum:  07.04.2018 09:49

DeanGreenhough, 2018-04-13:
Can someone kindly explain what I need to do to use this: arduino-esp32-stickbreaker-V0.2.0.zip

I have unzipped. Do I need to find the existing files and replace them with the unzipped files?  Sorry to have to ask, as a newbie, just a little concerned. I believe this will really help me an i2C issue I have encountered. Many thanks for your time

stickbreaker, 2018-04-13:
@DeanGreenhough yes, just replace the existing files with the ones from the zip archive.

esp* in to cores/esp32/
Wire* into libraries/Wire/ 

Chuck.

DeanGreenhough, 2018-04-14:
Hi Chuck
Many thanks for your prompt reply. Running my previous code the maximum loops I was able to  carry out was 200, with your modification and running overnight Im looking at 956 loops and continuing without crashing my program.

I woke up at 4 am, just to check, thank you so much for your work, this has caused me weeks of issues and has made my weekend, month maybe even year!!!!

Bless you Chuck...Big Thumbs Up

DeanGreenhough, 2018-04-14:
Chuck

Just to confirm, the issue is with the esp core library?, I assume this will get sorted out in development?

As this has caused me so much pain, thought maybe I should at least try to have a basic understanding of why there is an issue.

regards

Dean

stickbreaker, 2018-04-14:
@deangreenhough the Main branch i2c hardware abstraction layer(HAL) is based on a polling model.  It tries to control the hardware by reading status bits in a tight loop, but, the esp32 environment is a multi process, interrupt driven RTOS. WiFi, Serial, Timer ... or some other hi priority task "interrupts" this tight loop.  This interruption results in missing a status change. The i2c HAL will only work reliably as a very simple sketch. My HAL is designed to use the hardwares capabilities instead of work around them. For more info read the Wiki in my fork.

Chuck.

DeanGreenhough, 2018-04-14:
Chuck

Thank you very much for your detailed and patient replies, as instructed I will head over to WiKi. Many thanks for enlightening me on  an almost impossible task. Very grateful and still chuckling at the stability over 5000 loops, no errors.

cpassuel, 2018-04-21:
Thanks for your work Chuck

I had weird issues with I2C and no clues what problem I was facing and then your code fixed everything (arduino-esp32-stickbreaker-V0.2.0.zip on PlateformIO).

joaocarlosss84, 2018-05-07:
Hi everybody,

I was facing several problems using I2C until I found the @stickbreaker branch, thank you very much!!!
I downloaded it, fixed the compiling problem when I merged the latest version of ESP32 for ArduinoIDE for Windows at: 
https://github.com/espressif/arduino-esp32.git
and the "arduino-esp32-stickbreaker-V0.2.0.zip" at:
https://github.com/stickbreaker/arduino-esp32/releases

Now I can debug why my I2C code is not working with my sensor.

Any idea when espressif will add this bugfix to their main branch?

stickbreaker, 2018-05-07:
There is a staging branch in this repo stickbreaker-i2c.  So, work is progressing.

Chuck.

Rob58329, 2018-06-19:
@stickbreaker

Many thanks for this "github.com/stickbreaker/arduino-esp32" (latest v24May18) software - it fixed my i2c hanging issue perfectly!

FYI. I was using a "NodeMCU-32S" with Arduino-IDE and the latest "github.com/espressif/arduino-esp32" software.  I was having an issue with an i2c LED 8x8Matrix consistently hanging the i2c bus every 6 hours or so (requiring an "ESP.restart();" to recover from). Your above i2c software has fixed this (1 week running now without any hangs - yay!)

stickbreaker, 2018-06-19:
@Rob58329 You're welcome.

I would recommend you try the staging branch of the official repo.  Instead of my repo.

[espressif-arduino/tree/stickbreaker-i2c](https://github.com/espressif/arduino-esp32/tree/stickbreaker-i2c)

We need a lot of testing before it is moved into the main branch.  So far there have not been any issues found, but me-no-dev is being caucus before it replaces the existing i2c subsystem.

Chuck.

OslafEman, 2018-11-24:
Thank you very much Stickbreaker. I am completely new in this fascinating field. My project has scaled from Arduino Uno to ESP32, going through Arduino Mega and ESP8266. I almost went crazy looking for the reason why my DS1307 and my keypad (with PCF8574) stopped working well with the ESP32 if they worked perfectly with the previous ones.
I believe that today I will sleep calmly.

hazee007, 2018-11-28:
@stickbreaker thank you for the library but am still having a bit of an issue. am trying to using the esp32 with accelorometer and gyroscope module (gy 512).
the issue is that theres a delay with the reading of the module into the serial monitor compare to esp 8266 and arduino which was better. i know the library u gave is in early stages but is there anyway i can get upto or same speed on arduino to esp32.
thanks

stickbreaker, 2018-11-28:
@hazee007 open a new issue, post your code.  I'll have to see your code before I can advise.

There are no delays encoded in the drive, But, the i2c slave device can stretch SCL to pause communications. 

I use it to animate a 128x64 oled display at 50frames/sec.  So it is pretty fast!

Chuck.

stickbreaker, 2018-11-28:
`Wire.endTransmission(false);` will return 7 to indicate ReSTART is queued until a STOP is encountered.

stickbreaker, 2018-11-29:
Open a new issue, your problem has nothing to do with this issue.

Post your current code in the new issue using "markup" commands. The "markup" commands cause github to render the code in a readable format.

Make it easier for me to help you.

hazee007, 2018-11-29:
Sorry was sending from my mail just saw this now, heres the code.
`#include "Wire.h"
#include "SPI.h"
#include "SD.h"

#define MPU6050_ACCEL_XOUT_H       0x3B   // R 
#define MPU6050_PWR_MGMT_1         0x6B   // R/W
#define MPU6050_PWR_MGMT_2         0x6C   // R/W
#define MPU6050_WHO_AM_I           0x75   // R

#define MPU6050_I2C_ADDRESS 0x68
typedef union accel_t_gyro_union
{
  struct
  {
    uint8_t x_accel_h;
    uint8_t x_accel_l;
    uint8_t y_accel_h;
    uint8_t y_accel_l;
    uint8_t z_accel_h;
    uint8_t z_accel_l;
    uint8_t t_h;
    uint8_t t_l;
    uint8_t x_gyro_h;
    uint8_t x_gyro_l;
    uint8_t y_gyro_h;
    uint8_t y_gyro_l;
    uint8_t z_gyro_h;
    uint8_t z_gyro_l;
  } reg;
  struct
  {
    int16_t x_accel;
    int16_t y_accel;
    int16_t z_accel;
    int16_t temperature;
    int16_t x_gyro;
    int16_t y_gyro;
    int16_t z_gyro;
  } value;
};

// Use the following global variables and access functions to help store the overall
// rotation angle of the sensor
unsigned long last_read_time;
float         last_x_angle;  // These are the filtered angles
float         last_y_angle;
float         last_z_angle; 
float         last_gyro_x_angle;  // Store the gyro angles to compare drift
float         last_gyro_y_angle;
float         last_gyro_z_angle;

void set_last_read_angle_data(unsigned long time, float x, float y, float z, float x_gyro, float y_gyro, float z_gyro) {
  last_read_time = time;
  last_x_angle = x;
  last_y_angle = y;
  last_z_angle = z;
  last_gyro_x_angle = x_gyro;
  last_gyro_y_angle = y_gyro;
  last_gyro_z_angle = z_gyro;
}

inline unsigned long get_last_time() {return last_read_time;}
inline float get_last_x_angle() {return last_x_angle;}
inline float get_last_y_angle() {return last_y_angle;}
inline float get_last_z_angle() {return last_z_angle;}
inline float get_last_gyro_x_angle() {return last_gyro_x_angle;}
inline float get_last_gyro_y_angle() {return last_gyro_y_angle;}
inline float get_last_gyro_z_angle() {return last_gyro_z_angle;}

[... 209 lines cut ...]
  n = Wire.write(start);        // write the start address
  if (n != 1)
    return (-20);

  n = Wire.write(pData, size);  // write data bytes
  if (n != size)
    return (-21);

  error = Wire.endTransmission(true); // release the I2C-bus
  if (error != 0)
    return (error);

  return (0);         // return : no error
}

int MPU6050_write_reg(int reg, uint8_t data)
{
  int error;

  error = MPU6050_write(reg, &data, 1);

  return (error);
}

  `