espressif/arduino-esp32 #1077: How can I turn off the radio on ESP32?

JLuisAlcaraz, 2018-02-04:
Hi,
I need, in order to save battery, turn off the radio and the BLE. How can I do that on Arduino IDE?

CelliesProjects, 2018-02-04:
I turn off BLE with `btStop();`
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-bt.c

JLuisAlcaraz, 2018-02-04:
Thank you very much!

CelliesProjects, 2018-02-04:
For wifi you can use:

`WiFi.mode( WIFI_MODE_NULL );`

should work in Arduino IDE.

JLuisAlcaraz, 2018-02-09:
With "WiFi.mode( WIFI_MODE_NULL);" The current compsuntion is the same.
I teste with and without that line and I dont see current variation.
Is there another way to turn off the radio part on the ESP32??

CelliesProjects, 2018-02-12:
I am not sure, the above info came from issue https://github.com/espressif/arduino-esp32/issues/75.

CelliesProjects, 2018-02-12:
Maybe turn of both radios?
They may be implemented as a single radio that is multiplexed.

copercini, 2018-02-12:
There is an example for this: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiBlueToothSwitch/WiFiBlueToothSwitch.ino

to turn wifi and Bluetooth off:
```cpp
WiFi.mode(WIFI_OFF);
btStop();
```

JLuisAlcaraz, 2018-02-13:
Thank you very much, friends!

closed 2018-02-13 by copercini

daton89, 2019-07-28:
Stopping bluetooth with `btStop();` will save energy?

mrdazhuk, 2021-03-10:
`btStop` doesn't work. It freezes the app

me-no-dev, 2021-03-10:
@mrdazhuk should be fixed in the latest master

mrdazhuk, 2021-03-10:
@me-no-dev the problem still exists.

me-no-dev, 2021-03-10:
This works for me in Master and I can no longer reproduce your problem. Did you uninstall the core from the board manager first?
```cpp
  SerialBT.begin("ESP32test");
  delay(1000);
  SerialBT.end();//calls btStop();
```

mrdazhuk, 2021-03-10:
Let me check. Thank you

mrdazhuk, 2021-03-10:
@me-no-dev could you please share your configuration from platformio.ini. 
seems to me i'm doing something wrong, because BluetoothSerial.cpp is little different. I've removed everything, but still different

me-no-dev, 2021-03-10:
@mrdazhuk sorry, I do not use PIO