espressif/arduino-esp32 #1078: CONFIG_CLASSIC_BT_ENABLED

0x1abin, 2018-02-04:
Hi, I want enable Classic BT use the SPP to implement the BT Serial Arduino library, but the **sdkconfig** default config isn't enable **CONFIG_CLASSIC_BT_ENABLED** and **CONFIG_BT_SPP_ENABLED**.
It will enable it by default in the future?  Or provide a easy way to enable it? 
Thank you.

yoh-there, 2018-02-18:
+1. Classic BT support would be very, very welcomed.

anteph, 2018-03-04:
Hi! I've been able to run Bluetooth Classic stuff by replacing the libbt.a file from the following PR in my ESP32 Arduino core installation:
https://github.com/espressif/arduino-esp32/pull/1144

I did a tutorial on a very basic application to make the ESP32 discoverable. Not sure if it helps, but I will leave it here:
https://techtutorialsx.com/2018/03/04/esp32-arduino-bluetooth-classic-getting-started/

Best regards,
Nuno Santos

copercini, 2018-08-05:
It's enabled by default now, when there are some code using Classical BT, for example https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

closed 2018-08-05 by copercini