espressif/arduino-esp32 #1079: COM port busy after "WiFi.begin();"

p99will, 2018-02-04:
### Hardware:
Board:							ESP32D0WDQ6 (revision 1) / WEMOSLOLIN32. It has a built-in OLED screen.
Core Installation/update date:			xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip -- not too sure? but i got the latest with git clone.
IDE name:							Arduino IDE, Windows 10 x64
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
COM port busy after "WiFi.begin();"

I'm unable to upload new sketches or Monitor serial when this funtion is called in my code.

The only way I can is by booting the board with Boot button held, and/or held while uploading.

My diagnostic summery:
Changed ports on PC,
Changed assigned COM port,
Reinstall USB drivers with drivers provided for board,
Checked for other software opening the COM port,
Changing boaud rates on: Board, IDE and COM driver.
Downloading latest build of espressif libaries and boards.
Running Arduino's WiFi Libary (this worked with no COM port problems).
Attempting to connect to COM port using wireshark, putty etc.. Same problem.

Literally removing that one line of code.

I suspect the esp32 WiFi lib or some SPI idk... is using a GPIO that conflicts with the serial controller on the board.

I tried looking through the code for libary but couldnt make sence of it. I don't know much about things and stuff.

### Sketch:
```cpp

#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  
  WiFi.begin("test", "hello");  // Omitting this line stops the problems with COM ports.

}

void loop() {

}
```

### Debug Messages:
```
Error opening serial port 'COM2'. (Port busy) // when opening serial monitor
--
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM2': WindowsError(121, 'The semaphore timeout period has expired.') //when uploading.

```

moriczgergo, 2018-02-04:
Some other program is reading from COM2.

AIWIndustries, 2018-02-05:
I have the same problem that just started yesterday.

if I include Wifi.begin(ssid,password) the serial port stops working. If I remove it, port works fine

Does not work
`#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.println(F("Test"));
  WiFi.begin("RSM2","rsmhome1230");

  while (WiFi.status() != WL_CONNECTED) {
    //Serial.println("Connecting to the WiFi network");
    delay(500);
  } 
}
void loop() {
  Serial.println(F("Test"));   
  delay(1000);
}`

Works:
`void setup() {
  Serial.begin(115200);
  Serial.println(F("Test")); 
}

void loop() {
  Serial.println(F("Test"));   
  delay(1000);
}`

AIWIndustries, 2018-02-05:
I loaded Arduino 1.8.5 and reloaded the ESP32 core. No change. I'm running windows 10. 

My laptop with Windows 10 works just fine so I copied over the esp32 hardware folder to my desktop and now it works. 

Is there a way to look in the esp32 folder for a specific file that would identify the version? If so, I'll post what is working.

AIWIndustries, 2018-02-05:
This ended up notbeing an issue with the esp32 library at all. It was a power issue. What I found was that even though I got it to work once after transferring the folder from laptop to desktop it was still occurring. If I shutdown the serial window and reopened it after the wifi had already connected then everything worked. I transferred the USB cord from one port on the front of the desktop to one that was connected on the back and everything works just fine. 

Assumption was that during WiFi.begin( ) there is a large power surge and there was not enough current available from the USB port I was using to sustain comms via the silicon labs IC.

tablatronix, 2018-02-07:
Please close the issue then

closed 2018-02-10 by p99will