espressif/arduino-esp32 #1081: WiFi - Static IP fails to connect wifi

mistergreen, 2018-02-05:
### Hardware:
Board:							WEMOS lolin32
Core Installation/update date:			Jan/2018
IDE name:							Arduino IDE

### Description:

```
IPAddress ip(192,168,1,129);
IPAddress gateway(192,168,1,1);	
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8); //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional

if (!WiFi.config(ip, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
  }

  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
```

WiFi.status() won't connect.

If I get of WiFi.config, going with the localIP, Wifi can connect.

jliegner, 2018-02-09:
the esp fire in this mode not the event SYSTEM_EVENT_STA_CONNECTED but only SYSTEM_EVENT_STA_GOT_IP. 
In WiFiGeneric.cpp Line 318:
``````
        WiFiSTAClass::_setStatus(WL_IDLE_STATUS);
    } else if(event->event_id == SYSTEM_EVENT_STA_GOT_IP) {
// JL    
//         if(WiFiSTAClass::status() == WL_IDLE_STATUS)
           {
            WiFiSTAClass::_setStatus(WL_CONNECTED);
        }
    }
``````
i comment out the line that check of WL_IDLE_STATUS. This help.

baggior, 2018-02-11:
Same as #1086

baggior, 2018-02-20:
created the PR #1129 to fix this

copercini, 2018-03-05:
Closed due it's fixed here: https://github.com/espressif/arduino-esp32/pull/1129

Thanks @baggior !

PR #1129 (2018-03-04): fix issue 1081 and 1086 Wifi connection error when using STATIC IP settings
fixes
#1081 
#1086 

(to be merged into master as soon as possible please)
closed 2018-03-05 by copercini