espressif/arduino-esp32 #1083: analogRead problem.

JLuisAlcaraz, 2018-02-05:
Hi all!
I'm trying to read the output of a electret micro board. The board is this:
https://it.aliexpress.com/item/CJMCU-9812-MAX9812L-Electret-Microphone-Amplifier-Development-Board-Sensor-Module-For-Arduino-Free-Shipping/32769959503.html?spm=a2g0y.search0104.3.256.2719444fmmSDsr&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10344_10068_10547_10342_10343_10340_10548_10341_10084_10083_10618_10307_10301_10303_5722315_10313_10059_10184_10534_100031_10103_441_10624_442_10623_10622_10621_10620_10142,searchweb201603_2,ppcSwitch_5&algo_expid=6251e8c8-0330-42a2-a638-640a41db8892-38&algo_pvid=6251e8c8-0330-42a2-a638-640a41db8892&priceBeautifyAB=0

The wiring is:
Micro          ESP32
VCC             3.3V
GND            GND
OUT            GPIO35

To read the signal, I tested 2 ways, the analogRead(35) and this way:
On SetUp:
`adc_power_on();
  adc1_config_width(ADC_WIDTH_12Bit);
  adc1_config_channel_atten(ADC1_CHANNEL_7,ADC_ATTEN_11db); //Micro`

On Loop:
`      nivel = adc1_get_raw(ADC1_CHANNEL_7);`

In both cases, the readings for ambient sound start at 1500 and go down slowly to reach 0. If I speak on the micro, the values are correct, but I dont understand why the signal is going down for the ambient sound

lrmoreno007, 2018-02-05:
MAX9812 has a power save mode.
http://blog.digit-parts.com/pdf/max9812-max9813l.pdf

JLuisAlcaraz, 2018-02-05:
Thanks!
I don't understand. Same wiring with an Arduino Nano gives me a stable reading on ambient sound.

lrmoreno007, 2018-02-05:
which is the top mark in your MAX9812? ABJ ?

JLuisAlcaraz, 2018-02-05:
Yes, ABJ.

lrmoreno007, 2018-02-05:
This is your circuit:
[image]

It's exactly as this one:
[image]

With this schema power save never must be triggered. OK.

---
On the other hand, your MAX9812 works with 3.3v and maybe it is damaged because you used it with an Arduino Nano with 5v.
[image]

I can't think of anything else.

JLuisAlcaraz, 2018-02-05:
I power the module by 3.3V out from Nano, and the values read are stable.

lbernstone, 2018-02-07:
Can you put a multimeter on the pins and see if there is actual drift matching what you see on the esp32, or if it is reading incorrectly?  That would rule out power issues.

closed 2019-08-15 by stale[bot]