espressif/arduino-esp32 #1084: ESP32 not showing serial port after Mac installation

danthegoodman1, 2018-02-06:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			Yesterday
IDE name:							Arduino IDE
Flash Frequency:					N/A
Upload Speed:						N/A

### Description:
So I am using an ESP32 Dev Module, and a 2017 MBP 13" and after following the arduino installation tutorial for the ESP32 I have the boards, but the ESP32 dev module does not show up as a serial port.

Any help would be great, thanks.

huming2207, 2018-02-06:
Have you installed the serial port driver?

Please run this command ```ls /dev | grep tty``` and paste the result here.

danthegoodman1, 2018-02-06:
I have not installed any serial port drivers (I think?)
Here is the absurd output:
`tty
tty.Bluetooth-Incoming-Port
tty.MALS
tty.SOC
ttyp0
ttyp1
ttyp2
ttyp3
ttyp4
ttyp5
ttyp6
ttyp7
ttyp8
ttyp9
ttypa
ttypb
ttypc
ttypd
ttype
ttypf
ttyq0
ttyq1
ttyq2
ttyq3
ttyq4
ttyq5
ttyq6
ttyq7
ttyq8
ttyq9
ttyqa
ttyqb
ttyqc
ttyqd
ttyqe
ttyqf
ttyr0
ttyr1
ttyr2
ttyr3
ttyr4
ttyr5
ttyr6
ttyr7
ttyr8
ttyr9
ttyra
ttyrb
ttyrc
ttyrd
ttyre
ttyrf
ttys0
ttys000
ttys001
ttys002
ttys1
ttys2
ttys3
ttys4
ttys5
ttys6
ttys7
ttys8
ttys9
ttysa
ttysb
ttysc
ttysd
ttyse
ttysf
ttyt0
ttyt1
ttyt2
ttyt3
ttyt4
ttyt5
ttyt6
ttyt7
ttyt8
ttyt9
ttyta
ttytb
ttytc
ttytd
ttyte
ttytf
ttyu0
ttyu1
ttyu2
ttyu3
ttyu4
ttyu5
ttyu6
ttyu7
ttyu8
ttyu9
ttyua
ttyub
ttyuc
ttyud
ttyue
ttyuf
ttyv0
ttyv1
ttyv2
ttyv3
ttyv4
ttyv5
ttyv6
ttyv7
ttyv8
ttyv9
ttyva
ttyvb
ttyvc
ttyvd
ttyve
ttyvf
ttyw0
ttyw1
ttyw2
ttyw3
ttyw4
ttyw5
ttyw6
ttyw7
ttyw8
ttyw9
ttywa
ttywb
ttywc
ttywd
ttywe
ttywf`

mistergreen, 2018-02-06:
Your board might use USB to UART. You'd need another driver for that.
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

danthegoodman1, 2018-02-06:
> Your board might use USB to UART. You'd need another driver for that.
Will try that tomorrow and drop in here how it goes!

mistergreen, 2018-02-06:
My ESP32 didn't show up on my MAC either until I installed that driver.

danthegoodman1, 2018-02-06:
So it doesn't seem like it is installing them on my laptop. The files from the uninstall script are not there.

richardfuchs, 2018-02-13:
> So it doesn't seem like it is installing them on my laptop. 

For MacOS X 10.9 and 10.10, see the folder called "Legacy MacVCP Driver".
It may also help if you drag&drop out the file "Silicon Labs VCP Driver.pkg" to any folder (e.g. Downloads) and start it from there

JohnGEyles, 2018-02-27:
ESP32 crashes my OSX 10.11 machine when I try to upload my sketch (filed a separate issue awhile back, but no answers).

mistergreen, 2018-02-27:
If your OS X is crashing, it could be because of the USB driver. My OS X crashed when I plugged in my ESP8266.
Try https://github.com/adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver

WeiboboHe, 2018-05-05:
Hi danthegoodman1. I almost met the same problem with ESP32. Did you figures it out now?

JohnGEyles, 2018-05-05:
FWIW (I think it's mentioned in the other thread I started), mine worked when I used one of the USB ports on the back of my Apple monitor (a hub there, presumably) instead of the ones on the front of my Mac Pro tower.   I guess this wouldn't apply to the OP's Macbook Pro though.   Maybe try it with a powered hub ?   Try a shorter thicker cable ?   Add a battery to the ESP32 so it has plenty of power ?

Mr-taweep, 2018-05-20:
Goog that "richardfuchs"

bakzs, 2018-06-21:
Hi, I have the same problem too. downloaded the recommended silicon labs drivers for usb to uart but still no luck. My Mac OS version is High Sierra 10.13.5. 

Does anyone have the same problem too?

Chris2060, 2018-06-23:
@bakzs  Yeah I do have the exact same problem on my macbook pro... I have a Esp 32 Devkit V1 and installed the driver from espressif and the recomended one above, but I still cant see it in the arduino IDE. Im sad...
Someone any idea how to fix it?

JohnGEyles, 2018-06-23:
Try the various USB ports, try a different cable, try a USB hub ...

Chris2060, 2018-06-23:
@JohnGEyles Thanks for your advice!

I used the port at the back right with a different usb cable after a restart of the laptop. Now it works... ;-)

n8leon, 2018-07-04:
Installing the silabs Silicon Labs VCP Driver.pkg worked for me also. 👍 
(on OS X 10.11.6)

Seems obvious that we need the drivers, but I thought I already had them after going thru the espressif/arduino-esp32 install steps.
e.g. My "System Report / Hardware / USB" showed
   CP2102 USB to UART Bridge Controller driver

For others' future reference, you should see something like:
n8s-iMac:nleon$ ls -al /dev/tty.*
crw-rw-rw-  1 root  wheel   18,   2 Jul  4 11:14 /dev/tty.SLAB_USBtoUART

Regardless, thanks for the pointer!

AcrimoniousMirth, 2018-07-10:
Having this issue too. 
I downloaded all the drivers I found from different sources including the ones mentioned here. It flashed up as a recognised device briefly but then vanished again and I can't reproduce the result. 
My board doesn't have a visible chip so I'm assuming its under the RF shield. Navigating to `About this Mac →  System Report →  USB` shows a _"Composite Device"_ with Product ID _"0xea60"_ by Silicon Labs so I hunted that down and its one of the **CP2102s**...

Still hunting.

mattcony, 2018-07-14:
Anyone notice MAC drivers were updated yesterday - still no joy though

CP210x Macintosh OSX VCP Driver 5.0.5    - July 13, 2018

Known Issues and Limitations
----------------------------

	This release includes the Macintosh OSX driver for 64 bit Intel 
	Macintosh computers running MacOS X 10.11 or greater.
	For MacOS X 10.9 and 10.10, see the folder called "Legacy MacVCP Driver".

AcrimoniousMirth, 2018-07-14:
Yeah, so I’ve tried identifying the device several other ways and it keeps showing up the same. My father’s old MacBook Pro also identifies it the same but when trying to view it in the Arduino IDE can’t find it, gets the same SOC and MALS. 
Haven’t tried a Windoze machine because don’t have one handy!

AcrimoniousMirth, 2018-07-14:
I’m actually wondering if this is more an issue with the Arduino IDE? Anyone had any luck with other IDEs?

mattcony, 2018-07-14:
Just caved and tried a win8.1 machine, up and running in 15min. I thought my new ESP32 board was bad, tried every usb cable in the house with no Joy on my MAC. Windows machine is fine, my MAC cannot even see the board in terminal so has to be driver related still.

danthegoodman1, 2018-07-14:
Guys there are 2 drivers, the CP210x and CH340. I got those installed and
now all esp8266 and esp32 work with my 2017 mbp.

kiranshashiny, 2018-07-17:
I am unable to see the SiLabs driver for the ESP32 Port  on my Mac OS - High Sierra 10.13.3 for the ESP32.
After going over others comments above - I installed both the CP2102 and the CH340 ( since I play with the Arduino and ESP8266 ) but the port for the ESP32 is not showing up.
I have multiple ports and so I plugged in the ESP8266 - and I got the "cu.wchusbserial1410" and this is good. But the /dev/cu.SLAB_USBxxx for the ESP32 is not showing !

Note : I rebooted my Mac after installing the driver.

lionep, 2018-07-22:
I had the same issue on a Macbook pro 2017.
I was using USB-C to USB-A adaptor, and USB-A to micro usb cable.  
I added a basic usb 2.0 hub between the adaptor and the cable, and it's now detected.

Odi55555, 2018-07-27:
Had the same issue with bei 2017 MBP. I bought a new USB A to Micro USB cable and now it works.

JakeHartnell, 2018-08-23:
Had the same issue on a 2018 MBP, spent ages trying different cables and reinstalling drivers. On my Linux machine it just works but I have to use the MBP for work so to get it working:
- [disabled Gatekeeper](https://www.easymacsupport.com/blog/august-29th-2017) and reinstalled the driver
- Disable [System Integrity Protection](https://www.macworld.co.uk/how-to/mac/how-turn-off-mac-os-x-system-integrity-protection-rootless-3638975/)
- Disabled kext signing

Note, you may not need all those steps; I went overboard cause I am also trying to get a Linux partition working. Anyway, hope that helps someone. : )

lbernstone, 2018-08-23:
@JakeHartnell : Seems like it would just be easier to install virtualbox and a linux vm.

adddog, 2018-09-02:
[Silabs drivers](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) worked for me. Had to run the `uninstaller.sh` and then, in my case, install Legacy.

Now I see `/dev/tty.SLAB_USBtoUART` in  Arduino IDE.

pallikas, 2018-09-09:
I had the same issue, then I realized that the usb cable i was using was set up for power only, and was not passing the signals.  I was burned by this kid of cable once before, and need to put label tags on the power only cables. Works fine, with the CP2102 Drivers and the proper cable.

twoblokeswithapostie, 2018-09-13:
@ mistergreen Your silabs driver recommendation worked for me on High Sierra. I just had to allow loading in System Preferences and unplug and plug in the board. I can now see tty.SLAB_USBtoUART under /dev/. Thank you!

daspuru, 2018-09-28:
I solved it this way,

1. Use terminal to go to this directory: /Library/Extensions
2. Run  sudo kextload SiLabsUSBDriver.kext/
3. If the output is an alert, go to System Preferences, and click the Allow button like this:
[image]

4. Then run again the command, if nothing is in the output, you are in.
5. Use 115,200 BPS setup for programming.

jstarr, 2018-10-03:
Had the same problem (macpro running High Sierra).  AcrimoniousMirth said we needed both CP210x and CH340 drivers.  I only found the CP210X driver on the Silab site.  Did a search for CH340 macos drivers and found https://medium.com/@tony92151/arduino-ch340-chip-driver-for-high-sierra-39e621dae1e4 which pointed to https://github.com/adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver/raw/master/CH34x_Install_V1.3.pkg which ended up working for me.   The ls /dev/cu.* returns:
/dev/cu.usbmodem1411.

artwebstudios, 2018-11-05:
Hi have the same issue. 

I have tried both drivers CP210x and CH340. Both of them works occasionally but after restart the computer does not work. Even when I am trying to reinstall them. It does not work. 

Any idea how to solve this issue?

Tobils, 2018-11-27:
daspuru i have the same issue on my macbook air, with high siera. the problem was solved by click button allow on System Preferences the choose Security and Setting, then on bellow of the  page click allow. dont forget to choose . driver that you've installed. it's working for me.

Tobils, 2018-11-27:
> Hi have the same issue.
> 
> I have tried both drivers CP210x and CH340. Both of them works occasionally but after restart the computer does not work. Even when I am trying to reinstall them. It does not work.
> 
> Any idea how to solve this issue?

try to click allow on Security & Privacy button.

Tobils, 2018-11-27:
> I solved it this way,
>
> 1. Use terminal to go to this directory: /Library/Extensions
> 2. Run  sudo kextload SiLabsUSBDriver.kext/
> 3. If the output is an alert, go to System Preferences, and click the [...]

 i have the same issue on my macbook air, with high siera. the problem was solved by click button allow on System Preferences the choose Security and Setting, then on bellow of the page click allow. dont forget to choose . driver that you've installed. it's working for me.

djaramil, 2018-12-31:
I tried all kinds of things to get my ESP32 to work from here and from SLAB website to no avail.  My ESP8266 worked fine with no problem.
The one thing that eventually led me to figuring out the problem was to go to the About Mac / System Report / USB and that is when I saw that my ESP did not register under USB.  I then got another ESP32 and tried that one and it then showed up on the list and the drivers also showed up too after that.
[image]
Sometimes it is a hardware problem.  :-(
I hope this helps.

JohnGEyles, 2019-01-01:
Yep, I think mine may have been a bad USB cable too.

marctheshark3, 2019-01-17:
Hello, I followed the instructions: download the two different drivers and allow in the system preferences in the security settings. What is weird is that I have a flora board that is able to connect no problem but my huzzah esp32 has no luck. So I know my ports and cable work properly. 
OS: Mojave: 10.14.2

Anything else I can try guys??

marctheshark3, 2019-01-17:
> I tried all kinds of things to get my ESP32 to work from here and from SLAB website to no avail. My ESP8266 worked fine with no problem.
> The one thing that eventually led me to figuring out the [...]

So my system seems to be able to find the device like above yet the arduino program isnt picking it up. So is it possible I have a bad board too??

AhmedShaalan, 2019-01-18:
Same Problem here, My module is shown as "Composite Device" when connected via AmazonBasics USB Type C to Micro B 2.0 Cable with CP2102 Drivers installed on my MacBook Pro 2016. Although the cable is labeled as data and charging
However, it works fine with other USB-A to Micro USB.

Any idea why is that happening?

AhmedShaalan, 2019-01-18:
> I found a thread discussing my issue with Mojave OS and recommended installing this driver: https://www.ftdichip.com/Drivers/VCP.htm
> 
> I installed it and after I was able to load my code to my huzzah32. Success! Hope this is able to help others.

It didn't work with my USB-C cable. 
Thanks anyways

marcelodib, 2019-02-01:
- Download driver: 
    https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
- Open zip and .dmg
- Go to legacy folder and install
- Open terminal
- Go to /Library/Extensions
- Run sudo kextload SiLabsUSBDriver.kext/

It solve the problem for me.
my board: DOIT ESP32 DEVKIT V1
MBP 2017

mbbender, 2019-03-07:
Took me a while to read the readme... but for me I did not receive a warning when installing but I still had to execute

    To unblock, open the System Preferences Security & Privacy pane and
    unblock the system extension. See Apple Technical Note TN2459
    "User-Approved Kernel Extension Loading" for more information.

thelinguist, 2019-03-07:
@mbbender it took me at least 5 hours of wondering why my driver wasn't working to read the readme. Mac asks for your permission to run the installer, but afterwards you must ALSO give permission for the driver to run. Once I did what it said in the readme, and opened System Preferences to give permission, it worked!

besi, 2019-03-08:
I had some USB-C hubs that were not working. So if `/dev/cu.SLAB_USBtoUART` does not show up in your file system you might want to try another USB-C to USB hub.

djaramil, 2019-03-09:
Also some USB cables are not compatible.  I have seen that myself. 

Also, you have to install the right version of driver for the version of Mac OS too.  Older Mac OS wants the older driver.  

Newer Mac OS new driver plus all those security pop ups, etc

Regards
David Jaramillo 
djaramil@fau.edu
561-703-6860

oricoh, 2019-03-18:
> I solved it this way,
>
> 1. Use terminal to go to this directory: /Library/Extensions
> 2. Run  sudo kextload SiLabsUSBDriver.kext/
> 3. If the output is an alert, go to System Preferences, and click the [...]

I wasted hours, you are the best!

millergd, 2019-04-08:
FYI about the security pop ups, it only popped up in security preferences during the download. Once the download was over, **it disappeared**.

flevin58, 2019-05-02:
Thank you all for the tips I found here.
Today I got my ESP32-WROOM 32 from Amazon and could not connect.

I have a Mac Book Pro Retina mid 2012 with Mojave 10.14.4

Solved the issue installing as suggested the Silicon Labs USB to UART drivers from this site:
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

The issue was also to choose the correct Arduino settings:

Board: ESP32 Dev Module
Upload Speed: 921600
Port: /dev/cu.SLAB_USBtoUART
Programmer AVRISP mkII

When uploading the code must press BOOT on device to avoid timeout issues

Thanks again !!!!
Fernando

MacRaeMac, 2019-05-11:
I am using MBP Mid 2009 which using OSX Mojave, tried this driver https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers, the `/dev/cu.SLAB_USBtoUART` appear but when it upload sketch to device it failed and keep timeout.

flevin58, 2019-05-11:
I solved that issue doing as follows:
1) first click the upload button from Arduino IDE (or VS Code PlatformIO extension)
2) you should see some dots on the terminal indicating the sw is trying to connect to device
3) press BOOT on ESP32 for a second and then release
4) you should see the upload progress on the terminal.

Note: I found out (at least with my device) that when you press BOOT you should "feel" a click on your finger, otherwise the timeout will persist.

Hope it helps.

ankita-anand1, 2019-08-13:
I changed the USB cable and it worked

jdieg0, 2019-08-15:
> Your board might use USB to UART. You'd need another driver for that.
> https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

Thank you! I used to write to the board (a TTGO [Paxcounter](https://github.com/cyberman54/ESP32-Paxcounter)/`ttgov21new`) on my MacBook Pro (macOS High Sierra 10.13.6) from mid 2010. I'm not sure, how I managed to get the board running on my old laptop, but on my new MacBook Air 2018 running macOS Mojave 10.14.6 the board did not appear anymore as `/dev/tty.SLAB_USBtoUART`. Installing [this driver](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) did the job!

It appears under `/dev` again:

`ls -la /dev/tty*`

And uploading via Platformio works again.

```
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.SLAB_USBtoUART
Uploading .pio/build/usb/firmware_ttgov21new_v1.7.841.bin
```

So just in case someone is searching for the upload error, I got this one:

```
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
=============================== [ERROR] Took 9.92 seconds ===============================

======================================= [SUMMARY] =======================================
Environment ota [IGNORED]
Environment usb [FAILED]
Environment dev [IGNORED]
========================= 1 failed, 0 succeeded in 9.93 seconds =========================
The terminal process terminated with exit code: 1
```

nalzok, 2019-08-31:
I'm using [this ESP32 development board](https://www.amazon.com/HiLetgo-ESP-WROOM-32-Development-Microcontroller-Integrated/dp/B0718T232Z), and [this USB cable](https://www.amazon.com/gp/product/B00UUBRX0Y/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) from Amazon. After install [the driver from Silicon](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers), I see a "Composite Device" in System Information, but `ls -la /dev/cu.*` still can't find anything new.

Any advice? I'm using macOS Mojave 10.14.6 on MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports).

[image]

mistergreen, 2019-08-31:
try another usb port.

nalzok, 2019-09-01:
I have two ports in total. Neither works.  :(

ankita-anand1, 2019-09-01:
Try using a different USB cable, not all of them work...

monomente, 2019-09-29:
For me I had to install the driver AND change to a different USB cable!

[75 more replies]

closed 2022-03-29 by VojtechBartoska