espressif/arduino-esp32 #1086: Is 'WL_IDLE_STATUS' correct if 'STA_GOT_IP' ?

mijnmodelbaan, 2018-02-06:
https://github.com/espressif/arduino-esp32/blob/c8596cb9bb9f15578cc6ac0582edcd81976e585c/libraries/WiFi/src/WiFiGeneric.cpp#L319

I'm trying to create a static IP address on an ESP32 WROOM, but IMHO the ESP is not IDLE anymore if it has gotten an IP address. Please correct me if I'm wrong about this but I've been searching for some kind of time table to find out when exactly the IDLE status is, but couldn't find it. Creating a dynamic IP causes no problem (working correctly) but creating a static IP gives a problem in WiFiGeneric.cpp, line 319.

mistergreen, 2018-02-06:
I started a Static ip issue yesterday.  status() returns a 6 - disconnect with a static ip config().

baggior, 2018-02-11:
I an confirm this issue

baggior, 2018-02-20:
created the PR #1129 to fix this

copercini, 2018-03-05:
Closed due it's fixed here: https://github.com/espressif/arduino-esp32/pull/1129

Thanks @baggior !

closed 2018-03-05 by copercini