espressif/arduino-esp32 #1089: Can not use vTaskList?

akaneko1019, 2018-02-07:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP-WROOM-32?
Core Installation/update date:			2018/02/07
IDE name:							?Arduino IDE?
Flash Frequency:					?80Mhz?
Upload Speed:						?115200?

### Description:

I could not use vTaskList.
Both configuse_TRACE_FACILITY and configUSE_STATS_FORMATTING_FUNCTIONS are defined to 1.

Please help me.

### Sketch:

TaskHandle_t th[4];
char ptrTaskList[250];

void task1(void *pvParameters) {
  while (1) {
    delay(10);
  }
}
void task2(void *pvParameters) {
  while (1) {
    delay(10);
  }
}
void task3(void *pvParameters) {
  while (1) {
    delay(10);
  }
}
void task4(void *pvParameters) {
  while (1) {
    delay(10);
  }
}

void setup() {

  Serial.begin(115200);
  xTaskCreatePinnedToCore(task1, "task1", 2048, NULL, 1, &th[0], 1);
  xTaskCreatePinnedToCore(task2, "task2", 2048, NULL, 2, &th[1], 1);
  xTaskCreatePinnedToCore(task3, "task3", 2048, NULL, 3, &th[2], 0);
  xTaskCreatePinnedToCore(task4, "task4", 2048, NULL, 4, &th[3], 0);

  vTaskList(ptrTaskList);
  Serial.println(F("**********************************"));
  Serial.println(F("Task  State   Prio    Stack    Num"));
  Serial.println(F("**********************************"));
  Serial.print(ptrTaskList);
  Serial.println(F("**********************************"));
}

void loop() {
  delay(10);

}

### Debug Messages:
Archiving built core (caching) in: d:\tmp\arduino_cache_304660\core\core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_32289f39719329fa04ee6ee8ff063aa9.a
sketch\vtasklist_test.ino.cpp.o:(.literal._Z5setupv+0x4c): undefined reference to `vTaskList'

sketch\vtasklist_test.ino.cpp.o: In function `setup()':

D:\arduino\vtasklist_test/vtasklist_test.ino:32: undefined reference to `vTaskList'

collect2.exe: error: ld returned 1 exit status

exit status 1
ボードESP32 Dev Moduleに対するコンパイル時にエラーが発生しました。

forthlightning, 2018-02-07:
you may need to go into `/esp-idf/components/freertos/include/freertos/FreeRTOSConfig.h` and change to 1

```
#define configUSE_TRACE_FACILITY		0		/* Used by vTaskList in main.c */
#define configUSE_STATS_FORMATTING_FUNCTIONS	0	/* Used by vTaskList in main.c */
```

akaneko1019, 2018-02-08:
I tried it but it was useless.

beegee-tokyo, 2018-02-08:
According to https://github.com/espressif/esp-idf/issues/416 vTaskList is not a part of esp-idf.

akaneko1019, 2018-02-09:
Oh really. It is very disappointing.
By the way, do not know how to know the status of all the other tasks?
If you have something, I would like to tell.
Thank you.

lonerzzz, 2018-02-09:
@akaneko1019 Keep a handle to the task yourself and you can then check its state later on:
```
TaskHandle_t handle;
xTaskCreate( task_loop, name, stackSize, &parameterList, 1, &handle );
...
eTaskGetState( handle );
```

akaneko1019, 2018-02-14:
Thank you for a good reply. I actually let you use that function, but what I want to know most is all of the tasks generated within ESP32.
As for what it means, if you use WiFi etc., the total number of tasks required by "uxTaskGetNumberOfTasks ()" is a large number such as 18.
I need to know all this breakdown.

Please lend me a little more.

lonerzzz, 2018-02-15:
Not all of the functionality of FreeRTOS is currently exposed so you will be limited in that manner. You may want to look at the IDF project that the Arduino project uses as the latest release exposes more of the FreeRTOS functionality. However, it still may not expose all that you would like to see.

akaneko1019, 2018-02-15:
understood.
Thank you everyone.
I will close it.

However, if these commands are supported, we will be able to manage more precisely, so we want to wait for it.

closed 2018-02-15 by akaneko1019

yogeshwaran774, 2019-12-09:
> 
> 
> @akaneko1019 Keep a handle to the task yourself and you can then check its state later on:
> 
> ```
> TaskHandle_t handle;
> xTaskCreate( task_loop, name, stackSize, &parameterList, 1, &handle );
> ...
> eTaskGetState( handle );
> ```

i am also trying to vTaskList() from FreeRTOS library in arduino but same compilation error is coming and i changed the #define configUSE_TRACE_FACILITY and #define configUSE_STATS_FORMATTING_FUNCTIONS  to 1. But error persist and if i use eTaskGetState(handle), does it show current task status like suspended, resume, blocked and delete?

rene1997, 2020-01-20:
I added the following lines in `/esp-idf/components/freertos/include/freertos/FreeRTOSConfig.h`

`#define CONFIG_FREERTOS_USE_TRACE_FACILITY 1`
and
`#define CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS 1`

after rebuilding the project it was possible to use the function `vTaskList`

kraxor, 2021-01-09:
I know this was a year ago, but maybe this helps others who landed here...

Not sure if this was the case back then, but now you can update `CONFIG_FREERTOS_USE_TRACE_FACILITY` and `CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS` in `menuconfig` so you don't need to edit the IDF source, which saves you from a lot of headache, e.g. you don't need to depend on a custom version of IDF, settings remain project-specific, etc.

RadioDisney, 2024-05-19:
> I know this was a year ago, but maybe this helps others who landed here...
>
> Not sure if this was the case back then, but now you can update `CONFIG_FREERTOS_USE_TRACE_FACILITY` and [...]

Can you tell me the next detailed steps?