espressif/arduino-esp32 #1090: SD - card Data logger.

shaddow501, 2018-02-07:
Hello

I am trying to implement the data logger on the ESP32.

this is my part of the code.

   myFile = SD.open("/logs/Battery.csv", FILE_WRITE);
   myFile.printf("%02d-%02d-%02d ", now.date(), now.month(), now.year());
   myFile.printf("%02d:%02d:%02d ", now.hour(), now.minute(), now.second());
   myFile.printf(",");
    if (sinput.startsWith("Battery")){
    String Battery = getValue(sinput, ',', 1);
    myFile.println(String(Battery));
    
    // close the file:
    myFile.close();
     }
     
  } else {
    // if the file didn't open, print an error:
    DBG_OUTPUT_PORT.println("error opening Battery.csv");
  }

The specific code did work good with the ESP8266, but with the ESP32 it only update the first line of the CSV file but do add other lines, it just replace the first line each time...

All the examples I did find in google show that the command myFile.println(String(Battery)); (with the "ln") should add a new line, but it only replace the first line each time.

What do  I do wrong? why doesnt it work the same as it did with the ESP8266?

pipi61, 2018-02-07:
try FILE_APPEND
FILE_WRITE: open the file for writing, starting at the beginning of the file and replacing any existing content.
FILE_APPEND: open the file for reading and writing, starting at the end of the file.

shaddow501, 2018-02-07:
Thanks a lot it works :)

I have another question about the SD file creation, the file created with date of 1980 something.
How can I create a file that will take the current date from the clock and create the file with the current date?

In the older ESP8266 I used Sdfat lib that I could have this parameter -  root.ls(LS_R | LS_DATE | LS_SIZE);
But now I cannot use it, all the info that I have found is to use this  - root.ls(LS_R | LS_DATE | LS_SIZE); but it doesnt work in the ESP32,  so does anyone know how to create the file with the current date?

shaddow501, 2018-02-15:
No one knows?

pipi61, 2018-02-15:
check SdFat-master/SdFat-master/html/class_fat_file.html
timestamp...dateTimeCallback...

shaddow501, 2018-02-15:
Dear pipi61
Thanks for your answer, Yes as I have said I used SDFAT on the esp8266 but with the ESP32 this version you  entered does not compile.

CGB01, 2018-05-28:
I'm having the same issue.

You can get the file timestamp by calling;
time_t t= file.getLastWrite();

But the SD.h nor the FS.h has no provision to set a callback to get the current date/time to properly write the file's date attributes.

Anybody?

closed 2019-08-15 by stale[bot]