espressif/arduino-esp32 #1093: Erro:

kaline, 2018-02-09:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:esp32							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:arduino							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:
Upload Speed:

### Description:
I tried upload a code to esp32 but i found out that memory problem when I try see in the monitor.

### Sketch:
```cpp

![2018-02-08](https://user-images.githubusercontent.com/31135896/36004780-40a066a4-0d13-11e8-81bb-4b0d10ef236d.png)

lbernstone, 2018-02-09:
What memory error?  It says there is a problem with your I2C.  Most likely you have the wrong address, or your I2C device is connected to the wrong pins (usually 21 & 22).  Please run an I2C scanner and see if it finds the device. https://playground.arduino.cc/Main/I2cScanner

BlackTiger44, 2018-02-09:
welcome to the club

https://github.com/espressif/arduino-esp32/issues/1071

https://github.com/espressif/arduino-esp32/issues/1020

lonerzzz, 2018-02-09:
Have either of you tried to use the separate project by @stickbreaker https://github.com/stickbreaker/arduino-esp32 ? This is working for now.

eohan, 2018-02-10:
@lonerzzz , How do you use the project by @stickbreaker in platformio ? I am using the mainstream esp32 arduino library.

marian-craciunescu, 2018-02-11:
@eohan  you can manually copy the file from stickbreaker in your .piolibdeps folder

eohan, 2018-02-14:
Copying the entire folder from 
https://github.com/stickbreaker/arduino-esp32 to a .piolibdeps doesn't work.

lonerzzz, 2018-02-15:
Try copying just the files associated with the changes, namely
libraries/
  Wire.h
  Wire.cpp
cores/esp32
  esp32-hal-i2c.h
  esp32-hal-i2c.h

to replace the files you currently have. I don't use PlatformIO so don't know what else is required.

ytemelli, 2018-08-01:
void setup() {
  pinMode(SDA,INPUT_PULLUP);
  pinMode(SCL,INPUT_PULLUP);

copercini, 2018-09-21:
Fixed with I2C core RC1

closed 2018-09-21 by copercini