espressif/arduino-esp32 #1095: EEPROM library not saving or reading bytes

lonerzzz, 2018-02-10:
### Hardware:
Board:							ESP-WROOM-32
Core Installation/update date:			2017/11/26
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
We recently received a shipment of manufactured boards that use the ESP-WROOM-32 module. The packaging of this module is different from the previous ESP-WROOM-32 that we had been using during development with markings for CE and FCC certifications now present. We are able write code to this module and to write to SPIFFs using the updater tool so flash writes are possible.

However, we are unable to either read or write successfully using the EEPROM library. With both read and write, no errors are generated and the esp_partition_write call in the commit() function is returning a 0/ESP_OK response. No errors occur, but subsequent attempts to read the flash just written always read as high bits giving 0xFF bytes and 0xFFFF shorts for example.

Is there a need for a new type of module definition with different partitions for the new ESP-WROOM-32 module? Currently the "ESP32 Dev Module" is the type we have been using. 

Are there changes in IDF version that may have affected this?

As a reference, the code works without issue on the ESP-WROOM-32 modules we used for development.

Unfortunately we are totally stuck without the ability to write to flash from code.

lonerzzz, 2018-02-12:
Two issues were appearing at the same time, bad FLASH on some modules and a broken version of mkspiffs (https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/6).

closed 2018-02-12 by lonerzzz

markome, 2020-03-06:
I have had the same problem migrating from classical Arduino to ESP32 and I solved it using two lines of code.

The first line was put into the setup() method:
1: EEPROM.begin(512); // need to add this to esp32 in order for eeprom to work. 512 was arbitrary chosen size of Flash to use.

after writing the data to the flash, (EEPROM.put(address, value);)
a call to the function which commits changes was required:
2: EEPROM.commit(); //also needs to commit in order to save cahnegs.

ssp5zone, 2020-07-25:
To anyone who stumbles here. Giving a slight delay after begin() before you read/write anything solved my problem.

```C
EEPROM.begin(0x20);
delay(2000); // Some delay
```