espressif/arduino-esp32 #1097: ESP WROOM 02 - A fatal error occurred: Timed out waiting for packet header

vjkanth, 2018-02-10:
Hello

I'm very new to the whole Arduino scene and recently purchased https://www.aliexpress.com/item/Lolin-ESP32-wemos-ESP32-WiFi-Modules-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-X-Battery-Shield/32810419853.html?spm=a2g0s.9042311.0.0.T7MAHc
I'm trying to upload successfully compiled samples from espressif such as "SimpleTime.ino" but each time I get "A fatal error occurred: Timed out waiting for packet header" error. So far I've tried:
1. Keeping the FLASH button pressed before plugging it in the USB and releasing after the esptool.py scripts shows "Connecting..."
2. Tried changing the board type to ESPea32 but no joy
3. Changed the baud rate to the lowest and everything in between no joy
4. Changed the USB cable (I know its unlikely but clutching at straws here!) still stuck at the error.

The current selected board is WEMOS LOLIN32 set at flash frequency 80MHz (as per the sample in the product link above)

Anyone tried this and know how to resolve this?

copercini, 2018-02-11:
-Use the best and short USB cable that you have, usually, those that manufacturers send with smartphones are good enough
-The upload speed should work fine with 115200 
-Try with the flash frequency as 40Mhz
-And as you said: Keeping the FLASH button pressed before plugging it into the USB and releasing after the esptool.py script show "Connecting..."

If none of them works can be a bad soldering, take a look [in this issue](https://github.com/espressif/arduino-esp32/issues/491) it have a lot of tips about it

vjkanth, 2018-02-11:
@copercini thanks for your response. Tried all of the above still no luck... I've ordered a new one to confirm if it's not the board itself....

lbernstone, 2018-02-13:
I like to test connectivity using the esptool directly.  Open a command prompt, and run `python esptool.py chip_id` . This will query the chip, and will run whether the program button is pushed or not.  It might also give a bit more detail than the arduino ide terminal will show. I really like how they list the "spontaneous blue light" as a feature on that board.

vjkanth, 2018-02-21:
I bought a new board (https://www.amazon.co.uk/MakerHawk-NodeMcu-Development-ESP8266-Compatible/dp/B071S8MWTY) but unfortunately the behaviour is exactly the same. Before I throw the whole arduino crap out of the window and call it a day, is there anyone who'd genuinely faced this issue and can help? I've trawled through all forums with people facing similar issues and tried all sorts of button presses and changes to baud rates and even tried different board managers. If these things are selling, surely people must've had some success with them?

vjkanth, 2018-02-22:
OK so I managed to get the new wemos D1 mini board programmed with a simple world clock example (yay!). The issue was actually with power. It appears that for whatever reason when the upload is performed the power isn't enough to do the job. I had read this in other threads here but didn't take much notice. I applied external power and that resolved it in an instant.
Now, the new board works the OLED display from the previous kit doesn't. I can see the 'output' being generated by the board in the Serial Monitor. However the display is not lighting up with anything. I'm using the display that came with the initial kit, I believe I have the right set of drivers and have changed the example code matching the pinout (D1/D2 for SCL and SDA) of the new board. Also used external 5v power still no joy. Any help would be appreciated.

closed 2018-02-22 by vjkanth

WayneRiesterer, 2018-08-15:
I got mine working by unplugging the OLED module. This would be the same situation mentioned by @vjkanth .

ofSanjay, 2018-10-01:
Recently i bought esp32 module with ch340G. I try everything but it always show time out. Now i try .5 sec interval of switching flash switch constantly. 
When this will appear stop pressing button...

Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2621.5 kbit/s)...
Hash of data verified.
Compressed 15088 bytes to 9755...

Writing at 0x00001000... (100 %)
Wrote 15088 bytes (9755 compressed) at 0x00001000 in 0.9 seconds (effective 136.5 kbit/s)...
Hash of data verified.
Compressed 599232 bytes to 360799...

Writing at 0x00010000... (4 %)
Writing at 0x00014000... (8 %)
Writing at 0x00018000... (13 %)
Writing at 0x0001c000... (17 %)
Writing at 0x00020000... (21 %)
Writing at 0x00024000... (26 %)
Writing at 0x00028000... (30 %)
Writing at 0x0002c000... (34 %)
Writing at 0x00030000... (39 %)
Writing at 0x00034000... (43 %)
Writing at 0x00038000... (47 %)
Writing at 0x0003c000... (52 %)
Writing at 0x00040000... (56 %)
Writing at 0x00044000... (60 %)
Writing at 0x00048000... (65 %)
Writing at 0x0004c000... (69 %)
Writing at 0x00050000... (73 %)
Writing at 0x00054000... (78 %)
Writing at 0x00058000... (82 %)
Writing at 0x0005c000... (86 %)
Writing at 0x00060000... (91 %)
Writing at 0x00064000... (95 %)
Writing at 0x00068000... (100 %)
Wrote 599232 bytes (360799 compressed) at 0x00010000 in 32.4 seconds (effective 148.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 768.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

and it work done uploaded

froderi1, 2018-11-19:
I had the same exact problem. It is this one https://de.aliexpress.com/item/WeMos-D1-Esp-Wroom-02-Motherboard-ESP8266-Mini-WiFi-Nodemcu-Module-18650-Battery/32824587872.html?spm=a2g0x.search0104.3.9.5a601bd7diBuvJ&ws_ab_test=searchweb0_0,searchweb201602_2_10320_10065_10068_10547_319_10548_317_5733915_10696_10924_453_10084_454_10083_5733715_10618_10920_10921_10304_10922_10307_10820_10821_537_10302_536_5733815_10843_10059_10884_5734015_10887_100031_10319_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=1b8c39ce-9d24-446e-aff1-47443219699e-1&algo_pvid=1b8c39ce-9d24-446e-aff1-47443219699e 
Chip says ESP-WROOM-02.
Then I tried to compile and flash using libs, code and board settings for ESP8266 and it worked just fine. So just pretend it is a NODEMCU 12E and you will be fine.

chegewara, 2018-11-19:
@froderi1 But you understand that you have esp8266 chip on this board and this is not the same as esp32?
WROOM-02 board is not esp32.

wan420, 2018-12-05:
Cant believe I read this entire post only to realize its a esp8266 problem and not a esp32 one

hal9000-x, 2019-03-15:
i had the same problem. tried a good power supply, no result. but after cleaning and re-soldering the ESP-WROOM-32D module, it's now working.

basavaraj1995, 2019-03-27:
change the baud rate to 115200

x-cubed, 2019-04-21:
I was having this issue with a cheap ESP-WROOM-32 dev board that I had purchased online.
After much mucking around, I discovered that the SD1 pin was somehow tied down to ground, preventing communication with the flash chip in the module. After desoldering the castellation and scraping away the remaining metal on the castellation, I was able to get esptool to recognize the flash chip correctly.

furqan4545, 2019-05-24:
I Got the solution, watch this video on youtube and u will get it, i suffered from the same problem too but this video resolved the problem, anyways thanks me later, the link is below:
https://youtu.be/ji71cHaGW8w

contactlogicone, 2019-06-09:
mine working by unplugging the OLED module

hn-nguyen, 2019-06-09:
Mine working by change `Upload speed` to 115200.
```
esptool.py v2.6
Serial port /dev/cu.wchusbserial640
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
```

karen3980, 2019-09-09:
i enter this code:
#include <SoftwareSerial.h>
#define TIMEOUT 5000 // mS  

SoftwareSerial esp01(7, 6); // RX, TX
String AP = "gratis";
String PASS = "prayogo27";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  esp01.begin(9600);
  esp01.println("AT+CWMODE=1");
  delay(1000);
  esp01.println("AT+CWJAP=\"" + AP + "\",\"" + PASS + "\"");
  delay(1000);
}

void loop() {
  // put your main code here, to run repeatedly:
  SendData();
  delay(5000);

}

void SendData()
{
ulang :
  String cmd = "AT+CIPSTART=\"TCP\",\""; cmd += "192.168.43.74" ;
  cmd += "\",80";
  esp01.println(cmd);
  long t = millis() + TIMEOUT;
  while (millis() < t)
  {
    if (esp01.available()) {
      char ch = esp01.read();
      Serial.print(ch);
    }
  }
  Serial.println();

  if (esp01.find("ERROR"))
  {
    Serial.println("Koneksi error");
    esp01.println("AT+CIPCLOSE");
    goto ulang;

  }

  cmd = "GET /afhd/index.php/afhd/status_?value=123\r\n\r\n";
  esp01.print("AT+CIPSEND=");
  esp01.println(cmd.length());
  t = millis() + TIMEOUT;
  while (millis() < t)
  {
    if (esp01.available()) {
      char ch = esp01.read();
      Serial.print(ch);
    }
  }
  if (esp01.find(">"))
  {
    Serial.println("Terhubung");
  }
  else if (esp01.find("ERROR"))
  {
    Serial.println("Koneksi error.");
    esp01.println("AT+CIPCLOSE");
    goto ulang;

  }
       
  /* t=millis()+TIMEOUT;
    while(millis() < t){
    if (esp8266.available()){
     char ch = esp8266.read();
     Serial.print(ch);
    }
    }
    Serial.println();
  */
  esp01.print(cmd);
}

and then i got this error:

Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), ck, 40 MHz, 80MHz, DOUT (compatible), 512K (no SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Sketch uses 268520 bytes (53%) of program storage space. Maximum is 499696 bytes.
Global variables use 27584 bytes (33%) of dynamic memory, leaving 54336 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
_
  File "C:\Users\Karen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/Karen/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/Users/Karen/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

what the problem here....btw im new to this thing
thanks for the help

beegee-tokyo, 2019-09-09:
Wrong Repository, here is for ESP32 not for ESP8266. You might post your issue [here](https://github.com/esp8266/Arduino/issues)

karan6190, 2019-09-09:
> @karan6190 Wrong Repository, here is for ESP32 not for ESP8266. You might post your issue [here](https://github.com/esp8266/Arduino/issues)

@karen3980 this repo is for ESp32, please post your answer on ESP8266 repo as mentioned by @beegee-tokyo

beegee-tokyo, 2019-09-09:
@karan6190 sorry, typo.

joehou45, 2020-10-20:
I encountered the same problem, and the problem disappeared after changing the Dupont line.

Druder123, 2022-12-28:
I fixed it by disconnecting the GND, I think that some input that you use is harmful. I recommend you disconnect all the cables and see if that works.