espressif/arduino-esp32 #1098: Low Power API's

LRagji, 2018-02-11:
### Hardware:
Board:							DOIT
Core Installation/update date:		Not Sure
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600

### Description:
I am making a DIY wall clock with internal battery of 6000mAh & Matrix display, i need my ESP32 to be in low power mode googling got me following pointers which i need to do to reduce my power consumption i need API for them are the available

1. Use only one core
2. Reduce Speed of the core to 80Mhz from 160Mhz
3. Put you program in ULP Core

Are these API's available s that i cant program my ESP32 accordingly ...?

copercini, 2018-02-11:
These options are on `make menuconfig` of [ESP32 IDF](https://github.com/espressif/esp-idf)

For your project, I think you can put everything on ULP without using the main processors (or maybe just use it to sync the time with some NTP servers), so the battery will have a long life, but this is not so easy to code. 

In any case, you should carefully choose your hardware (most of development boards components will waste more power than the ESP32 itself)

LRagji, 2018-02-11:
I am doing the same i am just using the wifi core to get NTP time which consumes 200mA for 5 to 10 sec then i shut off the wifi radio.. which reduces the power down to 100mA but thats still too much power for batteries.  So was thinking to put the program on ULP do we have any example sketch for these said apis for arduino core?

And i have bypassed all the other components like the LDO, pwr led by connecting the batteries directly to 3v pin on board, is there anything else to get rid off? i heard even the FTDI chip will draw 15mA which i have left for now, untill i have settled on my final program

Also the strange observation is if i keep my esp on deep sleep it forgets the time so after every deep sleep i have reconnect to get NTP sync and then continue which is also taking up too much power...

Also tried BLE options but that too is eating up 150mA too high kind of defeats the purpose of calling it BLE..,

Is there any other recommendation?

LRagji, 2018-04-16:
I dont think ULP api's will be out soon, also reading through different blogs tells me RTC on esp is not straight forward, it is a mess with different clock source which makes it inaccurate :(, until they release some rock-solid implementation for battery powered devices with arduino core, i have settled for Atmega with HC-12 module, kind of lesser power consumption

ucrania, 2019-05-23:
Try to go to "make menuconfig" of ESP32 IDF and activate MODEM Sleep.
It will put on sleep the communication modules and wake them up every (100ms - 1000ms) to maintain the necessary comunications.
Generaly it reduces the current consumption to less than 30 mA.

artin-phares, 2019-05-27:
I've easily switched CPU Frequency from 240 MHz to 80 MHz in Arduino Board Configuration (using ESP32 Dev Module board type, which works for DOIT DevKit v1).

Any idea how to shut down second core?

lbernstone, 2019-05-27:
https://github.com/lbernstone/arduino-esp32-solo

artin-phares, 2019-05-27:
@lbernstone 

I see 240MHz frequency in sdkconfig.h
```
#define CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ 240
```

It means that if I'll use arduino-esp32-solo lib files and set 80MHz in Arduino Board Config, resulting core frequency will still be 80 or 240MHz?

lbernstone, 2019-05-27:
This does not affect the cpu freq.  That option calls the newish setCpuFrequencyMhz function.  You can always check it with getCpuFrequencyMhz.

closed 2019-08-14 by stale[bot]