espressif/arduino-esp32 #1106: improvement implementation on WiFi.begin(ssid,pw); with string parameters

DasVoelkel, 2018-02-13:
NO ISSUES!
Just some improvement idea.

Currently 
WiFi.begin(ssid,pw); 
doesn't support ssid and/or pw to be of the type string. Firstly i solved this with malloc, but really i think using malloc too often can be super dangerous. So for newer programmers it would be nice they didn't have to malloc ( and potentially forget the free(..) command).
I solved the whole thing with smart pointer that freed the space as soon as something else happened.
```
WiFi.begin(String ssid,String pw){
  std::unique_ptr<char[]> cptr1_(new char[network._ssid.length() + 1 ]);
  network._ssid.toCharArray(cptr1_.get(), (network._ssid.length() + 1 ));

  std::unique_ptr<char[]> cptr2_(new char[network._password.length() + 1 ]);
  network._password.toCharArray(cptr2_.get(), (network._password.length() + 1 ));

  WiFi.begin(cptr1_.get(), cptr2_.get());
}
```

everslick, 2018-02-15:
This is a VERY big general API annoyance. IMHO all functions that take a const char * should take const String & instead by default. and the passing by value instead of const references is a cardinal sin anyway, found in many places in many libs and also in the core I think.

lonerzzz, 2018-02-15:
We should have both those API calls taking String and those taking a const char * given that this API is intended to be as compatible with Arduino as possible and that we want to take advantage of more advanced language constructs.

everslick, 2018-02-15:
functions taking ```const String &``` are happy to consume ```const char *``` e.g.

```
void foo(const String &bar) {
}

foo("test");
foo(String("test"));
```

lonerzzz, 2018-02-15:
Learned something new. Hadn't realized there was an operator making that happen.

JohnsonUTPB, 2018-03-14:
Wow, you guys are awesome... I have been trying to build WiFi.begin params from values stored on a sd card... After many hrs without progress; google has led me here where I now know the shortcomings of the ESP32 API as well as the work around in which I was trying to avoid.
Thanks.....

DasVoelkel, 2018-03-15:
@JohnsonUTPB what exactly is your issue?

JohnsonUTPB, 2018-03-15:
Re: What is your issue:
Nothing more than y'all have already outlined; being, the ESP32 API is lacking a function/function overload in which WiFi.begin can be called with string params. 
What I do have:
I have a contribution in which I made a template project that:
   creates an array of chars (loop through action containing FileIO reads)
   allocate memspace sized for initializing WiFi.begin parameter
   binary copy of data from corresponding data array to ptr of allocated memspace
   and then use of these pointers allowed WiFi.begin(*,*) to connect using data from a sdcard during run time.

I am new to the github community and I am not trying to contribute to extensive forum clutter on this topic... but if y'all are interested; with a little guidance, my code might be a useful contribution.

closed 2019-03-04 by DasVoelkel