espressif/arduino-esp32 #1108: Is there any way to access the printed out ip address of smartconfig ??

DasVoelkel, 2018-02-13:
it seems after smartconfig is successful the log prints it out , but then it is discarded.
As far as i can see there is no way to get the ip adress of the sender atm, or am i missing something?

looked through the src code, only thing i found was where it was printed to the log.
In WiFiSTA.cpp
```
...
void WiFiSTAClass::_smartConfigCallback(uint32_t st, void* result) {
    smartconfig_status_t status = (smartconfig_status_t) st;
    log_d("Status: %s", sc_status_strings[st % 5]);
    if (status == SC_STATUS_GETTING_SSID_PSWD) {
        smartconfig_type_t * type = (smartconfig_type_t *)result;
        log_d("Type: %s", sc_type_strings[*type % 3]);
    } else if (status == SC_STATUS_LINK) {
        wifi_sta_config_t *sta_conf = reinterpret_cast<wifi_sta_config_t *>(result);
        log_d("SSID: %s", (char *)(sta_conf->ssid));
        sta_conf->bssid_set = 0;
        esp_wifi_set_config(WIFI_IF_STA, (wifi_config_t *)sta_conf);
        esp_wifi_connect();
        _smartConfigDone = true;
    } else if (status == SC_STATUS_LINK_OVER) {
        if(result){
            ip4_addr_t * ip = (ip4_addr_t *)result;
            log_d("Sender IP: " IPSTR, IP2STR(ip));
        }
        WiFi.stopSmartConfig();
    }
}
...
```

lbernstone, 2018-02-14:
I know nothing about smart config, but the normal way to get your IP is WiFi.localIP(), and the gateway is WiFi.gatewayIP().  A captive network IP (802.11x) is usually considered irrelevant after you get your info, and is typically discarded after connection even in a full network stack.

DasVoelkel, 2018-02-14:
I don't mean my IP address I mean the system who sent the address. 

When using the smart config app for the previous esp, the app can see the IP address of the freshly connected esp (when using smart config) 
And when displaying debug messages, the serial line prints out the IP address of the sender of the smartconfg package, but the this IP address is never accessible .

As i've said, the IP address gets passed through the smart config callback, but is never saved and therefore discarded, although it's not irrelevant information

closed 2018-04-21 by DasVoelkel