espressif/arduino-esp32 #1110: Does SPIFFS. mkdir() work?

tony-ozrelic, 2018-02-14:
Seems I can create/delete a file like "/foo.txt", but making a directory like "/foo_dir" does not work - it does not show up in the directory listing. 

Is mkdir/rmdir not working or am I not using it correctly?

char dirname[128];
sprintf(dirname, "/foo_dir");
if(!SPIFFS.exists(dirname))
{
   if(SPIFFS.mkdir(dirname)) printf("Can't make directory '%s'\n", dirname);
}

lrmoreno007, 2018-02-14:
You can't create a directory in SPIFFS, it doesn't exist. And the method SPIFFS.mkdir() doesn't exist.

In SPIFFS you can create a file named "/foo_dir/foo.txt", with this you simulate a directory with a file inside, but the directory doesn't exist, exist a file named "/foo_dir/foo.txt".

Read this for more information (talk about ESP8266 but is the same filesystem):
http://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#file-system-limitations

tony-ozrelic, 2018-02-14:
OK, thanks for the info!

closed 2018-02-14 by tony-ozrelic