espressif/arduino-esp32 #1111: ESP32 INTERRUPT ON BOTH FALLING AND RISING EDGE

usmanshahid001, 2018-02-14:
Hiii i am using proximity sensor with esp32 and it is calling ISR routine on both Falling and Rising Edge
i have verified it using nodemcu which is working properly i have tried all things like Falling Rising Change HIGH but its showing the same behaviour  kindly help me in this regard...

dazjd02, 2018-02-16:
I am using an interrupt on io34 and having the same issue; I have a 10K resistor pulling it high to try to stabilized it.    It seems to show the same behavior whether FALLING or RISING is used.  Sometimes interrupt occurs on the falling edge and other times it occurs on the rising edge and switches back every few seconds.

usmanshahid001, 2018-02-16:
actually its working on change no matter which mode is selected...

usmanshahid001, 2018-04-26:
i have checked the link but how to resolve this issue for arduino core

usmanshahid001, 2018-04-26:
i am having interrupts both on falling and rising edge .........how to resolve that problem related to gpio

yashfaith, 2018-04-29:
i also have the same problem. I am using an interrupt on pin 27, i tried using FALLING and RISING but it seems that the ISR got trigger on CHANGE. are there any solution to this problem?

usmanshahid001, 2018-06-04:
i have check using esp-idf gpio code that is working fine for positive edge trigger....but not working in arduino............

usmanshahid001, 2018-06-05:
waitnig for any response....................:(

usmanshahid001, 2018-06-06:
#include "driver/gpio.h"

#define GPIO_OUTPUT_IO_0    18
#define GPIO_OUTPUT_IO_1    19
#define GPIO_OUTPUT_PIN_SEL  ((1ULL<<GPIO_OUTPUT_IO_0) | (1ULL<<GPIO_OUTPUT_IO_1))
#define GPIO_INPUT_IO_0     4
#define GPIO_INPUT_IO_1     5
#define GPIO_INPUT_PIN_SEL  ((1ULL<<GPIO_INPUT_IO_0) | (1ULL<<GPIO_INPUT_IO_1))
#define ESP_INTR_FLAG_DEFAULT 0
extern "C" {
  #include "soc/gpio_struct.h"
}

static xQueueHandle gpio_evt_queue = NULL;

static void IRAM_ATTR gpio_isr_handler(void* arg)
{
    Serial.println("yes");
    
}

void setup()
{

  Serial.begin(115200);
    gpio_config_t pGPIOConfig
    {
      GPIO_INPUT_PIN_SEL,
      GPIO_MODE_INPUT,
      GPIO_PULLUP_ENABLE,
      GPIO_PULLDOWN_DISABLE,
      GPIO_INTR_POSEDGE,
    };
    pGPIOConfig.intr_type=GPIO_INTR_POSEDGE;
    gpio_config(&pGPIOConfig);
    gpio_set_intr_type(GPIO_NUM_5, GPIO_INTR_POSEDGE);
  gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
   
    gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(GPIO_NUM_4, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(GPIO_NUM_5, gpio_isr_handler, (void*) GPIO_INPUT_IO_1);
     gpio_isr_handler_remove(GPIO_NUM_4);
    //hook isr handler for specific gpio pin again
    gpio_isr_handler_add(GPIO_NUM_4, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
   
    int cnt = 0;
}
void loop()
{
  
}

usmanshahid001, 2018-06-06:
working in this way

musskopf, 2018-07-05:
I can confirm the issue is happening to me as well. My test code looks like:

```c
// User Buttons
#define USER_BT1      25

/*
 * Interrupt Definitions
 */
volatile uint8_t bt1Pressed = 0;

void bt1IntHandler()
{
  bt1Pressed++;
}

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Up");
  
  Serial.println(" - User Buttons");
  pinMode(USER_BT1, INPUT_PULLUP);

  // Some Delay for things to calm down
  delay(30);

  // Add Interrupts
  attachInterrupt(digitalPinToInterrupt(USER_BT1), bt1IntHandler, FALLING);
}

void loop() {
  // Print the number of interrupts every 3 second
  Serial.print("Number of Interrupts: ");
  Serial.println(bt1Pressed);
  delay(3000);
}
```

I also capture the button press on my scope, no noise in the signal.
[image]

The output on the Serial Print always increment by "2" every time I press the button. If I press and hold, it increments 1, but it'll increment 1 again once I release it.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
Starting Up
 - User Buttons
Number of Interrupts: 0
Number of Interrupts: 2 <- Quick Press and Release
Number of Interrupts: 4 <- Another Press and Release
Number of Interrupts: 5 <- Press and Hold
Number of Interrupts: 5 <- Still holding
Number of Interrupts: 6 <- Released
Number of Interrupts: 6
```

usmanshahid001, 2018-07-05:
yes it was the problem...however i had solved it earlier

usmanshahid001, 2018-07-05:
try this code

#include "driver/gpio.h"

#define GPIO_OUTPUT_IO_0 18
#define GPIO_OUTPUT_IO_1 19
#define GPIO_OUTPUT_PIN_SEL ((1ULL<<GPIO_OUTPUT_IO_0) | (1ULL<<GPIO_OUTPUT_IO_1))
#define GPIO_INPUT_IO_0 4
#define GPIO_INPUT_IO_1 5
#define GPIO_INPUT_PIN_SEL ((1ULL<<GPIO_INPUT_IO_0) | (1ULL<<GPIO_INPUT_IO_1))
#define ESP_INTR_FLAG_DEFAULT 0
extern "C" {
#include "soc/gpio_struct.h"
}

static xQueueHandle gpio_evt_queue = NULL;

static void IRAM_ATTR gpio_isr_handler(void* arg)
{
Serial.println("yes");

}

void setup()
{

Serial.begin(115200);
gpio_config_t pGPIOConfig
{
GPIO_INPUT_PIN_SEL,
GPIO_MODE_INPUT,
GPIO_PULLUP_ENABLE,
GPIO_PULLDOWN_DISABLE,
GPIO_INTR_POSEDGE,
};
pGPIOConfig.intr_type=GPIO_INTR_POSEDGE;
gpio_config(&pGPIOConfig);
gpio_set_intr_type(GPIO_NUM_5, GPIO_INTR_POSEDGE);
gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));

gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
//hook isr handler for specific gpio pin
gpio_isr_handler_add(GPIO_NUM_4, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
//hook isr handler for specific gpio pin
gpio_isr_handler_add(GPIO_NUM_5, gpio_isr_handler, (void*) GPIO_INPUT_IO_1);
 gpio_isr_handler_remove(GPIO_NUM_4);
//hook isr handler for specific gpio pin again
gpio_isr_handler_add(GPIO_NUM_4, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);

int cnt = 0;
}
void loop()
{

}

usmanshahid001, 2018-07-05:
attach interrupt function is not working as expected

musskopf, 2018-07-05:
Yes, I saw it, many thanks!

In any case, I was highlighting that this still a problem with the Arduino "attachInterrupt" function.

dazjd02, 2018-07-05:
I had a similar problem that turned out to be noise on the interrupt input line.  I think it was resolved with a .1uF capacitor.

william-ferguson-au, 2018-07-08:
Just to be clear this happens with the ESPIF libraries too. Not just the Arduino attachInterrupt() function.

Using the same board that @musskopf provided images for above and the following:

```
volatile uint8_t triggerPressed = 0;

static void IRAM_ATTR triggerIntHandler(void *args) {
    triggerPressed = 1;
}

void setup_hardware() {
    log("Initialising Pins");

    // input pins
    gpio_config_t gpioInputConfig;
    gpioInputConfig.pin_bit_mask = 1ULL << TRIGGER_BT;
    gpioInputConfig.mode = GPIO_MODE_INPUT;
    gpioInputConfig.pull_up_en = GPIO_PULLUP_ENABLE;
    gpioInputConfig.pull_down_en = GPIO_PULLDOWN_DISABLE;
    gpioInputConfig.intr_type = GPIO_INTR_POSEDGE;
    ESP_ERROR_CHECK(gpio_config(&gpioInputConfig));

    ESP_ERROR_CHECK(gpio_install_isr_service(ESP_INTR_FLAG_IRAM));
    gpio_isr_handler_add(TRIGGER_BT, triggerIntHandler, NULL);
}

void loop() {
    while (1) {
        if (triggerPressed) {
            triggerPressed = 0;
            Serial.println("Responding to trigger pressed");
        }

        ...
    }
}

```

The trigger fires when it is depressed and then once again when it is released.
But it should fire only for GPIO_INTR_POSEDGE.

As you can see from the screenshots from @musskopf the signal is clean.
SO what needs to be done to get it to fire on a single edge instead of both edges?

william-ferguson-au, 2018-07-08:
Note also, that it also fires twice if I use `GPIO_INT_NEGEDGE`

william-ferguson-au, 2018-07-08:
@stickbreaker when you say "shared" interrupt, what do you mean?

I do want to have interrupts for several pins, but I am getting this behaviour when I have only configured an interrupt for a single pin.

PepeTheFroggie, 2019-07-09:
Same problem here.

bradley-douglas, 2019-08-27:
same problem using ESP32 Wroom 32u only falling edge works for me , if i select rising edge it triggers on both rising and falling , no noise on input as i have 100nf there.

bradley-douglas, 2019-08-27:
On further testing what i found is weird but it solves this double trigger issue with ESP32
There must be no debounce capacitor on gpio line it seems that the slew rate of the edge of falling side of logic if slow creates a 2nd trigger. Also the logic must be the following, you must pull the gpio to gnd and have a pullup resistor. if you follow this principal the interrupt works fine for RISING and FALLING perfectly.

rafaelalvesferreira, 2019-09-12:
> On further testing what i found is weird but it solves this double trigger issue with ESP32
> There must be no debounce capacitor on gpio line it seems that the slew rate of the edge of falling side [...]

The info is perfect. You can use the internal pullup resistor using:
 -  pinMode(Pin_number, INPUT_PULLUP);
Be aware that GPIOs 34-39 are input-only *and* **do not have internal pull-up or pull-down circuitry**.

MacLeod-D, 2019-12-21:
My experience:
Pressing a button gives a lot of noise - pulses in down to some ns.
For that mostly you have to debounce a button press.
Try:
 - Use an output pin to produce pulses
- Connect this pin with the interrupt pin

I had never a problem with an interrupt using the ESP32.

Maybe you want to look at:
https://github.com/MacLeod-D/ESp32-Fast-external-IRQs

There you can find the measurements for external interrupts.
And a polling solution which is VERY fast.

cascabel2k2, 2020-01-23:
/*FLANCO DE SUBIDA EN PIN 21, CONMUTA EL ESTADO DEL LED EN PIN 22*/

#define pulsador  21 
#define led       22 
#define t_rebote  10   // ms

volatile int numberOfButtonInterrupts = 0;
volatile bool lastState;
volatile uint32_t debounceTimeout = 0;

uint32_t saveDebounceTimeout;
bool saveLastState;
int save;

portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR isr() {
  portENTER_CRITICAL_ISR(&mux); 
   numberOfButtonInterrupts++;
   lastState = digitalRead(pulsador);  
   debounceTimeout = xTaskGetTickCount(); 
  portEXIT_CRITICAL_ISR(&mux);
}

void setup() {
  pinMode (led, OUTPUT);
  pinMode(pulsador, INPUT_PULLUP);  
  attachInterrupt(digitalPinToInterrupt(pulsador), isr, **RISING**);
  digitalWrite(led, LOW); 
}

void loop() {
    portENTER_CRITICAL_ISR(&mux); 
     save = numberOfButtonInterrupts;
     saveDebounceTimeout = debounceTimeout;
     saveLastState  = lastState;
    portEXIT_CRITICAL_ISR(&mux); 

    bool currentState = digitalRead(pulsador); 

    if(currentState != saveLastState)
      saveDebounceTimeout = millis();
    if( (millis() - saveDebounceTimeout) > t_rebote && (save != 0) ) {
      **if(currentState)**              
        digitalWrite(led, !digitalRead(led));
      else { }
      portENTER_CRITICAL_ISR(&mux);
       numberOfButtonInterrupts = 0;
      portEXIT_CRITICAL_ISR(&mux);
    }
}

cascabel2k2, 2020-01-23:
/*FLANCO DE SUBIDA (RISING) EN PIN 21, CONMUTA EL ESTADO DEL LED EN PIN 22*/

#define pulsador  21 
#define led       22 
#define t_rebote  10   // ms

volatile int numberOfButtonInterrupts = 0;
volatile bool lastState;
volatile uint32_t debounceTimeout = 0;

uint32_t saveDebounceTimeout;
bool saveLastState;
int save;

portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR isr() {
  portENTER_CRITICAL_ISR(&mux); 
   numberOfButtonInterrupts++;
   lastState = digitalRead(pulsador);  
   debounceTimeout = xTaskGetTickCount(); 
  portEXIT_CRITICAL_ISR(&mux);
}

void setup() {
  pinMode (led, OUTPUT);
  pinMode(pulsador, INPUT_PULLUP);  
  attachInterrupt(digitalPinToInterrupt(pulsador), isr, RISING);
  digitalWrite(led, LOW); 
}

void loop() {
    portENTER_CRITICAL_ISR(&mux); 
     save = numberOfButtonInterrupts;
     saveDebounceTimeout = debounceTimeout;
     saveLastState  = lastState;
    portEXIT_CRITICAL_ISR(&mux); 

    bool currentState = digitalRead(pulsador); 

    if(currentState != saveLastState)
      saveDebounceTimeout = millis();
    if( (millis() - saveDebounceTimeout) > t_rebote && (save != 0) ) {
      **if(currentState)**
        digitalWrite(led, !digitalRead(led));
      else { }
      portENTER_CRITICAL_ISR(&mux);
       numberOfButtonInterrupts = 0;
      portEXIT_CRITICAL_ISR(&mux);
    }
}

cascabel2k2, 2020-01-23:
NOTAS.-
/*Flanco de subida en pin 21 (RISING), conmuta el estado del led en pin 22*/
28.   attachInterrupt(digitalPinToInterrupt(pulsador), isr, RISING);
44.       if(currentState == true)

 /*Flanco de bajada en pin 21 (FALLING), conmuta el estado del led en pin 22*/ 
28.   attachInterrupt(digitalPinToInterrupt(pulsador), isr, FALLING);
44.       if(currentState == false)

 /*Ambos flancos en pin 21 (CHANGE), conmuta el estado del led en pin 22*/ 
28.   attachInterrupt(digitalPinToInterrupt(pulsador), isr, CHANGE);
44.       if((currentState==false) || (currentState==true))

[ESP32.txt](https://github.com/espressif/arduino-esp32/files/4105455/ESP32.txt)

cascabel2k2, 2020-01-23:
NOTAS.-
/*Flanco de subida en pin 21 (RISING), conmuta el estado del led en pin 22*/
28.   attachInterrupt(digitalPinToInterrupt(pulsador), isr, RISING);
44.       if(currentState == true)

 /*Flanco de bajada en pin 21 (FALLING), conmuta el estado del led en pin 22*/ 
28.   attachInterrupt(digitalPinToInterrupt(pulsador), isr, FALLING);
44.       if(currentState == false)

 /*Ambos flancos en pin 21 (CHANGE), conmuta el estado del led en pin 22*/ 
28.   attachInterrupt(digitalPinToInterrupt(pulsador), isr, CHANGE);
44.       if((currentState==false) || (currentState==true))

[publicarIntEsp32.zip](https://github.com/espressif/arduino-esp32/files/4105527/publicarIntEsp32.zip)

cascabel2k2, 2020-01-23:
> Hiii i am using proximity sensor with esp32 and it is calling ISR routine on both Falling and Rising Edge
> i have verified it using nodemcu which is working properly i have tried all things like Falling Rising Change HIGH but its showing the same behaviour kindly help me in this regard...

NOTAS.-
/Flanco de subida en pin 21 (RISING), conmuta el estado del led en pin 22/
28. attachInterrupt(digitalPinToInterrupt(pulsador), isr, RISING);
44. if(currentState == true)

/Flanco de bajada en pin 21 (FALLING), conmuta el estado del led en pin 22/
28. attachInterrupt(digitalPinToInterrupt(pulsador), isr, FALLING);
44. if(currentState == false)

/Ambos flancos en pin 21 (CHANGE), conmuta el estado del led en pin 22/
28. attachInterrupt(digitalPinToInterrupt(pulsador), isr, CHANGE);
44. if((currentState==false) || (currentState==true))

[publicarIntEsp32.zip](https://github.com/espressif/arduino-esp32/files/4105578/publicarIntEsp32.zip)

DougArmstrong, 2020-06-20:
Have there been any updates on this issue?  I have studied this issue closely and it appears that when in rising or falling mode the ISRs trigger randomly in the other mode.  (This is not a key bounce issue as have been discussed above.  This is a test square wave fed directly to the MCU) 

In the image below, the yellow square wave is fed into PIN 17 on the ESP32.  Each time the ISR fires, it toggles the output pin shown in purple.  To the left you can see that the ISR fired on the rising edge of the input, however in the following two pulses it fired correctly.  

The ISR setup and  code follows:
------------
#define RPM_PIN			17		
	pinMode(Pin, INPUT_PULLUP);			// Pull Up the pin and make it an input
	pinMode(Light, OUTPUT);

	// Be sure the ISR is fully initialized by this point.
	attachInterrupt(digitalPinToInterrupt(Pin), RPM_ISR, FALLING); 
--------------------

void IRAM_ATTR RPM_ISR()
{
	static  uint64_t StartValue = 0;				// First interrupt value
	BaseType_t xHigherPriorityTaskWoken;
	uint32_t PeriodCount;
	uint64_t TempVal;

	xHigherPriorityTaskWoken = pdFALSE;

	TempVal = timerRead(RPM_Timer);			// value of timer at interrupt
	PeriodCount = (TempVal - StartValue);		// period count between rising edges in 
	
	
// ESP 32 appears to have a bug where sometimes different edges fire regardless of rising/falling
// Some test code to see the timing on the scope
digitalWrite(Light, !digitalRead(Light));

	xQueueSendToBackFromISR(RPM_Queue, &PeriodCount, &xHigherPriorityTaskWoken);
										// puts latest reading as start for next calculation
        StartValue = TempVal;
	
} // RPM_ISR

[image]

JimDrewGH, 2020-07-03:
I have seen this same comment from several different people, and the only thing in common that everyone has had that I noticed was that they are all using INPUT_PULLUP.  I wonder if the ESP32 disconnects the pin internally during the some part of the interrupt process and reconnecting it sometimes retriggers the interrupt.

DougArmstrong, 2020-07-03:
ESP32 does not require clearing the interrupt pending flag (or at least if it does this is done in background).  I will try an external pull up to see if that helps.  Some posts are about key-bounce which is clearly not what is happening here.  It does seem like a noise problem.  A previous post suggested that using the IDF instead of Arduino FW creates code that works correctly.  I try to stick with Arduino since I write for so many different CPUs.  Thanks for the feedback Jim, I'll run some tests and let you know what I find out.

BjarneBitscrambler, 2020-07-15:
@DougArmstrong - have you made any discoveries regarding internal vs external pullups?  

And, in your scope shot - is the yellow trace being probed right at the ESP32 input pin, or is there some additional wire length or passive components between the scope probe and the GPIO pin?  The scope shot certainly does look like a clean and fast-edged signal.

gehowi, 2020-07-16:
Random interrupts happen on ‘input only’ pins 34-39 that are pulled up or stay high. It has nothing to do with EMI.  
If the GPIO pin is normally low,  a positive pulse will generate a normal interrupt in any mode.  
If the pin stays high, interrupts will fire continuously, no matter which mode is used.
Use Input/Output pins for interrupts. Test with GPIO 4.

zfields, 2020-12-12:
Similar to the OP, I had a clean signal, but experienced `CHANGE` behavior when I had set the interrupt for `FALLING`.

My interrupts were originally configured as follows:

```c++
  pinMode(21, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(21), isr_1, FALLING);
  pinMode(14, INPUT);
  attachInterrupt(digitalPinToInterrupt(14), isr_2, RISING);
```

After reading @stickbreaker 's comment...
> "all pin interrupt sources use one interrupt(shared)"

I configured the interrupts to fire on the same edge, and I was able to resolve my issue.

```c++
  pinMode(21, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(21), isr_1, RISING);
  pinMode(14, INPUT);
  attachInterrupt(digitalPinToInterrupt(14), isr_2, RISING);
```
Minimally, it appears to be a bug in the Arduino Board Package implementation.

> Board: Adafruit Huzzah32 ESP32 Feather
> Arduino IDE: v1.8.13
> Additional Boards Manager URL: `https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json`
> Boards Package: esp32 (Espressif Systems) v1.0.4

ALEBORTOT, 2021-06-19:
ESP32 is really a "fake cpu".
Please see 
>> 3.14. The ESP32 GPIO peripheral may not trigger interrupts correctly at  
https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf

I use a PIC16C54 at 1997 and the interrupts works fine.

Now, at 2021, It's unbelievable that a CPU has serious problems interrupting a GPIO. Its the fantastic ESP32.

And there is no easy workaround .

LMBernardo, 2021-09-05:
This issue is still occurring for me, specifically on GPIO 35 (which is input only and has no pull up / pull down). I'm setting the config for all GPIO 0 and GPIO 35 at the same time, so they are identical, but GPIO0 behaves as expected and reacts on falling edge only while GPIO 35 will randomly interrupt on rising edges as well. Not sure if this is a hardware thing or a software thing or what. I did consider adding a pull up resistor to the pin, but it seems that GPIO_INTR_ANYEDGE works as expected on both GPIO0 and GPIO35.

This being the case, I was able to work around the issue by just triggering on all edges and implementing the falling edge detector in software. Just keep track of the previous input state - when an edge event happens, check the previous state for that input. If the previous state was high and the new state is low, then you have a falling edge.

zfields, 2021-09-07:
@LMBernardo I also had difficulties when trying to use RISING and FALLING edges simultaneously. Once I set my logic to use only RISING or FALLING, then the interrupts behaved as expected.

wanggaoteng, 2022-04-30:
> I can confirm the issue is happening to me as well. My test code looks like:
>
> ```c
> // User Buttons
> #define USER_BT1      25
>
> /*
>  * Interrupt Definitions
>  */
> volatile uint8_t bt1Pressed = 0;
>
> void [...]

Hi, today is 2022.04.30, my test code is very similar to yours, and the problem is still exists.

florentbr, 2023-01-15:
I faced the same issue with a signal from an optoisolator. The unexpected interrupts happened due to a significant rise/fall time. The ESP32 does not work well with curved edges. It needs **straight edges** to avoid the unexpected interrupts.

There's multiple ways to handle/avoid this issue:
* de-bounce with a timer in the interrupt
* read the input in the interrupt and skip if the level is not right
* reduce the rise/fall time with a transistor on the input to quickly jump into saturation.
* clean the input with a trigger buffer

timr49, 2023-06-21:
My experience is similar. I am using a water flow meter that pulses high-low-high with a frequency related to water flow rate.  With an ESP8266 and FALLING mode interrupt, it worked as expected. I replaced the ESP8266 with an ESP32. On the test bench with short wires to the flow meter, it behaved similarly (though it may have been double-counting pulses without me realising). But in the field, with longer wires between CPU and pulse meter, it reported an impossibly high number of pulses. I implemented software filtering that looked at the pin level to distinguish real falling edges (2nd option per @florentbr  above) and also included counting  of the 4 states as found by the ISR (previous state-current state = high-high, low-low, high-low and low-high).  As well as there being the same number of rising as falling interrupts, there was a very large number of the "high-high" case.

Probably nothing new here compared to discussion above but it seemed interesting that making the wires longer, and presumably the skew due to resultant capacitance, had a measureable effect on the number of "false" interrupts.

If there is a next hardware revision, it will likely have a Schmitt trigger on the input pin.

matei-miron, 2023-10-19:
Hi, I am experience the same problem.
Checked the input with the scope. It is really clean. Had extra caps and resistors added to debounce the signal "just in case".
Using 5.1.1

```
#include "driver/gpio.h"
#include "esp_log.h"
#include "soc/gpio_reg.h"           // For GPIO Registers. GPIO_OUT_REG (31..0) . GPIO_OUT1_REG (32..39) . REG_READ(GPIO_IN1_REG)
#include "freertos/FreeRTOS.h"
#include "freertos/queue.h"
#include "freertos/task.h"

#define crankPin 32
#define wheelPin 33
#define bothPins ((1ULL << crankPin) | (1ULL << wheelPin))

uint32_t wheelRevs = 0;
uint16_t wheelTime = 0;
uint16_t crankRevs = 0;
uint16_t crankTime = 0;
uint32_t wheelRevsOld = 0;
uint16_t wheelTimeOld = 0;
uint16_t crankRevsOld = 0;
uint16_t crankTimeOld = 0;

uint32_t interruptCrankOldTime = 0;
uint32_t interruptWheelOldTime = 0;

static const char *INTTAG = "Register Output:";

static void IRAM_ATTR gpio_isr_handler(void* arg) {
    uint32_t gpioInValue = REG_READ(GPIO_IN1_REG); // Read the whole GPIO input data register
    uint32_t timestamp = xTaskGetTickCount();

    uint32_t data[2];
    data[0] = gpioInValue;
    data[1] = timestamp;

    // Pass the GPIO data and timestamp to the queue
    xQueueSendFromISR(gpio_evt_queue, data, NULL);
}

void updateRecords(void* arg) {
    uint32_t receivedData[2];

    for (;;) {
        if (xQueueReceive(gpio_evt_queue, receivedData, portMAX_DELAY)) {
            //(((receivedData[0] >> 1) & 1) << 1) | (receivedData[0] >> 0) & 1
            uint32_t interruptTime = receivedData[1] * 10;
            uint32_t deltaTimeCrank = interruptTime - interruptCrankOldTime;
            uint32_t deltaTimeWheel = interruptTime - interruptWheelOldTime;
            
            if ((receivedData[0] >> (crankPin % 32)) & 1) {
                // 250 ms are 4 Hz or 240 rpm.
                if (deltaTimeCrank >= 250) {
                    crankRevs++;
                    ESP_LOGI(INTTAG, "Crank rotations: %u, %lu. Delta: %lu", crankRevs, interruptTime, deltaTimeCrank);
                }
                interruptCrankOldTime = interruptTime;
            }

            if ((receivedData[0] >> (wheelPin % 32)) & 1) {
                // 100 ms are 10 Hz or 600 rpm. About 72 km/h for 29 inch (700C) wheels 
                if (deltaTimeWheel >= 100) {
                    wheelRevs++;
                    ESP_LOGI(INTTAG, "Wheel rotations: %lu, %lu. Delta: %lu", wheelRevs, interruptTime, deltaTimeWheel);
                }
                interruptWheelOldTime = interruptTime;
            }
        }
    }
}

void app_main() {
    esp_log_level_set("*", ESP_LOG_INFO);

    gpio_config_t pinConfig = {};
    pinConfig.intr_type = GPIO_INTR_POSEDGE;
    pinConfig.mode = GPIO_MODE_INPUT;
    pinConfig.pin_bit_mask = bothPins;
    pinConfig.pull_down_en = GPIO_PULLDOWN_ENABLE;
    pinConfig.pull_up_en = GPIO_PULLUP_DISABLE;

    ret = gpio_config(&pinConfig);

    // Create a queue to handle GPIO events from ISR
    gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t[2]));

    // Start the GPIO task
    xTaskCreate(updateRecords, "update_records", 2048, NULL, 10, NULL);

    // Add the ISR
    gpio_install_isr_service(ESP_INTR_FLAG_LOWMED);
    gpio_isr_handler_add(crankPin, gpio_isr_handler, (void*) NULL);
    gpio_isr_handler_add(wheelPin, gpio_isr_handler, (void*) NULL);

    while (1) {
        // Add a suitable delay to control the timing of your application
        vTaskDelay(pdMS_TO_TICKS(1000)); // Example: 1-second delay
    }
}
```

This produces an interrupt as expected if the signal is applied for a short time.
But it produces an interrupt on the raising edge and on the falling edge if the signal is high for a "long enough time". Let's say 1 second.
It also panics the core if I set pinConfig.intr_type = GPIO_INTR_HIGH_LEVEL;

MarcinSzymanek, 2024-05-14:
> Hi, I am experience the same problem. Checked the input with the scope. It is really clean. Had extra caps and resistors added to debounce the signal "just in case". Using 5.1.1
> 
> ```
> #include "driver/gpio.h"
> #include "esp_log.h"
> #include "soc/gpio_reg.h"           // For GPIO Registers. GPIO_OUT_REG (31..0) . GPIO_OUT1_REG (32..39) . REG_READ(GPIO_IN1_REG)
> #include "freertos/FreeRTOS.h"
> #include "freertos/queue.h"
> #include "freertos/task.h"
> 
> #define crankPin 32
> #define wheelPin 33
> #define bothPins ((1ULL << crankPin) | (1ULL << wheelPin))
> 
> uint32_t wheelRevs = 0;
> uint16_t wheelTime = 0;
> uint16_t crankRevs = 0;
> uint16_t crankTime = 0;
> uint32_t wheelRevsOld = 0;
> uint16_t wheelTimeOld = 0;
> uint16_t crankRevsOld = 0;
> uint16_t crankTimeOld = 0;
> 
> uint32_t interruptCrankOldTime = 0;
> uint32_t interruptWheelOldTime = 0;
> 
> static const char *INTTAG = "Register Output:";
> 
> static void IRAM_ATTR gpio_isr_handler(void* arg) {
>     uint32_t gpioInValue = REG_READ(GPIO_IN1_REG); // Read the whole GPIO input data register
>     uint32_t timestamp = xTaskGetTickCount();
> 
>     uint32_t data[2];
>     data[0] = gpioInValue;
>     data[1] = timestamp;
> 
>     // Pass the GPIO data and timestamp to the queue
>     xQueueSendFromISR(gpio_evt_queue, data, NULL);
> }
> 
> void updateRecords(void* arg) {
>     uint32_t receivedData[2];
> 
>     for (;;) {
>         if (xQueueReceive(gpio_evt_queue, receivedData, portMAX_DELAY)) {
>             //(((receivedData[0] >> 1) & 1) << 1) | (receivedData[0] >> 0) & 1
>             uint32_t interruptTime = receivedData[1] * 10;
>             uint32_t deltaTimeCrank = interruptTime - interruptCrankOldTime;
>             uint32_t deltaTimeWheel = interruptTime - interruptWheelOldTime;
>             
>             if ((receivedData[0] >> (crankPin % 32)) & 1) {
>                 // 250 ms are 4 Hz or 240 rpm.
>                 if (deltaTimeCrank >= 250) {
>                     crankRevs++;
>                     ESP_LOGI(INTTAG, "Crank rotations: %u, %lu. Delta: %lu", crankRevs, interruptTime, deltaTimeCrank);
>                 }
>                 interruptCrankOldTime = interruptTime;
>             }
> 
>             if ((receivedData[0] >> (wheelPin % 32)) & 1) {
>                 // 100 ms are 10 Hz or 600 rpm. About 72 km/h for 29 inch (700C) wheels 
>                 if (deltaTimeWheel >= 100) {
>                     wheelRevs++;
>                     ESP_LOGI(INTTAG, "Wheel rotations: %lu, %lu. Delta: %lu", wheelRevs, interruptTime, deltaTimeWheel);
>                 }
>                 interruptWheelOldTime = interruptTime;
>             }
>         }
>     }
> }
> 
> void app_main() {
>     esp_log_level_set("*", ESP_LOG_INFO);
> 
>     gpio_config_t pinConfig = {};
>     pinConfig.intr_type = GPIO_INTR_POSEDGE;
>     pinConfig.mode = GPIO_MODE_INPUT;
>     pinConfig.pin_bit_mask = bothPins;
>     pinConfig.pull_down_en = GPIO_PULLDOWN_ENABLE;
>     pinConfig.pull_up_en = GPIO_PULLUP_DISABLE;
> 
>     gpio_config(&pinConfig);
> 
>     // Create a queue to handle GPIO events from ISR
>     gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t[2]));
> 
>     // Start the GPIO task
>     xTaskCreate(updateRecords, "update_records", 2048, NULL, 10, NULL);
> 
>     // Add the ISR
>     gpio_install_isr_service(ESP_INTR_FLAG_LOWMED);
>     gpio_isr_handler_add(crankPin, gpio_isr_handler, (void*) NULL);
>     gpio_isr_handler_add(wheelPin, gpio_isr_handler, (void*) NULL);
> 
>     while (1) {
>         // Add a suitable delay to control the timing of your application
>         vTaskDelay(pdMS_TO_TICKS(1000)); // Example: 1-second delay
>     }
> }
> ```
> 
> This produces an interrupt as expected if the signal is applied for a short time. But it produces an interrupt on the raising edge and on the falling edge if the signal is high for a "long enough time". Let's say 1 second. It also panics the core if I set pinConfig.intr_type = GPIO_INTR_HIGH_LEVEL;

The core panicking can happen because while the signal stays high, the interrupt keeps triggering?

MikeJohnsEng, 2024-06-22:
There is no Schmitt triggering on ESP32 inputs, interrupts on this MPU need a quick clean transition. 
I found with slow rise time transitions that interrupts are very unreliable, and even multiple interrupts can be triggered from one event . 

It's best to use an external Schmitt trigger buffer. 

The STM32 series have Schmitt on all inputs which make them more suitable for a lot of this sort of work. 

> I faced the same issue with a signal from an optoisolator. The unexpected interrupts happened due to a significant rise/fall time. The ESP32 does not work well with curved edges. It needs **straight [...]

Yes
There is no Schmitt triggering on ESP32 inputs, interrupts on this MPU need a quick clean transition. 
I found with slow rise time transitions that interrupts are very unreliable, and even multiple interrupts can be triggered from one event . 

It's best to use an external Schmitt trigger buffer, that cleans it all up. 

The STM32 series of MPU's have Schmitt on all inputs which make them a lot more robust, so I'll often interface less than ideal inputs with a Black-Pill and then send the data on to the esp32, works out cheaper than using external Schmitt buffers !

uygurrr, 2024-07-06:
falling edge and rising edge ramp should be vertical. there should be no capacitor. solution code:
static void kesmeIR()
{  
  if (digitalRead(ir_pin) == LOW) {
    say++;
    peryot = micros() - per ;
    per = micros();
  } 
}

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

mat2k, 2025-08-17:
This issue still exists, I have experienced it on different projects. Calling digitalRead kind of sidesteps the problem, but it adds unnecessary code to the interrupt.

StephanReti, 2026-01-07:
I am currently also experiencing this issue on a sideproject where I have a button connected through an optocoupler.

> This issue still exists, I have experienced it on different projects. Calling digitalRead kind of sidesteps the problem, but it adds unnecessary code to the interrupt.

In my setup (falling ISR) this approach fails about every 20th button press, so be careful.
I assume that the pin state sometimes hasn't fully settled when the ISR triggers and thus `digitalRead()` might return the "old" state. (The threshold for triggering the ISR hardware and for causing I/O register to return `high` could also be slightly different)

<details>

<summary>The test code: Immediate digitalRead()</summary>

```cpp
#include <Arduino.h>

constexpr int externalPin = 5;
constexpr int debugPinOne = 21;
constexpr int debugPinTwo = 45;

volatile bool externalPinTriggered = false;
volatile uint8_t coolDownCounter = 0;

void ARDUINO_ISR_ATTR PinISR() {
  bool state = digitalRead(5);
  digitalWrite(debugPinOne, HIGH);

  if (!state && coolDownCounter == 0) {
    externalPinTriggered = true;
    coolDownCounter = 10;

    digitalWrite(debugPinTwo, HIGH);    // Should theoretically only become high if it actually was a falling edge
    
  }
  delayMicroseconds(1); // Yes this line is a NoGo, but it makes the pulses wider and thus easier to catch with a scope
  digitalWrite(debugPinOne, LOW);
  digitalWrite(debugPinTwo, LOW);
}

void setup() {
  pinMode(debugPinOne, OUTPUT);
  pinMode(debugPinTwo, OUTPUT);
  pinMode(externalPin, INPUT_PULLUP);
  
  delay(100);                                          // Give pullup time to settle
  
  attachInterrupt(digitalPinToInterrupt(externalPin), PinISR, FALLING);

}

void loop() {
  coolDownCounter += (coolDownCounter > 0) ? -1 : 0;
  delay(1);
}
```
</details>

Here is a screenshot of such a "1 in 20 times" event:

[image]

$\color{#B8B800}{\textsf{Channel 1}}$: The actual button (before the optocoupler)
$\color{#03B800}{\textsf{Channel 2}}$: The ESP input pin
$\color{#88077D}{\textsf{Channel 3}}$: Debug pin 1
$\color{#1F5AFF}{\textsf{Channel 4}}$: Debug pin 2

There the ESP incorrectly triggered on a rising signal and the digitalRead() approach failed.

My current solution is to set a flag in the ISR and perform the digitalRead() in the main loop after a small delay.

<details>

<summary>Current solution: Delayed digitalRead()</summary>

```cpp
#include <Arduino.h>

constexpr int externalPin = 5;
constexpr int debugPinOne = 21;
constexpr int debugPinTwo = 45;

volatile bool externalPinTriggered = false;
volatile uint8_t coolDownCounter = 0;

void ARDUINO_ISR_ATTR PinISR() {
  bool state = digitalRead(5);
  digitalWrite(debugPinOne, HIGH);

  if (!state && coolDownCounter == 0) {
    externalPinTriggered = true;
    coolDownCounter = 10;
  }
}

void setup() {
  pinMode(debugPinOne, OUTPUT);
  pinMode(debugPinTwo, OUTPUT);
  pinMode(externalPin, INPUT_PULLUP);
  
  delay(100);                                          // Give pullup time to settle
  
  attachInterrupt(digitalPinToInterrupt(externalPin), PinISR, FALLING);

}

void loop() {
  coolDownCounter += (coolDownCounter > 0) ? -1 : 0;

  if (externalPinTriggered) {
    delayMicroseconds(10);// Allow the pin to settle
    if (digitalRead(externalPin) == LOW) {
      // The pin is low, so it actually was a falling edge
      digitalWrite(debugPinTwo, HIGH);
    } else {
      // False trigger
    }
    externalPinTriggered = false;
    digitalWrite(debugPinOne, LOW);
    digitalWrite(debugPinTwo, LOW);
  }
  delay(1);
}
```

</details>