espressif/arduino-esp32 #1113: Deep-Sleep needs WiFi switched OFF for low current consumption

rudi48, 2018-02-14:
### Hardware:
Board:	WEMOS LoLin32					
Core Installation/update date:  Jun  8 2016
IDE name:	Arduino 1.8.5 MacOS			
Flash Frequency:	80Mhz
Upload Speed:	115200

### Description:
Unfortunately it is not explained that for **deep-sleep** with about 150 uA mentioned **current consumption** you **need to switch OFF WiFi befor**e, if it was in use.
If you do not do that the **current consumption** is about **1600 uA - about 10 time what has been told.**

All **documentation** I found to that topic **did not tell about current consumption**.

**I propose to mention it on all documentation, where the 150 uA are mentioned.**

You can **save** in the case of an **internal** GPIO pull-down input **another 50 uA**, 
with an **external 120 KOhm resistor.**

Regards, Rudolf

### Sketch:
```cpp
#include <esp_wifi.h>
...
  esp_wifi_stop();  
  esp_deep_sleep_start(); // 0.25 mA sleep current (battery)
```

### References:
esp32_technical_reference_manual_en.pdf version 2.9 - does not tell about wifi_stop

esp32_datasheet_en.pdf version 2.1 - does not tell about wifi_stop
Table 4: Power Consumption by Power Modes
Deep-sleep The ULP co-processor is powered on. 150 μA

esp32_eco_and_workarounds_for_bugs_in_esp32_en.pdf verion 1.4 - does not tell about wifi_stop

esp32_hardware_design_guidelines_en.pdf version 2.1 - does not tell about wifi_stop

esp-wroom-32d_esp32-wroom-32u_datasheet_en.pdf version 1.0 - does not tell about wifi_stop
Table 7: Power Consumption by Power Modes
Deep-sleep The ULP co-processor is powered on. 150 μA

https://dl.espressif.com/doc/esp-idf/latest/
Search for "deep" shows 11 Search Results
Search for "wifi-stop" shows 1 Search Result - does not tell about power saving

https://github.com/espressif/esp-idf/blob/master/docs/api-reference/system/sleep_modes.rst
n deep sleep mode, wireless peripherals are powered down. Before entering sleep mode, applications must disable WiFi and BT using appropriate calls ( esp_bluedroid_disable, esp_bt_controller_disable, esp_wifi_stop).
Note: but it does not tell about "current consumption".

https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_sleep.h
 * esp_deep_sleep does not shut down WiFi, BT, and higher level protocol
 * connections gracefully.
 * Make sure relevant WiFi and BT stack functions are called to close any
 * connections and deinitialize the peripherals. These include:
 *     - esp_bluedroid_disable
 *     - esp_bt_controller_disable
 *     - esp_wifi_stop
Note: but it does not tell about "current consumption".

beegee-tokyo, 2018-02-15:
Added to the [Wiki](http://desire.giesecke.tk/index.php/2018/02/15/deep-sleep-needs-wifi-switched-off-for-low-current-consumption/) with credits to you.
Thank you for sharing your findings.

mikey60, 2019-02-12:
@brianaagaard, I had the exact same issue with esp_sleep_enable_ext0_wakeup after having used the WiFi.  I am using version 1.0.1 of arduino-esp32.  Even after executing WiFi.disconnect() and WiFi.mode(WIFI_OFF) before executing esp_deep_sleep_start() I saw ~1.3 ma in the deep sleep mode.  Thanks to you post I changed to esp_sleep_enable_ext1_wakeup and now my deep sleep current is  ~8 uA.

arnaudmz, 2019-05-04:
Same here.
Even with `esp_sleep_enable_ext1_wakeup`, you can go back to ~1.3 mA if you enable 
 the PERIPH RTC (with `ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON));`). I had to add an external Resistor as a pullup.

Stasikiii, 2019-05-20:
```c
#include "driver/adc.h"
adc_power_off();

rudi48, 2019-05-20:
@Stasikiii 
Tank you for the hint. Did you measure the power consumtion before and at sleep?

Stasikiii, 2019-05-21:
@rudi48 
Yep :)
Without adc_power_off - 1,3mA
With - 7uA (RTC SLOW/FAST MEM IS ON)
I use TTGO MICRO 32 module (ESP PICO D4).

closed 2019-08-14 by stale[bot]

our30K, 2020-05-27:
using Arduino's wifi turn on methods leads to ADC somehow start consuming power. 
so when turn off the wifi, it should turn off the ADC too. to save power.

our30K, 2020-05-27:
this is how I turn off wifi after wifi scan.  with or without adc_power_off(), there is about 1+mA power consumption difference.  I believe somehow when just use ext1 as wakeup source to sleep, ADC will be turnoff automatically when sleep, but if use ext0 as wakeup source, ADC don't get automatically turned off when enter deep sleep

  
WiFi.mode(WIFI_OFF);
  esp_wifi_stop();
  adc_power_off();

ligantx, 2021-03-23:
i run numerous tests (like 40 tests with multimeter) with the following lines:
```
//for arduino core
#include "esp_wifi.h"
#include "driver/adc.h"

//do many stuff
WiFi.mode(WIFI_MODE_NULL);
//comment one or both following lines and measure deep sleep current
esp_wifi_stop(); //you must do esp_wifi_start() the next time you'll need wifi or esp32 will crash
adc_power_off();
```
Summarized, I found that esp_wifi_stop() didnt make any difference, but adc_power_off, made deep_sleep current ~1-2mA less when wifi was used before the deep sleep.
In all scenarios, if you dont use wifi there wasnt any difference. 
The only strange thing is that when i run a no wifi task after code upload and then deep sleep i had much lower current, compared to if i run a wifi task and then deep sleep, and after that a no wifi task and deep sleep..

e-tinkers, 2021-05-22:
`adc_power_off()` is [deprecated](https://github.com/espressif/esp-idf/blob/1067b28707e527f177752741e3aa08b5dc64a4d7/components/driver/include/driver/adc_common.h#L155-L162). Use `adc_power_release()` instead.

Gabriel-Gardin, 2023-03-20:
adc_power_release() does not decrease the power usage during deep sleep.

mars000, 2023-04-13:
> You can **save** in the case of an **internal** GPIO pull-down input **another 50 uA**,
> with an **external 120 KOhm resistor.**

hello - very useful. Can you explain your statement "you can save in the case of internal GPIO pull-down input another 50uA". Are you suggested using pinMode(xx, INPUT_PULLDOWN) using more power when in deep sleep and therefore using an extenral resistor will save this power ?