espressif/arduino-esp32 #1114: Question about SNTP refresh/updates.

CelliesProjects, 2018-02-14:
I have 2 ESP32s running as led controllers.
Both run the same code, and use SNTP to sync the RTC.

Today after running for 10 days there is a time difference between the 2 devices of about 1 minute.

I suspect the SNTP code does not update the time after I set it with the following code:

````cpp
void ntpTask( void * pvParameters )
{
  const char * NTPpool         = "pool.ntp.org";
  const char * defaultTimezone = "CET-1CEST,M3.5.0/2,M10.5.0/3";
  char NTPpoolAdress[20];

  if ( COUNTRY_CODE_ISO_3166 )
  {
    snprintf( NTPpoolAdress, sizeof( NTPpoolAdress ), "%s.%s", COUNTRY_CODE_ISO_3166, NTPpool );
  }
  else
  {
    snprintf( NTPpoolAdress, sizeof( NTPpoolAdress ), "0.%s", NTPpool );
  }

  ESP_LOGI( TAG, "NTP syncing with %s.", NTPpoolAdress );

  configTzTime( preferences.getString( "timezone", defaultTimezone ).c_str(), NTPpoolAdress );

  time_t now;
  struct tm timeinfo;

  while ( timeinfo.tm_year < ( 2016 - 1900 ) ) {
    vTaskDelay( 50 / portTICK_PERIOD_MS );
    time( &now );
    localtime_r( &now, &timeinfo );
  }
  gettimeofday( &systemStart, NULL );

  ESP_LOGI( TAG, "NTP sync @ %s", asctime( localtime( &systemStart.tv_sec ) ) );

  /* start time dependent tasks */

  BaseType_t xReturned;

  xReturned = xTaskCreatePinnedToCore(
    dimmerTask,                     /* Function to implement the task */
    "dimmerTask",                   /* Name of the task */
    2500,                           /* Stack size in words */
    NULL,                           /* Task input parameter */
    dimmerTaskPriority,             /* Priority of the task */
    &xDimmerTaskHandle,             /* Task handle. */
    1);                             /* Core where the task should run */

  ESP_LOGI( TAG, "DimmerTask %s.", ( xReturned == pdPASS ) ? "started" : "failed" );

  xReturned = xTaskCreatePinnedToCore(
    loggerTask,                     /* Function to implement the task */
    "loggerTask",                   /* Name of the task */
    3000,                           /* Stack size in words */
    NULL,                           /* Task input parameter */
    loggerTaskPriority,             /* Priority of the task */
    &xLoggerTaskHandle,             /* Task handle. */
    1);                             /* Core where the task should run */

  ESP_LOGI( TAG, "LoggerTask %s.", ( xReturned == pdPASS ) ? "started" : "failed" );

  vTaskDelete( NULL );
}
````
How do I check/set the NTP refresh rate?

lbernstone, 2018-02-14:
From lwip/sntp_opts.h:
/** SNTP update delay - in milliseconds
 * Default is 1 hour. Must not be beolw 15 seconds by specification (i.e. 15000)
 */
#ifndef SNTP_UPDATE_DELAY
#define SNTP_UPDATE_DELAY           3600000
#endif

MarkyAD, 2018-02-14:
Its set in tools/sdk/include/lwip/apps/sntp/sntp_opts.h and default is 1 hour. An esp32 clock here has been up for over a month and it keeps to within about a second compared to a gps-synced clock sitting next to it. One thing to note is that those pool servers are on a round-robin dns, so you get a different server every time a hostname is looked up. Server offsets and jitter/drift can vary widely, as does network latency, its best to use a local ntp server.

MarkyAD, 2018-02-15:
Something to test would be using known reachable ntp servers instead of the pool names. I'm not sure how lwip handles sntp dns lookups as far as whether it resolves names with each sync or whether it does so only once when sntp is initialized. If it does it only at init, then its possible one of the esp resolved an unreachable server and ran unsynced the whole time. One minute offset over 10 days sounds like a whole lot but running unsynced would explain that. Running a local ntpsec server here, ntpmon shows my esp32's checking in every hour, and they do keep reasonable time over long periods.

CelliesProjects, 2018-02-15:
@MarkyAD 
I just reset the slow running device and it is in sync with the other device after the reset.
I compare them to a radio clock.

The device running unsynced would explain the difference.

To test I let some devices run a long time ( a while ago) without syncing them and the result was that cold ESP32s run their RTC slower than warm ESPs.
The difference I noticed while testing was about the same magnitude as the difference I now see. (couple of seconds a day)

I have been testing my network code lately, so the devices were connected/disconnected to WiFi a lot, and I suspect something goes wrong with time server resolution after (repeated) WiFi disconnects.
Without any feedback from SNTP it is hard to debug.

Can I let the SNTP code do some serial debug output?

CelliesProjects, 2018-02-15:
Found [this thread on esp32.com](https://esp32.com/viewtopic.php?f=2&t=4695) which is about the same subject.

CelliesProjects, 2018-02-15:
Just tried `#define SNTP_DEBUG                  LWIP_DBG_ON` in `lwip/sntp_opts.h` but I get no output.

beegee-tokyo, 2018-02-15:
If you are on PlatformIO then you need to enable log in the platformio.ini file. [Example in the Wiki](http://desire.giesecke.tk/index.php/2018/01/30/enable-log-levels-in-platformio/)
If you are using ArduinoIDE you need to enable the log level from the menu. I am not using Arduino IDE, so I don't know exactly which menu it is.

CelliesProjects, 2018-02-15:
@beegee-tokyo 
I set log level to `Verbose` but still no log output.

MarkyAD, 2018-02-15:
The #defines in those includes are cpp conditionals for files that are shipped as precompiled static libraries, which means recompiling the sdk to affect any changes.

CelliesProjects, 2018-02-16:
@MarkyAD So on Arduino IDE logging from SNTP is not possible?

MarkyAD, 2018-02-16:
Not without compiling lwip, however it will work if you use arduino as component of esp-idf. I assume youre looking for something like this...

sntp_retry: Next request will be sent in 3000 ms
I (5378) wifi: pm start, type:0
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Server address resolved, sending request
sntp_send_request: Sending request to server
sntp_process: Fri Feb 16 00:58:45 2018
, 839022 ussntp_recv: Scheduled next time request: 3600000 ms

You may even be able to compile a debugging version of liblwip.a on arduino as component and copy it to the proper place in arduino-ide and gain a debugging version of lwip that way.

MarkyAD, 2018-02-16:
> You may even be able to compile a debugging version of liblwip.a on arduino as component and copy it to the proper place in arduino-ide and gain a debugging version of lwip that way.

Confirmed this does indeed work, just save a copy of the original liblwip.a, the sntp debugging adds about 10KB to code size.

CelliesProjects, 2018-02-16:
@MarkyAD Nice, but I don't really understand. Could you elaborate?

MarkyAD, 2018-02-16:
When a project is built under arduino-as-component of esp-idf, it compiles its own liblwip.a. This allows making changes such as those you made to the include files (sntp_opts.h) and results in a linkable library that incorporates those features. All one has to do is copy the single file liblwip.a from arduino-as-component (in the project's "build/lwip" directory) to tools/sdk/lib in the arduino-ide location and you've got a debugging enabled lwip in arduino-ide. You don't have to port your project to arduino-as-component, a simple test sketch will work.

CelliesProjects, 2018-02-18:
@MarkyAD 
Installed Arduino as component.
I then copied the blink example sketch in the idf.
Enabled `Forward ESP_LOGx to Arduino log output ` in menuconfig.
It compiles and is flashed succesfully to my esp.
Then I copied `liblwip.a` from the `build/lwip` folder to `arduino tools/sdk/lib` folder.
Arduino IDE is set to log level verbose.
But I still get no log output from SNTP.

MarkyAD, 2018-02-21:
@CelliesProjects did you edit the include sntp_opts.h like you did in arduino-ide? Its in a different place... components/lwip/include/lwip/apps/sntp/sntp_opts.h

MarkyAD, 2018-02-21:
@CelliesProjects under the directory you installed components to, edit components/arduino/tools/sdk/include/lwip/apps/sntp/sntp_opts.h the same as you did in arduino-ide, and in components/arduino/tools/sdk/include/lwip/lwipopts.h find LWIP_DEBUG and set it to LWIP_DBG_ON.

tferrin, 2018-02-21:
You don't have to recompile the whole liblwip.a to do what you want if using the Arduino IDE. Just change the #defines you want in lwip/sntp_opts.h, copy the source module (in this case sntp.c) to you local sketch folder, and rebuild the sketch. The Arduino IDE will compile all the source files in the local sketch folder and explicitly include those when the linker is invoked. So your replacement sntp module will be used and any other unmodified LWIP modules will come from liblwip.a.

CelliesProjects, 2018-02-24:
@tferrin That would be a useful addition. Would that be IDF or Arduino IDE code?

tferrin, 2018-02-24:
This would need to be in the IDF, which will then (eventually) get incorporated into the arduino-esp32 code base. So it will take some time. I've got a version running now but it needs more testing against unreliable ntp servers.

CelliesProjects, 2018-03-06:
@tferrin 

> You don't have to recompile the whole liblwip.a to do what you want if using the Arduino IDE. Just change the #defines you want in Arduino/hardware/.../lwip/sntp_opts.h, copy the source module (in this case sntp.c) to your local sketch folder, and rebuild the sketch. The Arduino IDE will compile all the source files in the local sketch folder and explicitly include those when the linker is invoked. So your replacement sntp module will be used and any other unmodified LWIP modules will come from liblwip.a.

Got that part working, thank you.

For reference: the mentioned files reside at:

/home/<USER>/esp/esp-idf/components/lwip/apps/sntp/sntp.c
/home/<USER>/esp/esp-idf/components/lwip/include/lwip/apps/sntp/sntp_opts.h

CelliesProjects, 2018-03-08:
Now I got some messages:

The SNTP client sets the time OK first couple of times.
After a while it fails to resolve the DNS to IP which would explain why one of my ESP32s gets out of sync.
Strange thing is, the interval between syncs should be one hour (Scheduled next time request: 3600000 ms), but it tries to sync again after 6 minutes.

I am running a somewhat older SDK (v3.0-dev-1325-ge5b2c1cf) because of (other) issues, so this weekend I will update esp-idf and Arduino ESP32 to see if that changes things.

````bash
x⸮x⸮x⸮⸮ ⸮ x⸮ ⸮ x⸮ x⸮x⸮⸮x ⸮x⸮x⸮⸮⸮⸮0,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
[I][aquacontrol32.ino:297] setup(): aquacontrol32 V0.9.0
[I][aquacontrol32.ino:298] setup(): ESP32 SDK: v3.0-dev-1325-ge5b2c1cf
[I][aquacontrol32.ino:308] setup(): Forced ILI9341 start.
[I][aquacontrol32.ino:330] setup(): Found I2C device at address 0x3c.
[I][oledtask.ino:6] oledTask(): OLED initialized.
[I][temptask.ino:25] tempTask(): 0 Dallas sensors found.
[I][spiffstask.ino:3] spiffsTask(): Starting SPIFFS.
[I][wifitask.ino:10] wifiTask(): Connecting WiFi
[I][spiffstask.ino:11] spiffsTask(): SPIFFS started.
[I][wifitask.ino:77] wifiTask(): WiFi connected to huiskamer
[I][wifitask.ino:80] wifiTask(): Local IP: 192.168.0.189
[I][wifitask.ino:83] wifiTask(): MAC: 24:0a:c4:82:70:98
[I][webservertask.ino:863] setupMDNS(): Started MDNS service: 'testnaam1.local'
[I][ntptask.ino:16] ntpTask(): NTP syncing with nl.pool.ntp.org.
sntp_request: Waiting for server address to be resolved.
[I][webservertask.ino:24] webServerTask(): Startisntp_dns_found: Server address resolved, sending request
sntp_send_request: Sending request to server
ng webserver setup. 
[I][webservertask.ino:731] webServerTask(): HTTP server setup done.
sntp_process: Thu Mar  8 14:03:11 2018
, 58201 ussntp_recv: Scheduled next time request: 3600000 ms
[I][ntptask.ino:30] ntpTask(): NTP sync @ Thu Mar  8 14:03:11 2018

[I][dimmertask.ino:9] dimmerTask(): Default timers loaded.
[E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: channelminimum1 NOT_FOUND
[E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: channelminimum2 NOT_FOUND
[E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: channelminimum3 NOT_FOUND
[E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: channelminimum4 NOT_FOUND
[E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: pwmfrequency NOT_FOUND
[E][Preferences.cpp:354] getUInt(): nvs_get_u32 fail: pwmdepth NOT_FOUND
[I][dimmertask.ino:118] setupDimmerPWMfrequency(): PWM frequency set to 1.22 kHz.
[I][dimmertask.ino:119] setupDimmerPWMfrequency(): PWM bit depth set to 16 bits.
[I][dimmertask.ino:120] setupDimmerPWMfrequency(): Maximum raw value set to 0xffff or 65535 decimal.
[I][ntptask.ino:45] ntpTask(): DimmerTask started.
[I][ntptask.ino:56] ntpTask(): LoggerTask started.
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Server address resolved, sending request
sntp_send_request: Sending request to server
sntp_process: Thu Mar  8 14:09:11 2018
, 100142 ussntp_recv: Scheduled next time request: 3600000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Server address resolved, sending request
sntp_send_request: Sending request to server
sntp_process: Thu Mar  8 14:15:11 2018
, 122966 ussntp_recv: Scheduled next time request: 3600000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Server address resolved, sending request
sntp_send_request: Sending request to server
sntp_process: Thu Mar  8 14:21:11 2018
, 147184 ussntp_recv: Scheduled next time request: 3600000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Server address resolved, sending request
sntp_send_request: Sending request to server
sntp_process: Thu Mar  8 14:27:11 2018
, 176704 ussntp_recv: Scheduled next time request: 3600000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Failed to resolve server address resolved, trying next server
sntp_retry: Next request will be sent in 3000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Failed to resolve server address resolved, trying next server
sntp_retry: Next request will be sent in 6000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Failed to resolve server address resolved, trying next server
sntp_retry: Next request will be sent in 12000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Failed to resolve server address resolved, trying next server
sntp_retry: Next request will be sent in 24000 ms
sntp_request: Waiting for server address to be resolved.
sntp_dns_found: Failed to resolve server address resolved, trying next server
sntp_retry: Next request will be sent in 24000 ms
...
...
...
...
````

tferrin, 2018-03-08:
Can you post the code for ntpTask.ino please.

> ntp_recv: Scheduled next time request: 3600000 ms
This line indicates that the sntp module has set a timeout that will send the next ntp request in one hour. The fact that a request gets sent only 6 minutes later likely means that sntp_init() somehow got called again. I suspect something in ntpTask.ino is to blame.

Also, the LWIP DNS name resolver caches the lookups it does and so you should not be seeing "Waiting for server address to be resolved" except for the first one unless you are doing several other DNS lookups and the cache entry is getting wiped.

CelliesProjects, 2018-03-08:
@tferrin That would be this:
````cpp
void ntpTask( void * pvParameters )
{
  const char * NTPpool         = "pool.ntp.org";
  const char * defaultTimezone = "CET-1CEST,M3.5.0/2,M10.5.0/3";
  char NTPpoolAdress[20];

  if ( COUNTRY_CODE_ISO_3166 )
  {
    snprintf( NTPpoolAdress, sizeof( NTPpoolAdress ), "%s.%s", COUNTRY_CODE_ISO_3166, NTPpool );
  }
  else
  {
    snprintf( NTPpoolAdress, sizeof( NTPpoolAdress ), "0.%s", NTPpool );
  }

  ESP_LOGI( TAG, "NTP syncing with %s.", NTPpoolAdress );

  configTzTime( preferences.getString( "timezone", defaultTimezone ).c_str(), NTPpoolAdress );

  time_t now;
  struct tm timeinfo;

  while ( timeinfo.tm_year < ( 2016 - 1900 ) ) {
    vTaskDelay( 50 / portTICK_PERIOD_MS );
    time( &now );
    localtime_r( &now, &timeinfo );
  }
  gettimeofday( &systemStart, NULL );

  ESP_LOGI( TAG, "NTP sync @ %s", asctime( localtime( &systemStart.tv_sec ) ) );

  /* start time dependent tasks */

  BaseType_t xReturned;

  xReturned = xTaskCreatePinnedToCore(
    dimmerTask,                     /* Function to implement the task */
    "dimmerTask",                   /* Name of the task */
    2500,                           /* Stack size in words */
    NULL,                           /* Task input parameter */
    dimmerTaskPriority,             /* Priority of the task */
    &xDimmerTaskHandle,             /* Task handle. */
    1);                             /* Core where the task should run */

  ESP_LOGI( TAG, "DimmerTask %s.", ( xReturned == pdPASS ) ? "started" : "failed" );

  xReturned = xTaskCreatePinnedToCore(
    loggerTask,                     /* Function to implement the task */
    "loggerTask",                   /* Name of the task */
    3000,                           /* Stack size in words */
    NULL,                           /* Task input parameter */
    loggerTaskPriority,             /* Priority of the task */
    &xLoggerTaskHandle,             /* Task handle. */
    1);                             /* Core where the task should run */

  ESP_LOGI( TAG, "LoggerTask %s.", ( xReturned == pdPASS ) ? "started" : "failed" );

  vTaskDelete( NULL );

  //https://www.ibm.com/developerworks/aix/library/au-aix-posix/index.html#artdownload
  //https://github.com/espressif/esp-idf/blob/master/examples/protocols/sntp/README.md
  //https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-time.c
  //https://www.di-mgt.com.au/wclock/tz.html
  //http://www.catb.org/esr/time-programming/
  //http://www.lucadentella.it/en/2017/05/11/esp32-17-sntp/
  //https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/sntp/sntp.c
  //https://github.com/hwstar/freertos-avr/blob/master/include/time.h

  /* https://stackoverflow.com/questions/3673226/how-to-print-time-in-format-2009-08-10-181754-811 */

}
````

CelliesProjects, 2018-03-08:
@tferrin `...unless you are doing several other DNS lookups and the cache entry is getting wiped.`

Checking the rest of the code now...

tferrin, 2018-03-08:
In the above code, the NTPpoolAdress array is on the stack, so once the ntpTask() function returns its value is no longer valid. But configTzTime() passed a pointer to this array to sntp(), which subsequently tries to use the invalid array data. This is the primary cause of your problem.
But secondly, the call to vTaskDelete( NULL ) deletes the ntpTask and means that the heap space previously allocated to it is freed. I'm not sure, but sntp() may be using this space for its stack. I'd fix the first problem and if you still have issues then dig deeper into the possible second problem.

CelliesProjects, 2018-03-09:
@tferrin  Thanks for the code review & explanation.

closed 2018-03-09 by CelliesProjects

CelliesProjects, 2018-03-30:
After a lot of testing it seems that @tferrin set me on the right track.
Marking this issue as solved.