espressif/arduino-esp32 #1116: ESP32 SPIFFS Webserver

shaddow501, 2018-02-15:
Hello.

This is not an issue but a question.

I would like to run Webserver from the internal SPIFFS and from the local website to be able to load / access  files from the SD card.

I have the standard Webserver working and that works ok, also the SD card store log files from inputs that the ESP32 gets.

This all works, but how do I open the files from the SD card on the Webpage? what need to be done?

rndme, 2018-02-23:
there's an SD server example in the esp8266 package, it's going to be 90% the same for esp32. You basically need to alter the SPIFFs example's 404 handler to use SD.exists() instead of SPIFFS.exists, and the load routines as well.

shaddow501, 2018-02-27:
Dear rndme, thanks for you answer, Yes I did try this lib but my question is how to integrate both SPIFFS and SD in the same webserver and to be able to access them both, My logic is to put the website on the SPIFFS and read from the webserver the logs from the SD-Card, how to do that this is the question...

me21, 2018-08-24:
Unfortunately, no. Never worked with SD Card myself.

shaddow501, 2018-08-24:
lol so how can you be sure that your example works?

me21, 2018-08-24:
I'm not sure, try and we will see :)

closed 2019-08-15 by stale[bot]