espressif/arduino-esp32 #1121: 'class WiFiClass' has no member named 'mode'

Brunez3BD, 2018-02-16:
https://github.com/espressif/arduino-esp32/blob/a153f6ad280e5932be9c80341556348ff505d586/libraries/WiFi/examples/WiFiScan/WiFiScan.ino#L13.

As reported in the title, the member 'mode' is missing in the library

atanisoft, 2018-02-17:
The wrong WiFi library is being compiled. It is not using the one in the tree you are bringing above

Brunez3BD, 2018-02-17:
So what's the correct library? Why keeping the wrong one in the tree?

atanisoft, 2018-02-18:
If you are using platformio don't list it in the initial file and clean the project temp directories. It should just work.

Brunez3BD, 2018-02-20:
I've tried with no success.. I only need the correct library.
It makes no sense to have examples not working

Brunez3BD, 2018-02-20:
Using PLatformIO on Atom, not working

atanisoft, 2018-02-21:
@Brunez3BD I would recommend moving this issue over to https://github.com/platformio/platform-espressif32. This is not a bug in the library or in arduino-esp32. It is a bug in the LDF component of PlatformIO picking the wrong library, it is picking a generic library and not the built-in one from arduino-esp32

Brunez3BD, 2018-02-23:
Moved to [https://github.com/platformio/platform-espressif32/issues/72](url)

everslick, 2018-04-19:
issue moved to platformio -> closing

closed 2018-04-19 by everslick

antsa1, 2018-04-19:
@everslick why did you closed this issue? is your problem resolved? I still have got the same issue while trying to upload the code from freematics

everslick, 2018-04-19:
As atanisoft explained above, this is not an issue belonging in the arduino-esp32 issue tracker and was thus moved to platformio, where the issue can be further discussed.

If you feel this is still a bug in Arduino-ESP32, please reopen the issue and explain your reasoning.

antsa1, 2018-04-19:
@everslick ooh ok, now I get it. Thanks for the reply

akloster, 2018-06-29:
I'm getting this error from a fresh install of arduino nightly and esp32-arduino.

Hindurable, 2019-02-13:
I don't understand.. Where can I get proper WiFi class? Boards manager and git version both doesn't have .mode member. Is only way to start using Platformio to get this work?

SirBarcla, 2020-07-21:
hAVING THE SAME PROBLE

Michail1, 2020-07-25:
WifiManager library.
Versions up to 0.15.0 support the mode.
Versions after it (i.e. 2.0.0 and above) do NOT support it.

Simply downgrade to 0.15.0 in the library manger or lower to get the functionality back (and therefore compiles).

Michail1, 2020-07-25:
The problem people might be having is that they are trying to compile the esp32 version while having the board set to esp8266, etc.  So, make sure the correct board is selected.

MiteshSonagra, 2020-09-22:
if you face this error #  'class wificlass' has no member named 'softap'   #
  try this librery is <ESP8266wifi.h>

CodingCreate101, 2020-10-30:
`ESP8266` uses `<ESP8266wifi.h>`

lovepussy612, 2026-02-22:
same issue existed ,my dev env include vscode+plattformio ,and the ino is 

[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino