espressif/arduino-esp32 #1122: Timezone environment variable style issue

huming2207, 2018-02-17:
Hi all,

Just a question about time timezone issues when calling the `configTzTime()` API. I'm currently in China and I also live in Melbourne, Australia. For the first argument `const char *tz` in the `configTzTime()` API, it seems that there are two styles of the timezone strings, which are either:

- `AUSEST-10AUSDST-11,M10.5.0/02:00:00,M3.5.0/02:00:00` or `Australia/Melbourne` for Melbourne, Australia
- `CST-8` or `Asia/Shanghai` for mainland China

Since ESP32 IDF implementation follows POSIX standard, can I just simply specify the `CountryName/CityName` style timezone variable string for `configTzTime()`? 

Regards,
Jackson

lbernstone, 2018-02-17:
It is just passing this along to tzset, so you will need to use tzset syntax.

lbernstone, 2018-06-06:
Can this issue be closed?

huming2207, 2018-06-06:
Ah yea I will close it. Sorry I forgot that

closed 2018-06-06 by huming2207