espressif/arduino-esp32 #1123: fdev_setup_stream function missing

VioletGiraffe, 2018-02-17:
The Arduino platform for ESP32 is missing the `fdev_setup_stream` function. I came across it in this Linpack benchmark implementation for Arduino: https://gist.github.com/projectgus/8279947

This function is obviously not essential nor widely used, but adding it will improve compatibility with existing code base.

VioletGiraffe, 2019-08-01:
`fdev_setup_stream` still missing from `<stdio.h>`.

atanisoft, 2019-08-01:
@VioletGiraffe where is fdev_setup_stream defined? It is not part of the C stdio.h file.

VioletGiraffe, 2019-08-01:
@atanisoft : good point! Looks like it's specific to AVR libc, not sure why they put it `stdio.h` then. Sorry, should have realized it's not standard C.

closed 2019-08-01 by VioletGiraffe