espressif/arduino-esp32 #1126: Wake up from deep sleep by monitoring internal Hall sensor?

Abishek05, 2018-02-18:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?18/feb/2018?
IDE name:							?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Is it possible to wakeup Esp32 from deep sleep by monitoring the adc value of the internal Hall sensor readings?

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```

```

beegee-tokyo, 2018-02-18:
Here are all possibilities listed to wake up from deep sleep: [esp-idf deep sleep](http://esp-idf.readthedocs.io/en/latest/api-reference/system/sleep_modes.html). The hall sensor is not mentioned.

gabriel-milan, 2019-05-28:
Hey!

I was having a similar issue and realized there's few material on the internet for that.
I've written a very simple code for waking up ESP32 according to a threshold on hall sensor readings, hope it helps!

Link for the code: https://github.com/gabriel-milan/esp32_ulp_hall_wakeup

closed 2019-08-14 by stale[bot]