espressif/arduino-esp32 #1130: HTTP 2 support to WiFiClientSecure

mpatafio, 2018-02-20:
Hi!

I'm trying to use the WiFiClientSecure to post a request using HTTP 2, but I guess the underlying http client doesn't support this new standard 

got: "HTTP/2 client preface string missing or corrupt"

Do you know if it's possibile to get this for our esp32?

Thank you in advance

lbernstone, 2018-02-22:
HTTP/2 is not really a drop in replacement for HTTP/1.  It is unlikely that WFCS can really support it without significant structural changes.  The esp-idf has included nghttp2 (https://github.com/espressif/esp-idf/tree/master/components/nghttp), and you can access those api's within Arduino IDE.  They are C bindings, and so you won't have nice object classes.  If you want to make a library to provide those, the community would certainly appreciate it.

anteph, 2019-01-05:
Hi,

I've been putting together a couple of tutorials about HTTP/2 on the ESP32 using the Arduino core and an IDF wrapper called sh2lib. 

Please note that these are based on experiences I've been making and that I though might be useful for the community, I'm no expert on HTTP/2 :)

As @lbernstone mentioned, it doesn't have nice object classes, but this sh2lib wrapper removes a lot of the NGHTTP2 complexity and helps getting started with the basics much faster.

Not sure if this is exactly what you are looking for, but maybe it can give you some help getting started:

Install sh2lib wrapper (video):
https://www.youtube.com/watch?v=00P8oIbNK-Q

Basic operations: 
https://techtutorialsx.com/2018/11/18/esp32-http-2-connecting-to-server/   (also explains how to install sh2lib)
https://techtutorialsx.com/2018/11/19/esp32-arduino-http-2-get-request/
https://techtutorialsx.com/2018/12/19/esp32-http-2-post-request/

More advanced tutorials
https://techtutorialsx.com/2018/12/02/esp32-arduino-http2-using-query-parameters/
https://techtutorialsx.com/2018/11/25/esp32-http-2-adding-headers-to-get-request/
https://techtutorialsx.com/2019/01/04/esp32-http-2-post-request-with-headers/
https://techtutorialsx.com/2018/12/08/esp32-http-2-decompressing-brotli-encoded-content/ESP32 
https://techtutorialsx.com/2018/11/23/esp32-http-2-get-request-with-continuous-stream-response/

Also, for reference, here is the link for IDF's HTTP/2 tutorial:
https://github.com/espressif/esp-idf/tree/master/examples/protocols/http2_request

Hope this helps :)

Best regards,
Nuno Santos

closed 2019-08-15 by stale[bot]