espressif/arduino-esp32 #1131: Wrover Kit SD

Crunkstar, 2018-02-20:
### Hardware:
Board:							ESP32 Wrover Kit v3
Core Installation/update date:			20/Feb/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:

Not sure if this is the right place to ask this (sorry if it's not) but I can't get the SD adapter on my wrover kit v3 working and I can also not find any documentation, about using the wrovers SD card, online . Well, theres not much else to say about it else than my knowledge about ESP32s is very very small so sorry if I'm just too stupid.

### Sketch:
```
SDMMC_Test.ino

Only change to that file:
pinMode(2, INPUT_PULLUP);
in setup() in hopes of it helping
```

### Debug Messages:
```
E (95) sdmmc_req: handle_idle_state_events unhandled: 00000004 00000000
E (97) sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107
[E][SD_MMC.cpp:68] begin(): Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.
Card Mount Failed
```

lbernstone, 2018-02-20:
The SD card pins are not at the defaults on the WROVER kit.  Try replacing the SD.begin() line with this:
```
    SPI.begin(14, 2, 15, 13);
    if(!SD.begin(13)){
```

Crunkstar, 2018-02-20:
Well, thats embarassing, I checked the datasheet before and thought the pins were alright.

It now fails on the sd card initialization and I get a Guru Meditation Error, though.
I can't make out if this is due to my sd card or some other error on my side or a problem with the library (pretty sure it's not the latter)

This is the decoded Guru Meditation Error, which doesn't really tell me anything so I hope someone can help me out with this.

```0x400d2e07: sdcard_init(unsigned char, SPIClass*, int) at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\libraries\SD\src/sd_diskio.cpp line 686
0x400d3fe8: spiFrequencyToClockDiv at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-spi.c line 528
0x400d404a: spiFrequencyToClockDiv at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-spi.c line 528
0x400d2e07: sdcard_init(unsigned char, SPIClass*, int) at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\libraries\SD\src/sd_diskio.cpp line 686
0x400d1fff: fs::SDFS::begin(unsigned char, SPIClass&, unsigned int, char const*) at c:\users\frank\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h line 127 (discriminator 1)
0x400d11aa: setup() at C:\Users\Frank\AppData\Local\Temp\arduino_modified_sketch_565416/SD_Test.ino line 179
0x400ec387: loopTask(void*) at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 15
```

> Side note: My issue referred to usage of the "SD_MMC" library and not the "SD" library but for me it doesn't really matter as long as I can use the wrover kits SD card slot

lbernstone, 2018-02-21:
I'm not sure why you are using SD_MMC, as I'm pretty sure all microSD are SD(SC/HC/XC).  Your error seems to be related to setting the frequency, which is 1000000 (really low) by default.  I don't have a WROVER kit, so it is possible there is a frequency issue with the board. My dev board works with this exact code:
```arduino
#include "SD.h"
void setup() {
    Serial.begin(115200);
    SPI.begin(14, 2, 15, 13);
    if(!SD.begin(13)){
        Serial.println("Card Mount Failed");
        return;
    }
    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);
}
void loop() {}
```

Crunkstar, 2018-02-21:
As I said, for me it doesn't matter if I use SD_MMC or SD as long as it's working 

> (to be honest I don't know the difference between SD_MMC and SD, except that SD_MMC doesn't use SPI, I was only using it because the standard pinout looked like the one I found on the wrover kit v3 documentation)

I've also tried multiple different SPI frequencies but probably didn't do it the right way as I'm not exactly sure how to set the SPI frequency.

The ways I tried were, pasting some piece of code I don't remember, that I have found online and changing the frequency variable in sd_diskio.cpp. Probably both of those were wrong so it'd be awesome if you could correct me on how to change the frequency and maybe some frequencies to try.

Also, thank you for your help so far!

lbernstone, 2018-02-21:
On second look at your stack trace, I think it is getting through setup and then choking on loop.  Try putting delay(100); into your loop function.  I have seen instances where it upset things to have an empty loop.

Crunkstar, 2018-02-22:
I don't when or due to what the Guru Meditation Error changed but now it is:
```0x400d1a17: sdcard_init(unsigned char, SPIClass*, int) at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\libraries\SD\src/sd_diskio.cpp line 772
0x400d37a0: spiFrequencyToClockDiv at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-spi.c line 528
0x400d3802: spiFrequencyToClockDiv at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-spi.c line 528
0x400d1a17: sdcard_init(unsigned char, SPIClass*, int) at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\libraries\SD\src/sd_diskio.cpp line 772
0x400d0c8f: fs::SDFS::begin(unsigned char, SPIClass&, unsigned int, char const*) at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\libraries\SD\src/SD.cpp line 104
0x400d0ba9: setup() at C:\Users\Frank\AppData\Local\Temp\arduino_modified_sketch_121973/sketch_feb22a.ino line 5
0x400eba2f: loopTask(void*) at C:\Users\Frank\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 15
```
This stays the same whether I have an SD card in there or not and whether the loop() is empty or has a delay in it

kosso, 2018-05-29:
@Crunkstar Hi. Did you even get the SD card on the ESP-WROVER-KIT (v3) working?

I'm having the same issue. 

Since this is a genuine Espressif board (and an expensive one at that) I expected fully working examples of all the hardware features this dev board /kit comes with. ;)

lbernstone, 2018-05-29:
@kosso Have you tried https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card ?

kosso, 2018-05-29:
@lbernstone Hi. I just tried this method (using esp-idf tools) and initially it failed to mount, providing a message in the error about a flag to set for reformatting the card if the mounting fails. 

This appears to have worked, despite printing the same error as when I tried the Arduino libraries. 

My serial output was eventually: 

```
[0;32mI (253) example: Initializing SD card[0m
[0;32mI (253) example: Using SDMMC peripheral[0m
[0;31mE (293) sdmmc_req: handle_idle_state_events unhandled: 00000004 00000000[0m
Name: SU01G
Type: SDSC
Speed: default speed
Size: 968MB
CSD: ver=0, sector_size=512, capacity=1984000 read_bl_len=9
SCR: sd_spec=1, bus_width=5
[0;32mI (323) example: Opening file[0m
[0;32mI (333) example: File written[0m
[0;32mI (343) example: Renaming file[0m
[0;32mI (353) example: Reading file[0m
[0;32mI (353) example: Read from file: 'Hello SU01G!'[0m
[0;32mI (353) example: Card unmounted[0m

```

So it seemed to do OK when formatted by the example. (though now the card, an old 1Gb microSD, is now not even readable by macOS - which surprises me. It's usually quite good at reading most disk formats)

And now, the Arduino SD_MMC library test appears to read the card OK. 

```
SD_MMC Card Type: SDSC
SD_MMC Card Size: 968MB
...
```
and also with an 8Gb card: 
```
SD_MMC Card Type: SDHC
SD_MMC Card Size: 7580MB
...
```

So there's clearly a specific type, size and format which the SD reader will work with. (This is quite common in some 3D printers which often require FAT16 or FAT32) 

Interestingly, when I let the example code reformat the card after failing to mount, my macOS 'saw' it as 'ExFat' format - but could not read it.   
And a Linux Ubuntu laptop failed to open it at all, throwing lots of errors.

lbernstone, 2018-05-30:
esp-idf also has this function: https://esp-idf.readthedocs.io/en/v2.0/api/storage/fatfs.html#using-fatfs-with-vfs-and-sd-cards.  I'll play around and see if I can't get that working and try to put something together for arduino.

closed 2019-08-15 by stale[bot]

michaliskon, 2020-04-14:
This works fine for me and I can use the SD library.

SPI.begin(14, 2, 15, 13);
    if(!SD.begin(13)){

Any idea how I can use the SdFat library on WROVER?