espressif/arduino-esp32 #1134: Arduino-esq example of using BT RFCOMM/SPP from IDF updates

spmp, 2018-02-21:
I am grateful that RFCOMM/SPP have been implemented in the IDF, but try as I might to bend my head around the examples it is too low level and magic for my n00b/out of practice mind.

Can someone, maybe @anteph, please do a tutorial or give code snippets as to how we could get a RFCOMM like Serial send and receive such that we can debug and interact with our awesome esp32 projects from our phones with the likes of  https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal

Cheers

copercini, 2018-02-22:
[image]
Soon.....

spmp, 2018-02-22:
Tease!

UndefinedOrigin, 2018-02-22:
Ohhh... and I was just about to start installing the IDF and Eclipse

beegee-tokyo, 2018-02-23:
More teasing!!! Works like a charm!
My [ESP32 Weatherstation](http://desire.giesecke.tk/index.php/2018/01/29/esp32-weatherstation/) sending data over Bluetooth Serial to my tablet:
[image]
To get it running, I downloaded the Bluetooth Serial files as mentioned in this [pull request](https://github.com/espressif/arduino-esp32/pull/1144/files) and put them into my **_packages/framework-arduinoespressif32_** folder in the correct sub-folders.

anteph, 2018-02-24:
@spmp as soon as the BluetoothSerial.h is available in the Arduino core, I will try to make some tutorials and share here the links :) Currently I only have some tutorials using the BTStack library on IDF which, as you have mentioned, are pretty low level.

@copercini looking good and easy to use :) I'm looking forward to start playing with it, once again great work and thanks for making this available to the community!

anteph, 2018-03-04:
Hi! I've downloaded the binary file from the Pull Request and started playing around a bit with the IDF API on the Arduino core.

I've written a small tutorial on how to start the Bluetooth controller and host and how to make the ESP32 discoverable. Not sure if it helps, but I tried to explain what the function calls do, so it helps others understand better what is going on under the hood.

https://techtutorialsx.com/2018/03/04/esp32-arduino-bluetooth-classic-getting-started/

Just as a note, I'm no Bluetooth expert and just start playing with the API, so if you have the chance to read the tutorial please let me know if there is something wrong or some imprecision :)

Best regards,
Nuno Santos

Best regards,
Nuno Santos

UndefinedOrigin, 2018-03-04:
Awesome, thank you.  I'll play with that tonight!

spmp, 2018-03-07:
PR merged, example code so simple and amazing!
@anteph I think the example could be made easier by simply copying the example from BluetoothSerial/examples//....  no need to mention PR's or anything now.

Too easy!

Pretty ready to close this as the example provided is a good start, but a general purpose guide would also be great 8)

closed 2018-03-11 by spmp