espressif/arduino-esp32 #1135: Wifi + Bluetooth together works only for 2 to 30 min

Morayetl, 2018-02-21:
Hi!

I'm having issues to get wifi+BT working together with my ESP32 board, I've gone trough many forums and i cant find any solution for this issue and i read that this is known issue, but is there any solution for this?

beegee-tokyo, 2018-02-21:
Biggest problem is that WiFi and Bluetooth stack together don't fit into the default partition sizes.
I have WiFi and Bluetooth running together in my small project [ESP32 Weatherstation](http://desire.giesecke.tk/index.php/2018/01/29/esp32-weatherstation/). I needed to change the partition sizes before I got Bluetooth BLE even to compile without error. Here is a guide how to change partition size [Change partition size](http://desire.giesecke.tk/index.php/2018/01/30/change-partition-size/)

IvanDreamer, 2018-08-30:
Hi!  Does somebody know why Bluetooth does not work if I start Wifi with Wifi.begin command ? 
Thanks for help.

closed 2019-08-15 by stale[bot]