espressif/arduino-esp32 #1136: Turning Wifi off dont save current

JLuisAlcaraz, 2018-02-21:
Hia all!
I need to turn wifi and ble off, in order to save battery. I tried the example: WifiBluetoothSwitch, in Arduino examples, and I tested the current comsuption.

I tested with nothing connected to the Nodemcu ESP32 board.

When I first apply tester, to measure current (and supply voltage), the meter read 76,5mA.
When I press the button, the read is always 108.5mA, no matter how many times I press the button... always 108.5mA,

¿It is normal?

lbernstone, 2018-02-21:
This won't turn the wifi off but does put it in power save mode:
```arduino
#include <esp_pm.h>
#include <esp_wifi.h>
#include <esp_wifi_types.h>
#include <WiFi.h>

boolean flagPS = false;
boolean setPS = false;

void IRAM_ATTR set_ps() {
  flagPS = true;
}

void isr_pin0() {
// Puts an interrupt on pin 0 (the top button on Heltec boards)
  pinMode (0, INPUT);
  attachInterrupt(digitalPinToInterrupt(0), set_ps, RISING);
}

void setup() {
  Serial.begin(115200);
  isr_pin0();
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) {delay(500);}
  delay(200);
  Serial.println(WiFi.localIP());
}

void loop() {
  if (flagPS == true && setPS == false) {
    esp_wifi_set_ps(WIFI_PS_MODEM);
    setPS = true;
    Serial.println("Power Save set");
  }
  delay(100);
}
```

closed 2019-08-15 by stale[bot]

mrWheel, 2020-07-12:
Why is this issue closed?

The issue is still not resolved!

I have tried:
- esp_wifi_stop()
- WiFi.setSleep(true)
- WiFi.stop()

In al three cases the WiFi connection is closed, but the power consumption is still somewhere around 70mA.