espressif/arduino-esp32 #1137: Failed to execute script esptool the selected serial port Failed to execute script esptool does not exist or your board is not connected

Abdulaziz95, 2018-02-21:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Error when compiling a code to the microcontroller
this is the error message:

esptool.py v2.1
Traceback (most recent call last):
  File "esptool.py", line 2524, in <module>
  File "esptool.py", line 2517, in _main
  File "esptool.py", line 2246, in main
  File "esptool.py", line 177, in __init__
  File "site-packages\serial\__init__.py", line 88, in serial_for_url
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM8': WindowsError(2, 'The system cannot find the file specified.')
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected  

### Sketch:

#define led 2

void setup () {
   pinMode(led, OUTPUT);
}

void loop () {
digitalWrite (led, HIGH);
delay(500);
digitalWrite(led, LOW);
delay(500);
}

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-02-21:
https://www.arduino.cc/en/Guide/Troubleshooting

Abdulaziz95, 2018-02-23:
I tried that, nothing worked.
also in the Arduino IDE I can't select
port
tools > port
it is in gray color

beegee-tokyo, 2018-02-23:
I am guessing you are on Windows????

- When you plug in your ESP32 to your PC, do you hear a sound? If not, try a different USB cable.
- Does Windows say anything about installing drivers failed? You might need to manually install the drivers.
- If you open the device manager of windows are there any USB devices listed (after you plugged in the ESP32) that have no drivers installed? Then you most likely need to manually install FTDI (USBtoSerial) drivers.
- I needed to install separate drivers for the onboard USB of my ESP32. In my case it was for a CP2102 chip. You can find the drivers here: [ESP32.net](http://esp32.net/usb-uart/).
- Did you try to use a different USB cable?

And last, some ESP32 boards need to be set manually into bootloader mode. On my board that would be pressing the RESET button and keep it hold, then press the USR button and keep it hold, then release the RESET button, then release the USR button. Might be different for your board.

Abdulaziz95, 2018-02-26:
Thank you,,,
The problem was the driver CP210x is missing 
and the cable wasn't good, even though I tried 4 or 5 different cables and finally I got one working.

Issue Solved

beegee-tokyo, 2018-02-26:
Glad I could help.
Please push the CLOSE button for this issue.

Shirola, 2018-03-08:
Also, if you working on a laptop, make sure it is on full performance and plugged in.

closed 2018-03-08 by copercini

glkeeney1951, 2018-04-04:
Holding down the "Boot" button on my ESP-WROOM-32 while the Arduino IDE download the program worked. After the download I had to push the "EN" button to run the program. It did NOT start on it's own.

thedim-witted, 2019-05-13:
Hey, anyone here? I have a similar issue but unlike this one, I can select my COM Port and the drivers are also up to date. However, what I am finding out of place is that in the COM Port list I am getting some 17 options for COM1 even though nothing else is connected to the laptop. Any ideas?

rohithvishaal, 2019-05-22:
> Hey, anyone here? I have a similar issue but unlike this one, I can select my COM Port and the drivers are also up to date. However, what I am finding out of place is that in the COM Port list I am getting some 17 options for COM1 even though nothing else is connected to the laptop. Any ideas?

you can get those COM port options in Device Manager

AndrasSzep, 2020-06-06:
push the "BOOT" button worked! Thx guys.

amilanpathirana, 2020-12-16:
Worked after changing my USB cable

pasindu991218, 2021-04-30:
Hey anyone here? I get the error message saying that 

the selected serial port Failed to execute esptool  does not exist or your board is not connected.
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected

I also installed the CP210x USB to UART Bridge VCP Drivers as well.

Hope someone help me with this..

Thanks

AndrasSzep, 2021-05-01:
Did you select the proper serial port & speed? perhaps lower the speed to 115 or 230kbps

pasindu991218, 2021-05-01:
Hey

When I go to device manager it does not show any port. Is that mean I have
to change the USB cable..?

AndrasSzep, 2021-05-01:
IDE –> Tools -> Port -> you shall see here your device.

------------------

pasindu991218, 2021-05-02:
no i cant select it it is gray color

pasindu991218, 2021-05-02:
when I program it using the FTDI module I cannot select a port then I
programmed it from Arduino the I got the porn as COM5.

Mowais-430, 2021-06-16:
hey guys i m so tired from this error every thing is connected but it show me errors
esptool.py v3.0-dev
Serial port COM7
Traceback (most recent call last):
  File "esptool.py", line 3682, in <module>
  File "esptool.py", line 3675, in _main
  File "esptool.py", line 3329, in main
  File "esptool.py", line 263, in __init__
  File "site-packages\serial\__init__.py", line 88, in serial_for_url
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM7': WindowsError(2, 'The system cannot find the file specified.')
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected

Adaranijo-kenneth, 2021-06-21:
How do I know which of the com ports to select,I have 4 options before me and I am having the same error as the original question

Adaranijo-kenneth, 2021-06-21:
Please what is the solution to this error
esptool.py v2.6
Serial port COM3
Connecting...
Traceback (most recent call last):
  File "esptool.py", line 2959, in <module>
  File "esptool.py", line 2952, in _main
  File "esptool.py", line 2653, in main
  File "esptool.py", line 460, in connect
  File "esptool.py", line 440, in _connect_attempt
  File "esptool.py", line 379, in sync
  File "esptool.py", line 322, in command
  File "esptool.py", line 285, in write
  File "site-packages\serial\serialwin32.py", line 323, in write
serial.serialutil.SerialTimeoutException: Write timeout
Failed to execute script esptool
An error occurred while uploading the sketch
Please what is the solution to this error

Mowais-430, 2021-06-28:
sorry for the late reply ..i dont know buddy

Kapil-chn7, 2021-07-05:
Please help me with this,

Arduino: 1.8.15 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz 

(WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Sketch uses 198834 bytes (15%) of program storage space. Maximum is 1310720 bytes.

Global variables use 13248 bytes (4%) of dynamic memory, leaving 314432 bytes for local variables. Maximum is 327680 bytes.

esptool.py v3.0-dev

Serial port COM3

Connecting...

Traceback (most recent call last):

  File "esptool.py", line 3682, in <module>

  File "esptool.py", line 3675, in _main

  File "esptool.py", line 3330, in main

  File "esptool.py", line 512, in connect

  File "esptool.py", line 492, in _connect_attempt

  File "esptool.py", line 431, in sync

  File "esptool.py", line 369, in command

  File "esptool.py", line 332, in write

  File "site-packages\serial\serialwin32.py", line 323, in write

serial.serialutil.SerialTimeoutException: Write timeout

Failed to execute script esptool

An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

me-no-dev, 2021-07-05:
1. Make sure you have a good cable
2. Enter download mode manually by resetting the board while the boot button is pressed.

Mowais-430, 2021-07-06:
Thanks for the tip!

Soma-time, 2021-07-12:
Hey guys. I was getting this error and my work around was changing the "Flash Mode" from "QIO" to "DIO".

danielalpha007, 2021-07-23:
> 1. Make sure you have a good cable
> 2. Enter download mode manually by resetting the board while the boot button is pressed.

Hello, i have a ESP32-WROOM-32 and your Tip was the Solution for me. unplug the Device, hold Boot button and plug device, further hold boot button and Upload Firmware.

lowenna, 2021-08-13:
One other solution I found today not mentioned above that might help others. 

Insufficient power from the computers (Microsoft Surface Pro 6 in my case) USB port caused exactly the same issue, even though when plugging in directly, the device was fully recognised in device manager and showing up as COM3 in my case.  I knew the cable was good as I was able to flash firmware from a desktop machine.

Resolved by using either an externally powered USB hub, or a Surface docking station which among other things is also an externally powered USB hub. 

Note I also did upgrade the driver using the `CP210x Universal Windows Driver` from https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers as that was newer than the fresh install of Windows 10 (21H1) provided, but do not believe that was necessary at all.

My ESP32 board is an [AITrip ESP32-WROOM32D](https://www.amazon.co.uk/gp/product/B08DXSMZSB). Arduino IDE settings are (reading the bottom right of the IDE) `ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 40MHz, 4MB (32Mb), 115220, None on COM3`.

Hope that helps someone 😄

danielherrera1999, 2021-12-19:
> ### Hardware:
> Board: ?ESP32 Dev Module? Core Installation/update date: ?11/jul/2017? IDE name: ?Arduino IDE? ?Platform.io? ?IDF component? Flash Frequency: ?40Mhz? Upload Speed: ?115200?
> 
> ### Description:
> Error when compiling a code to the microcontroller this is the error message:
> 
> esptool.py v2.1 Traceback (most recent call last): File "esptool.py", line 2524, in File "esptool.py", line 2517, in _main File "esptool.py", line 2246, in main File "esptool.py", line 177, in **init** File "site-packages\serial__init__.py", line 88, in serial_for_url File "site-packages\serial\serialwin32.py", line 62, in open serial.serialutil.SerialException: could not open port 'COM8': WindowsError(2, 'The system cannot find the file specified.') Failed to execute script esptool the selected serial port Failed to execute script esptool does not exist or your board is not connected
> 
> ### Sketch:
> #define led 2
> 
> void setup () { pinMode(led, OUTPUT); }
> 
> void loop () { digitalWrite (led, HIGH); delay(500); digitalWrite(led, LOW); delay(500); }
> 
> ### Debug Messages:
> ```
> Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
> ```

see into manage device, look at the port com and select the port number you use, click on this, open the settings and configurate "port configuration" and change bits per seconds

desi2108, 2022-04-02:
i try to change my port in device manager, and it's worked

rajojon23, 2022-11-30:
I was using the wrong port as well! It stayed on "Connecting..." while compiling before the timeout error. It worked once I switched to the correct port.

pythonworkspy, 2023-03-07:
> 
Thanks it works for me.

HelloEveryonehh, 2023-03-23:
ues this to update ‘https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json’

quackduck, 2024-01-18:
Unplugging and plugging in the ESP32 worked for me, surprisingly

gungriffonbr, 2024-12-21:
> Thank you,,, The problem was the driver CP210x is missing and the cable wasn't good, even though I tried 4 or 5 different cables and finally I got one working.
> 
> Issue Solved

thanks for the tip solved a issue  for me!!