espressif/arduino-esp32 #1142: Compiling core at 80Mhz

dvizma, 2018-02-22:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:	       11/jul/2017
IDE name:					        Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
Hi,

I have been using Arduino IDE for ESP32. As per my search, ESP32 core for Arduino has been compiled with 240 Mhz clock. I have never used IDF before. Can someone tell the basics steps how to compile a new arduino core for esp32 with 80Mhz clock.

Thanks

JacoFourie, 2018-02-22:
You just set it in the options of the board in the IDE.

dvizma, 2018-02-22:
Isn't that option for flash frequency?
As per discussions in this issue thread https://github.com/espressif/arduino-esp32/issues/487 . I need to recompile the core at 80Mhz. Wanna know how. Thanks.
#487

lbernstone, 2018-02-23:
1) Install arduino-esp32 as per instructions (https://github.com/espressif/arduino-esp32)
2) Install esp-idf as per instructions (https://github.com/espressif/esp-idf)
3) Get esp-idf in line with arduino-esp32.  Looks through the commit log for the latest IDF update (I search for IDF).  Use git log to get the full uuid for that commit and then git checkout that version.
4) Follow the instructions to install arduino as an esp-idf component (https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).  You can use any project. I use the blink example.
5) Copy the sdkconfig from arduino-esp32/tools/sdk/sdkconfig into your project.
6) Run make menuconfig and set the cpu speed (or whatever change you want). Save that sdkconfig.
7) Make the project
8) Copy build/esp32/libesp32.a into arduino-esp32/tools/sdk/lib/ .
9) Profit??

Aalz, 2018-03-26:
When I copied the file and tried to load the sketch through the Arduino Ide, an error was displayed:

D:/ESPRESSIF/esp-idf/components/esp32/panic.c:648: undefined reference to `xPortInterruptedFromISRContext'

collect2.exe: error: ld returned 1 exit status

exit status 1

Aalz, 2018-03-26:
How to compare esp-idf versions?

lbernstone, 2018-05-13:
Return to step 3 and try again.  You must do a `git checkout <version>` in esp-idf to match the <version> that arduino-esp32 was compiled at. That currently is `git checkout f586f5e690e687f8db219a8b509bbf483fc416d6`

Buffalchill, 2018-05-13:
ok thanks! but i didnt unterstand it completely. 
how do i know which arduino core version is currently installed ?

did i unterstand it right that  > git checkout <version> will change the IDF to the newer or older commit (version) ?

lbernstone, 2018-05-14:
```
larryb@larryb-EZpad:~/arduino-1.8.5/hardware/espressif/esp32$ git log | grep -i idf | head -1
    Update IDF to f586f5e (#1296)
larryb@larryb-EZpad:~/arduino-1.8.5/hardware/espressif/esp32$ cd ~/esp32/esp-idf/
larryb@larryb-EZpad:~/esp32/esp-idf$ git log | grep f586f5e
commit f586f5e690e687f8db219a8b509bbf483fc416d6
larryb@larryb-EZpad:~/esp32/esp-idf$ git checkout f586f5e690e687f8db219a8b509bbf483fc416d6
M	components/bt/lib
M	components/esp32/lib
HEAD is now at f586f5e6 Merge branch 'feature/get-started-20180130' into 'master'
```

Buffalchill, 2018-05-15:
thank you man! <3

PhilColbert, 2018-05-24:
Hi, this seems to take the menuconfig back to the previous menu system without all the new bluetooth error and memory system... any ideas how to keep the new menus , while using ESPIDF as a component or can we not do that just yet?

lbernstone, 2018-05-24:
sdkconfig is just a text file that defines those menus.  If you diff the sdkconfig.defaults with the sdkconfig in arduino-esp32, it is not hard to pick out what you need to set.

RA5040, 2018-09-04:
Hi, I've followed these instructions carefully (both the 23rd Feb and 14th May posts) and I THINK I've done things correctly.

However I get a build error, as follows:
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp: In member function 'bool fs::SDMMCFS::begin(const char*, bool)':
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:56:26: error: 'sdmmc_host_io_int_enable' was not declared in this scope
         .io_int_enable = sdmmc_host_io_int_enable,
                          ^
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:57:24: error: 'sdmmc_host_io_int_wait' was not declared in this scope
         .io_int_wait = sdmmc_host_io_int_wait,
                        ^
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:59:5: error: 'sdmmc_host_t' has no non-static data member named 'io_int_enable'
     };
     ^
/home/robert/esp/esp-idf/make/component_wrapper.mk:273: recipe for target 'libraries/SD_MMC/src//SD_MMC.o' failed
make[1]: *** [libraries/SD_MMC/src//SD_MMC.o] Error 1
/home/robert/esp/esp-idf/make/project.mk:453: recipe for target 'component-arduino-build' failed
make: *** [component-arduino-build] Error 2

I have deselected "Autostart Arduino setup and loop on boot". My blink.cpp file is down to:

extern "C" void app_main()
#include <stdio.h>
#include "sdkconfig.h"
#include "Arduino.h"

extern "C" void app_main()
{
    initArduino();
}

I'm running on Ubuntu.

lbernstone, 2018-09-04:
There is no need to edit blink.c.  It should definitely not include Arduino.h.  If you will be using the libraries in Arduino IDE, you need to autostart Arduino setup and loop.  Maybe you should explain what you are trying to achieve by making these changes.

RA5040, 2018-09-05:
The reason why I am trying this is that I have a lot of Arduino code that I would like to convert to C++, but I don't have the time to do it manually. So if I could run the code in the way described here: 

> https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md 

it would allow me to do the migration bit by bit.

The document  gives 2 options, both of which have an include "Arduino.h". I've selected the 2nd option, which is to have an app_main, as this is moving in the right C++ direction.

My main program needs to be a .cpp as I will be developing in c++ on Eclipse.

lbernstone, 2018-09-05:
Please open a separate issue instead of hijacking this thread, which is pretty clearly about recompiling the libraries for 80MHz.

mehmetcanbalci, 2018-09-08:
> https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-388913018
What are these steps ? I did not understand anything.

2ni, 2019-04-02:
> 1. Install arduino-esp32 as per instructions (https://github.com/espressif/arduino-esp32)
> 2. Install esp-idf as per instructions (https://github.com/espressif/esp-idf)
> 3. Get esp-idf in line with [...]

Does anyone know how to "align" the esp-idf with platformio to make it work? I'd like to change the config CONFIG_ESP32_RTC_CLOCK_SOURCE_EXTERNAL_CRYSTAL and need to compile a libesp32.a for that matter?
I tried by using .platformio/packages/framework-arduinoespressif32/tools/sdk/sdkconfig, running make in esp-idf with the hello_world example and using the generated libesp32.a in platformio. (Compiling esp-idf asks a lot of questions in the terminal and seems to adapt the sdkconfig)
Unfortunately I get compiling errors when building the arduino example (which works with the origin libesp32.a):
```
.pioenvs/esp32dev/libFrameworkArduino.a(esp32-hal-misc.c.o):(.iram1.12.literal+0x4): undefined reference to `esp_timer_get_time'
.pioenvs/esp32dev/libFrameworkArduino.a(esp32-hal-misc.c.o): In function `millis':
esp32-hal-misc.c:(.iram1.12+0x3): undefined reference to `esp_timer_get_time'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o):(.iram1.23.literal+0x3c): undefined reference to `esp_timer_init'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o):(.iram1.23.literal+0x40): undefined reference to `esp_dbg_stubs_init'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o):(.iram1.23.literal+0x44): undefined reference to `esp_coex_adapter_register'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o): In function `start_cpu0_default':
/www/esp-idf/components/esp32/cpu_start.c:357: undefined reference to `esp_timer_init'
/www/esp-idf/components/esp32/cpu_start.c:357: undefined reference to `esp_dbg_stubs_init'
/www/esp-idf/components/esp32/cpu_start.c:377: undefined reference to `esp_coex_adapter_register'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o):(.iram1.31.literal+0x0): undefined reference to `ets_timer_arm_us'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o):(.literal.timer_setfn_wrapper+0x0): undefined reference to `ets_timer_setfn'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o):(.literal.timer_done_wrapper+0x0): undefined reference to `ets_timer_done'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o):(.iram1.30.literal+0x0): undefined reference to `ets_timer_disarm'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o): In function `timer_arm_us_wrapper':
/www/esp-idf/components/esp32/esp_adapter.c:253: undefined reference to `ets_timer_arm_us'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o): In function `timer_setfn_wrapper':
/www/esp-idf/components/esp32/esp_adapter.c:253: undefined reference to `ets_timer_setfn'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o): In function `timer_done_wrapper':
/www/esp-idf/components/esp32/esp_adapter.c:253: undefined reference to `ets_timer_done'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o): In function `timer_disarm_wrapper':
/www/esp-idf/components/esp32/esp_adapter.c:253: undefined reference to `ets_timer_disarm'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(esp_adapter.o):(.data.g_coex_adapter_funcs+0x4c): undefined reference to `esp_timer_get_time'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(ipc.o): In function `esp_ipc_init':
/www/esp-idf/components/esp32/ipc.c:99: undefined reference to `__stack_chk_fail'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(system_api.o):(.iram1.24.literal+0x2c): undefined reference to `rtc_wdt_flashboot_mode_enable'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(system_api.o): In function `esp_restart_noos':
/www/esp-idf/components/esp32/system_api.c:238: undefined reference to `rtc_wdt_flashboot_mode_enable'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libnewlib.a(time.o):(.literal.get_time_since_boot+0x4): undefined reference to `esp_timer_get_time'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libnewlib.a(time.o): In function `get_time_since_boot':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/time.c:220: undefined reference to `esp_timer_get_time'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libnewlib.a(time.o): In function `esp_set_time_from_rtc':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/time.c:251: undefined reference to `esp_timer_get_time'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(panic.o):(.literal.commonErrorHandler_dump+0x0): undefined reference to `__stack_chk_guard'
[... cut ...]
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp:849: undefined reference to `__stack_chk_fail'
/Users/denis/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libnvs_flash.a(nvs_page.o):/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp:849: more undefined references to `__stack_chk_fail' follow
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32dev/firmware.elf] Error 1
```

Anyone having an idea?

oliverjc, 2019-07-17:
I was having a similar issue and got it working - so I thought I'd share my process which adds to the advice from @lbernstone. I could make this shorter and give the answers, but I think it is important to know the process. 

1. Look through the latest commits of [arduino-esp32](https://github.com/espressif/arduino-esp32/commits/master) to see when it's last been brought up to date with esp-idf. 
2. I see that the latest version of esp-idf to be used is at commit 'a8916daeb'
3. I open a terminal in my esp-idf directory and run: `git log | grep a8916daeb`
4. That gives me the full branch name which I switch to with `git checkout <branchname>`
5. Once that's done 'git describe' will give you the version number. 
6. Go to the docs pages for [xtensa-esp32-elf](https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started/macos-setup.html) toolchain and [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started/index.html#get-esp-idf) and in the bottom left you can select the version of the docs. I selected v3.2.2, got rid of my current toolchain and esp-idf installations and followed the steps for this version and it worked. 

It took me a while so I hope this post saves someone the hassle.

lbernstone, 2019-07-17:
If you want 80MHz nowadays, all you need to do is `setCpuFrequencyMhz(80);`

oliverjc, 2019-07-17:
Maybe I should have specified, but my issue was unrelated to CPU frequency. Your posts helped in fixing some incompatibility issues I was having with Arduino as an ESP-IDF component.

aLoneStrider, 2019-07-19:
@lbernstone , `setCpuFrequencyMhz(80);` returns true but I still got 240 MHz from `ESP.getCpuFreqMhz();` but `getCpuFrequencyMhz()` returns 80 MHz. 
Is the frequency actually changed?

lbernstone, 2019-07-19:
Yes.  That old function should just call the new one.  I'll put in a PR.

ionini, 2019-08-02:
Hi!
Just tried this with the latest version of the idf sdk and it does not work.
I know that I should roll back to a more stable version but the latest idf sdk has a lot of new features and bug fixes, is there any plans to make this compatible with a newer idf version?
Please correct me if I got anything wrong.

atanisoft, 2019-08-02:
@ionini I'm not sure what you are referring to with IDF issues. This issue is not about IDF but an Arduino provided API.

As for IDF 4.0 (aka master of esp-idf), it should mostly work but there are a couple areas which may not. I'd suggest opening a different issue to cover compatibility issues for IDF and not hijack this one for that

closed 2019-08-20 by me-no-dev

tedder, 2019-10-16:
for the record, it's `ESP.getCpuFreqMHz()` .. note the double-caps in MHz.

pedros89, 2021-10-31:
Hello,
Adding Arduino as IDF component in version 4.3.1 did not work and produced the following errors, see image.
If anybody successfully managed to build with Arduino as compont could you point a way.
I need to add custom partitions.csv table. I tried with both docker https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder
and also with this branch 
https://github.com/summivox/arduino-esp32/tree/mod-idf-v4.2

the error produced in 4.3.1 is the following and regards I2C drivers
[image]

If anybody is so kind to point a way without docker I will then create a very crear guide for anybody that examplains how to do it and also how to encrypt esp32 flash with a sketch that includes OTA spiffs and firware so just following the procedure the person can
- add arduino component
- encrypt esp32 flash
- [do OTA update to his firmware stored on their servers](https://github.com/pedros89/ESP32-update-both-OTA-Firmware-and-OTA-SPIFFS-with-Arduino)
- do OTA spiffs to his SPIFFs stored on their server
At the moment the issue is such a mess for a new person approaching it. I commit to put everything in order but I need some guidance for those who already managed to add arduino as component and actually build a sketch, thank you.

lbernstone, 2021-10-31:
Please don't hijack comments that have nothing to do with your issue.  Be brave, open an issue where the title has something to do with your problem.

pedros89, 2021-10-31:
Sorry @lbernstone . I have published here because on the official IDF Arduino components webpage they refer to this very same issue if compilation errors occurs.

https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

kostuch, 2021-11-03:
Hi, any success with arduino-as-component?
I run into this same issue like pedros89, when tried to compile the simplest app from the template (on the PlatformIO with IDF extension, if this makes difference).

BTW: This is strange, that official doc points to unrelevant thread

atanisoft, 2021-11-03:
@pedros89 @kostuch Please open a new issue for your compilation issue or join us over on gitter and we can assist with your compilation issues. More than likely you are ending up with an incompatible combination of ESP-IDF and arduino-esp32 and PlatformIO is complicating matters further as they have yet to support arduino-esp32 2.0.0.

pedros89, 2021-11-03:
Yes, as pointed out by @atanisoft  it is exactely an issue of versions of IDF and Arduino component. 
It was this @lbernstone  post that was really a saver on how to manage to compile successfully for me. 
https://github.com/espressif/arduino-esp32/issues/4702#issuecomment-955787335

Please @kostuch  refer to this detailed post to compile successfully. I didn't publish on this thread so it more pertinent. 
https://github.com/espressif/arduino-esp32/issues/4702#issuecomment-960296078

CodeOn-ArK, 2022-05-04:
Hi!
I a using arduino as an ESP-IDF component
I am using IDF v4.4 and arduino-esp32 master branch. I have added arduinoJSON and MQTTPubSubCLient as IDF components.
It works till then.
Now when I try to use default arduino libs such as BLE or WiFiSecureClient I get errors as mentioned in the issue : https://github.com/espressif/arduino-esp32/issues/6680
So to use the default arduino libs do we have to so something extra?

me-no-dev, 2022-05-04:
Have you enabled BT/BLE and MbedTLS in menuconfig?

CodeOn-ArK, 2022-05-08:
I solved the WiFiSecureClient by enabling MbedTLS in menuconfig.
Wdym by enabling BT/BLE? where to enable them? Aren't they enabled by default?

marnov999, 2023-01-03:
So i have managed to rebuild the arduino componend in esp-idf, then i copyied "libfreertos.a" to "Arduino15\packages\esp32\hardware\esp32\2.0.5\tools\sdk\esp32s3\qio_opi" to test everything, and it was ok, but why i did this was because i wanted to use vTaskList(for debbuging, and learning), so i have went back, and added this to sdkconfig: "CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y
CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER=y" but after i rebuild again, and replaced the file in arduino folder, and uploaded my sketch, i got this error: 
"\0\x80ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037def5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
[   103][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface CDC enabled
[   104][D][esp32-hal-tinyusb.c:562] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace:0x403778fe:0x3fce18f00x4037aa7d:0x3fce1910 0x40380c85:0x3fce1930 0x4037de0d:0x3fce1a60 0x4037b756:0x3fce1aa0 0x420192f8:0x3fce1ae0 0x4200ae99:0x3fce1b30 

ELF file SHA256: 0000000000000000

Rebooting..."
andbody know what "assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)" means? 
(all check and esp-idf is in line with arduino-esp32) and i can use vTaskList in esp idf without problem.

dagi0627, 2023-11-24:
What is this error ``` /home/user/test/ESP32/esp-idf/led_blinking/components/arduino/cores/esp32/esp32-hal-rmt.c: In function 'rmtInit':
/home/usertest/ESP32/esp-idf/led_blinking/components/arduino/cores/esp32/esp32-hal-rmt.c:516:11: error: 'rmt_tx_channel_config_t' has no member named 'intr_priority'
  516 |     tx_cfg.intr_priority = 0;
      |           ^
/home/usertest/ESP32/esp-idf/led_blinking/components/arduino/cores/esp32/esp32-hal-rmt.c:541:11: error: 'rmt_rx_channel_config_t' has no member named 'intr_priority'
  541 |     rx_cfg.intr_priority = 0;
      |           ^
[0/1] Re-running CMake...-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/enhee/test/ESP32/esp-idf/led_blinking/sdkconfig ```

me-no-dev, 2023-11-24:
@dagi0627 that you are using older esp-idf. Use 5.1.2