espressif/arduino-esp32 #1143: Installation failling on macos

cacaodev, 2018-02-22:
I followed the instructions in docs/arduino-ide/mac.md

Having macos 10.10 and python 2.7.10 installed, I get the following error:
```
System: Darwin, Info: Darwin-15.6.0-x86_64-i386-64bit
Platform: x86_64-apple-darwin
Tool xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz already downloaded
Extracting xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
Tool esptool-4dab24e-macos.tar.gz already downloaded
Extracting esptool-4dab24e-macos.tar.gz
Downloading mkspiffs-0.2.1-osx.tar.gz
Traceback (most recent call last):
  File "get.py", line 139, in <module>
    get_tool(tool)
  File "get.py", line 94, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1049, in endheaders
    self._send_output(message_body)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 893, in _send_output
    self.send(msg)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 855, in send
    self.connect()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1274, in connect
    server_hostname=server_hostname)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 352, in wrap_socket
    _context=self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 579, in __init__
    self.do_handshake()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 808, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
```

wiiikiii, 2018-02-24:
the same problem here.
install brew and openssl. follow instruction adding path in .bash_profile
later install python and run the installed version

```
imac4-2:tools$ python2.7 get.py
System: Darwin, Info: Darwin-16.7.0-x86_64-i386-64bit
Platform: x86_64-apple-darwin
Tool xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz already downloaded
Extracting xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
Tool esptool-4dab24e-macos.tar.gz already downloaded
Extracting esptool-4dab24e-macos.tar.gz
Downloading mkspiffs-0.2.1-osx.tar.gz
Done
Extracting mkspiffs-0.2.1-osx.tar.gz
Renaming mkspiffs-0.2.1-osx to mkspiffs
Done
```

renegadeandy, 2018-02-28:
That didnt fix it for me. In the end I manually downloaded mkspiffs leaving the zip in the same directory as the others and reran.

powersoft, 2018-03-02:
Thanks have the same problem, but where to store mkspiffs?
Have extracted the tar.gz

augustj, 2018-03-03:
@powersoft 
Manually download the file, then move it to the `Arduino/hardware/espressif/esp32/tools/dist` directory. You should see other archives in that folder. Don't manually extract the file - just rerun the script.

tavdog, 2018-03-07:
I too had this problem.  Manually downloaded the archive and the get.py script finishes but the new board does not show up in Arduino.

makefitter, 2018-03-18:
After almost a day of dealing here is the solution;

[https://stackoverflow.com/a/46308535](url)

In this solution python and openssl are reinstalled and then linked. 

I tried it for mac OS High Sierra, python 2.7.14 and openssl 1.0.2n.

It's nice to get Blink example to compile :)

tferrin, 2018-04-16:
Manually fetching mkspiffs-0.2.2-arduino-esp32-osx.tar.gz into hardware/espressif/esp32/tools/dist and then re-running get.py worked for me.

Note that the root cause of the problem here is that the python included with OSX versions 10.12 (Sierra) and earlier only supports OpenSSL version 0.9.8, and that only supports TLS 1.1.  Recently github moved to requiring TLS 1.2 or later, and so get.py fails when fetching mkspiffs. It's okay fetching one or two packages manually, but over the long term I expect more and more sites will require TLS 1.2, so manual fetches will become more tedious and sooner or later OSX users will either need to upgrade to High Sierra (which does support a newer OpenSSL) or install a new python and openssl 1.0.2.

closed 2019-08-15 by stale[bot]

ivancerrato, 2020-02-14:
it works with the latest driver version:
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
and 
Arduino IDE 1.8.12 - macOS 10.15.3 

crw-rw-rw-   1 root  wheel           18,   7 Feb 14 08:02 cu.SLAB_USBtoUART