espressif/arduino-esp32 #1145: HardwareSerial end() / begin() does not appear to cleanly reinitialize port

rayozzie, 2018-02-23:
### Hardware:
Board: Adafruit Huzzah32 Feather
Core Installation/update date: 19-Feb-2018
IDE name: Arduino 1.8.5
Flash Frequency: 80MHz
Upload Speed: 921600

### Description:
I have been fortunate thus far that everything I've tried has functioned flawlessly, so thank you very much for that.

On the Adafruit module, I am using the UART (pins 16/17) on Serial1 via:
HardwareSerial Serial1(2);

I am initializing it to 9600bps, and all is good.  It works well.

However, I actually have a GPIO-driven 4-port UART multiplexer connected to those RX/TX pins.  I have used this mux design for years, and the general algorithm when switching is:
- Serial1.end() to deinit the driver
- digitalWrite to two GPIO pins to select the desired port
- Serial1.begin(newSpeed) to reinit the driver

At the current time, all devices connected to all ports just happen to be operating at 9600bps, so this switchover should be occurring in a flawless manner without any framing errors or any issues at all.

However, there appears to be a bug somewhere in the end()/begin() code path that is causing extremely bad behavior.  The way that it manifests itself is that after the end/begin reinit,

a) a fragment of the tail end of the data that had been previously successfully transmitted prior to the end() is spuriously transmitted redundantly after the begin()

b) buffer pointers appear to be getting somewhat confused, because after the reinit certain sequences of bytes are transmitted out-of-order - i.e. the string:
{"req":"time"}
...is commonly transmitted as...
e"}{"req:"tim

c) received data is wrong or reordered in a way that I have not yet been able to determine.

After many hours of debugging, I finally tried simply removing the calls to Serial1.end() and Serial1.begin(), knowing that (for now) my devices are all operating at 9600bps.  After doing this, I am having zero bugs or issues - which is why I'm filing this issue.

I regret that I am on my way to the airport and haven't had a chance to create a repro sketch for you, but I felt it best to file this while it is fresh in my mind rather than to wait two weeks.  Perhaps whoever is familiar with the HardwareSerial driver will recognize the failure to fully clear and reinitialize the driver context on begin().

Good hunting, and thanks again.

lbernstone, 2018-02-23:
I don't know anything about the internal workings of the esp-hal-uart, but for troubleshooting and perhaps a workaround, have you tried putting a flush in between the end and begin?

rayozzie, 2018-02-23:
I did, and it caused incredibly unstable behavior.  The flush call caused transmission of a large amount of garbage – consistent with the theory that the state of the buffer pointers is likely bad.

lbernstone, 2018-02-23:
Do you get the same behavior on HardwareSerial(1)?  From the esp32 technical manual:

Three UART controllers share a 1024 x 8-bit RAM space. As illustrated in Figure 79, RAM is allocated in different
blocks. One block holds 128 x 8-bit data. Figure 79 illustrates the default RAM allocated to Tx_FIFO and
Rx_FIFO of the three UART controllers. Tx_FIFO of UARTn can be extended by setting UARTn_TX_SIZE, while
Rx_FIFO of UARTn can be extended by setting UARTn_RX_SIZE.
NOTICE: Extending the FIFO space of a UART controller may take up the FIFO space of another UART
controller.
If none of the UART controllers is active, setting UART_MEM_PD, UART1_MEM_PD, and UART2_MEM_PD can
prompt the RAM to enter low-power mode.
In UART0, bit UART_TXFIFO_RST and bit UART_RXFIFO_RST can be set to reset Tx_FIFO or Rx_FIFO,
respectively. In UART1, bit UART1_TXFIFO_RST and bit UART1_RXFIFO_RST can be set to reset Tx_FIFO or
Rx_FIFO, respectively.
Note:
UART2 doesn’t have any register to reset Tx_FIFO or Rx_FIFO, and the UART1_TXFIFO_RST and UART1_RXFIFO_RST
in UART1 may impact the functioning of UART2. Therefore, these 2 registers in UART1 should only be used when the
Tx_FIFO and Rx_FIFO in UART2 do not have any data.

rayozzie, 2018-02-23:
Unfortunately, it’s not possible for me to try UART1 because the pins aren’t exposed on the Arduino Huzzah32.  Not using it.

I am using UART0 for output only because, in the Arduino IDE it’s wired to be used for the debug console.

lbernstone, 2018-02-23:
You should be able to map that to whatever pins you like:
```arduino
HardwareSerial Serial1(1);
Serial1.begin(9600, SERIAL_8N1, 16, 17);
```

negativekelvin, 2018-02-24:
Arduino has not fixed uart fifo reset bug that was fixed in idf

rayozzie, 2018-02-25:
This does feel somewhat similar to these:

https://github.com/espressif/esp-idf/issues/1202

https://github.com/espressif/arduino-esp32/issues/991

I am not familiar with the engineering processes and practices. If indeed there were fixes that resolved these bugs, and these or any other fixes have not made it into the Arduino driver, where does an issue need to be submitted to get some priority on resolution and delivery in that code base?

Thx

JacoFourie, 2018-02-27:
If I put a Flash before sending I am able to use all 3 the UARTS without a problem.

````
Serial1.flush();
````

rayozzie, 2018-02-27:
You are suggesting this as a workaround for the end() followed by a begin() to reinitialize ports?

I never even considered a flush before every write(), however I did try a flush() before the end() and after the begin(), and got quite unstable results.

JacoFourie, 2018-02-27:
I am using all UARTs all the time. Before I do a send on UART1 I do a flush.
UART 0 and 2 works fine. for me.

````
MyModbusSerial.flush();
MyModbusSerial.println("Testing");
````

rayozzie, 2018-02-27:
To be clear, the problem I am encountering has nothing at all to do with uarts or flush. They are all working.

The problem is that when a working Uart is uninitialized with end() and then reinitialized again with begin(), it puts the Uart into a messed up state in which there is misordering and reappearance of previously buffered contents.

JacoFourie, 2018-02-27:
OK. Why do you need to do that ? If they are working then let them work.

rayozzie, 2018-02-27:
My device has a UART multiplexer attached to RX/TX on one of the uarts (with a GPIO-driven selector), that fans out to four different serial devices. Each of the four can potentially be at a different baud rate, necessitating an end()/begin() to change the baud rare each time it is switched.

tgmaxx, 2018-03-13:
ESP32 was "released" 1-1/2 years ago, and the serial ports are still a complete mess.

johnty, 2018-05-15:
I've stumbled across this issue as well. The most recent [PR1408](https://github.com/espressif/arduino-esp32/pull/1408) appears to address this and applies the same software fix to the uartFlush() function as whats done in IDF. However, having tried the solution myself, I'm still seeing *occasional* bad behaviour of UART2 after reinit.

In my application I need to change baudrates so I perform a Serial1.end() and then begin() at the new rate. If I do this, the reads afterwards will *sometimes* go out of whack. I can see partially "correct" data in the stream but they can appear in the wrong position, along with invalid bytes. Its also obvious these bytes were seen previously in the buffer based on their values. Scoping the actual stream shows the sender is acting correctly.

If I don't perform this baudrate change, then receive buffer seems to work fine without any issues.

robdobsn, 2018-06-10:
I have a problem with Serial2 (running at a fixed 9600 baud) which seems to involve ring buffer errors.  My test code is as follows:

```
#include <Arduino.h>
#include <HardwareSerial.h>

HardwareSerial serialPort2(2);
unsigned long txTime = millis();

const uint8_t GETUIDCMD[] = { 0x02, 0x08, 0x00, 0x55, 0x2b, 0x01, 0x01, 0x00, 0x01, 0x00, 0x01, 0xf0, 0xc6};

void setup() {
    Serial.begin(115200);
    serialPort2.begin(9600);
    Serial.println("TestESP32HardwareSerial");
}

void loop() {
    if (serialPort2.available() > 0)
    {
        char tmpBuf[5];
        sprintf(tmpBuf, "%02X ", serialPort2.read());
        Serial.print(tmpBuf);
    }

    if ((millis() > txTime + 500) || (txTime > millis()))
    {
        Serial.println("Sending GetUID command");
        serialPort2.write(GETUIDCMD, sizeof(GETUIDCMD));
        txTime = millis();
    }
}
```
So it sends a sequence of bytes every 500ms on Serial2 (I'm trying to talk to an RFID chip called Eccel Tech B1 which should respond to the command with two packets of data).  The response that should come back looks like this:

02 03 00 AF F7 00 F0 E1 02 04 00 38 6E 08 20 C4 B0

There are two messages here, both starting with STX (0x02)

After a while (maybe an hour or two) it changes to something like:

20 C4 B0 02 03 00 AF F7 00 F0 E1 02 04 00 38 6E 08

This is the right data but the two messages are now conflated with the end of a previous one and the end of the second message is missing. 

This looks to me like some kind of circular buffer problem but I guess it could be due to a number of things.

To be sure that it is the ESP32 that is getting things wrong I've used a logic analyzer (Saleae Logic 16) to monitor the serial data received. It shows the following:

[image]

Which seems to clearly be correct even while the ESP32 is showing the wrong data.

lbernstone, 2018-06-10:
@robdobsn : Can you test this with a minimal esp-idf example (along the lines of https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart_async_rxtxtasks).  I think you will get a lot more traction with this issue if you can demonstrate whether the bug is in arduino-esp32 or the underlying esp-idf.

hex86, 2018-06-28:
Not sure if this is related but I recently upgraded from an older (late 2017/early 2018) ESP32 Arduino package to the current as of today and noticed a lot of my serial code was buggy/unstable using all 3 UARTs. Previously they all seemed to function properly but post upgrade UART2 wouldn't work whenever USB was plugged into the PC on an Adafruit Feather32 but when I disconnected USB so it was only on battery power UART2 would immediately start functioning. No reboot/nothing, simply disconnect USB and UART2 immediately worked. When going back to the older package everything resumed proper function.

closed 2019-08-15 by stale[bot]