espressif/arduino-esp32 #1146: WifiClientSecure missing peek() method!

chaeron, 2018-02-23:
P
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:				        40Mhz
Upload Speed:						115200

### Description:
The WifiClientSecure class is missing the peek() method, so the peek() from the parent class (WifiClient) is called instead.

The problem with this is that if you are using WifiClientSecure with the Update library to do OTA updates it will fail with a "Invalid Magic Byte" error when Update tries to read the stream and do a peek() at the first magic byte.  Likely because the byte isn't being decrypted.

This means that you currently cannot do an OTA update over HTTPS/SSL. :-(

brokentoaster, 2018-03-14:
`UpdateClass::writeStream()` could be modified to call `_verifyHeader()` inside the  `while(remaining())` loop. Making the call to peek unnecessary.

brokentoaster, 2018-03-14:
The `ESP_IMAGE_HEADER_MAGIC` is also checked in `UpdateClass::_writeBuffer()` when the buffer is committed to flash, I think the earlier checks are pointless as no changes will be committed to the flash unless that byte was read correctly.

brokentoaster, 2018-03-14:
try #1214

copercini, 2018-03-18:
Hi guys, I've implemented something here: https://github.com/copercini/arduino-esp32/tree/master/libraries/WiFiClientSecure/src but I don't have a server with SSL for test if it's working with the update library

copercini, 2018-04-14:
Now it has =) https://github.com/espressif/arduino-esp32/pull/1310

closed 2018-04-14 by copercini