espressif/arduino-esp32 #1147: wifi connection Problem

stickbreaker, 2018-02-24:
### Hardware:
Board:							WeMos bluetooth & battery
Core Installation/update date:			28/nov/2017
IDE name:						Arduino IDE 1.8/5
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Could not connect to WiFi router. Debug level set to ERROR, The sketch I was testing uses: 
```
#include <WiFi.h>
#include <PubSubClient.h>
#include <OneWire.h>
#include <SensorAverage.h>
#include "mqtt.h"
#include <ESPmDNS.h>
```
This sketch has be working unchanged for 2 months. 

### Sketch: Just the WiFi reconnect/connect code.
```cpp
void setup_wifi() {
  uint16_t count=0;
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  if(!sta_was_connected) WiFi.begin(ssid, password); //sta_was_connected is global, init'd to false.
  else WiFi.reconnect();
  while ((WiFi.status() != WL_CONNECTED)&&(count<30)) {
    delay(500);
    count++;
    if(Serial.available()){
      Serial.printf("WiFi.status=%d\n",WiFi.status());
      Serial.read();
      }
    else Serial.print(".");
    }
  sta_was_connected = WiFi.status()==WL_CONNECTED;
  randomSeed(micros());
  if(sta_was_connected){
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    
    if (!MDNS.begin("ESP32_sensor")) {
      Serial.println("Error setting up MDNS responder!");
      while(1){
        delay(1000);
        }
      }

    }
  else {
    Serial.println("");
    Serial.printf("WiFi Connection Failed, status=%d",WiFi.status());
    }
}
```
Function always failing with WiFi.status ==  WL_NO_SSID_AVAIL    = 1.

Uploaded WifiScan example. with Debug Level set to DEBUG, Scan reported 4 visible WiFi networks:
```
D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
Setup done
scan start
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 1 - SCAN_DONE
scan done
4 networks found
1: Belkin.464C (-79)*
2: todd (-86)*
3: Guest5323 (-89)*
4: TP-LINK_D2A4 (-89)*
```
Re upload original sketch, with Debug set to DEBUG Now it works?
```
Connecting to Belkin.464C
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 7 - STA_GOT_IP
.
WiFi connected
IP address: 
192.168.2.19
Scanning oneWire
os(0x3ffdbe94) id=0x28 0xff 0xb8 0x68 0x00 0x17 0x03 0xd5 , pin=17, to=0
os(0x3ffdbea4) id=0x28 0xff 0x8e 0x38 0x00 0x17 0x05 0xc8 , pin=17, to=0
Attempting MQTT connection to GX270 ...connected
8240 temp(0) 16.18 1618
8240 temp(1) 19.81 1981
```
re Upload sketch with Debug set to ERROR:
```
Connecting to Belkin.464C
..
WiFi connected
IP address: 
192.168.2.19
Scanning oneWire
os(0x3ffdbd28) id=0x28 0xff 0xb8 0x68 0x00 0x17 0x03 0xd5 , pin=17, to=0
os(0x3ffdbd38) id=0x28 0xff 0x8e 0x38 0x00 0x17 0x05 0xc8 , pin=17, to=0
Attempting MQTT connection to GX270 ...connected
8184 temp(0) 16.18 1618
8184 temp(1) 19.75 1975
```

Now it is correctly working?  What did uploading WifiScan fix?  Or was it changing the Debug level up then down?

I can't replicated the problem now.  I had not been able to get this Mqtt sketch to connect for 4 days?  I Upload it at the end of my I2C work day as a play toy for my Node-RED server.

Does anyone know what went wrong? or how to reset the WiFi stuff.  I tried power cycling, manual reset, loading other scripts.

Chuck.

stickbreaker, 2018-02-24:
@lbernstone If I can get it to fail again, I'll try.
What do I actually need to actually code?  Doing a quick google search show me that NVS is part of IDF.  I looked through readthedocs but could not find a NVS_list().  
```c++
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"

void app_main()
{
    // Initialize NVS
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );
```

Chuck.

lbernstone, 2018-02-25:
They seem to have done some security through obscurity on the nvs.  I think the "secret" keys and namespace for wifi will need to be provided by someone from Espressif (@igrr ).  There is a wrapper for the nvs functions in arduino, so you can use them right in arduino code.  The following will wipe all the nvs keys:
```
#include <nvs.h>
#include <nvs_flash.h>
void setup() {
    int err;
    err=nvs_flash_init();
    Serial.println("nvs_flash_init: " + err);
    err=nvs_flash_erase();
    Serial.println("nvs_flash_erase: " + err);
 }
```

lbernstone, 2018-02-25:
Hmm, a little googling came up with this: https://github.com/Edzelf/ESP32-Show_nvs_keys

stickbreaker, 2018-02-25:
@lbernstone I'll play around with ESP32-Show_nvs_keys, I'll have to mod it. As written, it does not show all namespaces, only it's ` namespace_ID = FindNsID ( nvs, "ESP32Radio" ) ; ` ESP32Radio.  Shouldn't be too hard to remove the conditional.  and just directly scan it.

Chuck.

lbernstone, 2018-02-25:
I'm playing with it too, but it sure looks like it is showing all the namespaces.  I don't have a ESP32Radio namespace on my device, and it is showing a whole lot of keys.

stickbreaker, 2018-02-25:
@lbernstone I'm part way through rewriting that example `show_nvs_keys`.
```

D: Partition nvs found, 20480 bytes
NameSpaces existing in NVS
            misc = 1
    nvs.net80211 = 2
             phy = 3

NameSpace:             misc = 1
001: 1 type:BLOB key=log data=size:0010
02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
056: 1 type:072 key=HH	H
HHH
HHHHHHH8H9H:H;H<H=H data=00000000481a4819
061: 1 type:084 key=XXXX data=000000005c055c04
021: 1 type:000 key=⸮@>`@0 data=0000000031313330
002: 2 type:000 key= data=00000000ffff0010

NameSpace:     nvs.net80211 = 2
005: 2 type:U8   key=country data=0
011: 2 type:U8   key=sta.authmode data=1
016: 2 type:BLOB key=sta.pmk data=size:0020
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
019: 2 type:U8   key=auto.conn data=1
023: 2 type:U8   key=sta.phym data=3
024: 2 type:U8   key=sta.phybw data=2
050: 2 type:BLOB key=ap.ssid data=size:0024
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff                                      ....            
055: 2 type:BLOB key=ap.passwd data=size:0041
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff                                               .               
059: 2 type:BLOB key=ap.pmk data=size:0020
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
061: 2 type:U8   key=ap.chan data=1
062: 2 type:U8   key=ap.authmode data=0
063: 2 type:U8   key=ap.hidden data=0
064: 2 type:U8   key=ap.max.conn data=4
065: 2 type:U16  key=bcn.interval data=100
066: 2 type:U8   key=ap.phym data=3
067: 2 type:U8   key=ap.phybw data=2
068: 2 type:U8   key=ap.sndchan data=1
005: 2 type:BLOB key=sta.ssid data=size:0024
0b 00 00 00 42 65 6c 6b 69 6e 2e 34 36 34 43 00  ....Belkin.464C.
a5 a5 a5 a5 a5 a5 a5 a5 d0 13 08 40 3e 60 08 40  ...........@>`.@
30 05 06 00                                      0...            
008: 2 type:BLOB key=sta.pswd data=size:0041
35 34 41 39 30 33 31 31 00 00 00 00 00 00 00 00  54A90311........
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00                                               .               
012: 2 type:U8   key=bssid.set data=0
013: 2 type:U8   key=sta.scan_method data=0
014: 2 type:U8   key=sta.sort_method data=0
015: 2 type:I8   key=sta.minrssi data=129
016: 2 type:U8   key=sta.minauth data=0
017: 2 type:BLOB key=sta.apsw data=size:0002
05 00                                            ..              
120: 2 type:U8   key=sta.chan data=11
058: 2 type:000 key=	
 data=0000000030013000
000: 2 type:BLOB key=sta.bssid data=size:0006
[... 156 lines cut ...]
10 01 ea 00 13 01 e7 00 13 01 e7 00 13 01 e7 00  ................
13 01 e7 00 10 01 e7 00 10 01 e7 00 10 01 e7 00  ................
10 01 e3 00 10 01 dd 00 10 01 d4 00 06 01 c4 00  ................
06 01 c4 00 0a 01 c4 00 0a 01 c4 00 0a 01 c4 00  ................
03 01 c4 00 06 01 c0 00 0a 01 c0 00 0e 01 c0 00  ................
14 01 c0 00 18 01 c0 00 06 01 bc 00 0a 01 bc 00  ................
17 01 bc 00 c4 00 ef 00 c2 00 ec 00 c4 00 ec 00  ................
c4 00 ec 00 c4 00 ec 00 c7 00 ec 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

still kinda messy

Chuck.

lbernstone, 2018-02-25:
```
#include <nvs_flash.h>
#include <nvs.h>
#include <WiFi.h>

bool wifiReset = false;

void IRAM_ATTR set_ps() {
  wifiReset = true;
}

void isr_pin0() {
// Puts an interrupt on pin 0 (the top button on Heltec boards)
  pinMode (0, INPUT);
  attachInterrupt(digitalPinToInterrupt(0), set_ps, RISING);
}

void resetStaInfo() {
    esp_err_t err;
    nvs_handle nvs_mqtt;
    
    Serial.println(F("Wifi reset"));
    err = nvs_flash_init();
    if (err) Serial.println("nvs_flash_init: " + err);
    err = nvs_open("nvs.net80211", NVS_READWRITE, &nvs_mqtt);    
    if (err) Serial.println("nvs_open: " + err);
    err = nvs_set_str(nvs_mqtt, "sta.ssid","");
    if (err) Serial.println("nvs_erase_ssid: " + err);
    err = nvs_set_str(nvs_mqtt, "sta.pswd","");
    if (err) Serial.println("nvs_erase_pswd: " + err);
    ESP.restart();
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  isr_pin0();
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  delay(200);
  Serial.println(WiFi.localIP());
}

void loop() {
  if (wifiReset) resetStaInfo();
  delay(100);
}
```

stickbreaker, 2018-02-25:
@lbernstone You are correct to blame NVS.  
I have spent the day working through validating my NVS partition,  It is hosed!
```
Partition nvs found, 20480 bytes
page[0].State=0xfffffffc INIT FULL  means FULL Seqnr=0
page[1].State=0xfffffffe INIT  means ACTIVE Sequence value InValid: page[1].Seqnr=5663 
page[2].State=0xffffffff  means UNINITIALIZED  Page[2] not active, State = 0xffffffff Seqnr=4294967295 
page[3].State=0xfffffffc INIT FULL  means FULL Sequence value InValid: page[3].Seqnr=183 
page[4].State=0xfffffffc INIT FULL  means FULL Sequence value InValid: page[4].Seqnr=5662 
```
The sequence number for the nvs pages are corrupt. The sequence values must fit in the nvs Partions (0..4).  One page must be left blank to all copying on reuse/modification. It looks to me that page[2] is the current blank page.  I manually created an inuse/translation index:
```
Page 0 -> 0
Page 1 -> 1
Page 2 -> 3
Page 3 -> 4
Page 4 -> 255
```
Logical page 4 is marked as inactive. Logical page 2 points to sector 3.
Here is what I read with that mapping:
```

NameSpace:            misc 
0-001:            log type:BLOB data=size:0010
02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

NameSpace:    nvs.net80211 
0-005:        country type:U8   data=0
0-011:   sta.authmode type:U8   data=1
0-016:        sta.pmk type:BLOB data=size:0020
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0-019:      auto.conn type:U8   data=1
0-023:       sta.phym type:U8   data=3
0-024:      sta.phybw type:U8   data=2
0-050:        ap.ssid type:BLOB data=size:0024
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff                                      ....            
0-055:      ap.passwd type:BLOB data=size:0041
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff                                               .               
0-059:         ap.pmk type:BLOB data=size:0020
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0-061:        ap.chan type:U8   data=1
0-062:    ap.authmode type:U8   data=0
0-063:      ap.hidden type:U8   data=0
0-064:    ap.max.conn type:U8   data=4
0-065:   bcn.interval type:U16  data=100
0-066:        ap.phym type:U8   data=3
0-067:       ap.phybw type:U8   data=2
0-068:     ap.sndchan type:U8   data=1
1-034:      sta.bssid type:BLOB data=size:0006
34 fc 3f 74 35 fc                                4.?t5.          
1-067:       sta.ssid type:BLOB data=size:0024
0b 00 00 00 42 65 6c 6b 69 6e 2e 34 36 34 43 00  ....Belkin.464C.
01 00 00 00 0a 00 00 00 d0 13 08 40 3e 60 08 40  ...........@>`.@
30 04 06 00                                      0...            
1-070:       sta.pswd type:BLOB data=size:0041
35 34 41 39 30 33 31 31 00 00 00 00 00 00 00 00  54A90311........
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00                                               .               
1-074:      bssid.set type:U8   data=0
1-075:sta.scan_method type:U8   data=0
[... 610 lines cut ...]
    }
 
 i=0;
 uint8_t j=0;
  while(i<255){
    if(i<NA->count){
      Serial.printf("\nNameSpace: %15s \n",NA->ns[i].name);
      j=NA->ns[i].nsID;
      }
    else {
      j = i+1;
      Serial.printf("\nOrphan UnNamed : %u \n",j);
      }
    listKeys(nvs,j);
    i++;
    }
    
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Chuck.

stickbreaker, 2018-02-25:
@lbernstone 
I just init'd nvs with your code:
```c++
  int err;
  err=nvs_flash_init();
  Serial.println("nvs_flash_init: " + err);
  err=nvs_flash_erase();
  Serial.println("nvs_flash_erase: " + err);
```
I then reran [Show_nvs_keys](https://github.com/stickbreaker/ESP32-Show_nvs_keys).  It correctly reported an empty nvs.
```
Partition nvs found, 20480 bytes
page[0].State=0xffffffff  means UNINITIALIZED  Page[0] not active, State = 0xffffffff Seqnr=4294967295 
page[1].State=0xffffffff  means UNINITIALIZED  Page[1] not active, State = 0xffffffff Seqnr=4294967295 
page[2].State=0xffffffff  means UNINITIALIZED  Page[2] not active, State = 0xffffffff Seqnr=4294967295 
page[3].State=0xffffffff  means UNINITIALIZED  Page[3] not active, State = 0xffffffff Seqnr=4294967295 
page[4].State=0xffffffff  means UNINITIALIZED  Page[4] not active, State = 0xffffffff Seqnr=4294967295 

Logical Page to Physical Sector
Page 0 -> 255
Page 1 -> 255
Page 2 -> 255
Page 3 -> 255
Page 4 -> 255
```
then uploaded my mqtt sketch, which connected and reloaded nvs, After rerunning [Show_nvs_keys](https://github.com/stickbreaker/ESP32-Show_nvs_keys)
everything looks good!  Thanks!
```
Partition nvs found, 20480 bytes
page[0].State=0xfffffffc INIT FULL  means FULL  Seqnr=0 
page[1].State=0xfffffffc INIT FULL  means FULL  Seqnr=1 
page[2].State=0xfffffffe INIT  means ACTIVE  Seqnr=2 
page[3].State=0xffffffff  means UNINITIALIZED  Page[3] not active, State = 0xffffffff Seqnr=4294967295 
page[4].State=0xffffffff  means UNINITIALIZED  Page[4] not active, State = 0xffffffff Seqnr=4294967295 

Logical Page to Physical Sector
Page 0 -> 0
Page 1 -> 1
Page 2 -> 2
Page 3 -> 255
Page 4 -> 255

NAME SPACES
           misc = 1
   nvs.net80211 = 2
            phy = 3

NameSpace:            misc = 1

NameSpace:    nvs.net80211 = 2

NameSpace:             phy = 3

NameSpace:            misc 
0-001:            log type:BLOB data=size:0010
02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

NameSpace:    nvs.net80211 
0-010:   sta.authmode type:U8   data=1
0-015:        sta.pmk type:BLOB data=size:0020
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0-018:      auto.conn type:U8   data=1
0-020:      sta.bssid type:BLOB data=size:0006
ff ff ff ff ff ff                                ......          
0-022:       sta.phym type:U8   data=3
0-023:      sta.phybw type:U8   data=2
0-053:        ap.ssid type:BLOB data=size:0024
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[... 192 lines cut ...]
15 01 ea 00 12 01 ea 00 12 01 e6 00 12 01 e6 00  ................
12 01 e3 00 12 01 dc 00 12 01 d2 00 07 01 c1 00  ................
07 01 c1 00 0d 01 c1 00 0d 01 c1 00 0d 01 be 00  ................
06 01 be 00 06 01 be 00 0b 01 be 00 0f 01 be 00  ................
18 01 be 00 1c 01 be 00 08 01 bb 00 0c 01 bb 00  ................
1a 01 bb 00 bf 00 ef 00 c2 00 ec 00 c2 00 ec 00  ................
c2 00 ec 00 c4 00 ec 00 c4 00 ec 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

```

Thanks for the Help.

Chuck.

closed 2018-02-25 by stickbreaker

beegee-tokyo, 2018-02-26:
Added to Wiki [WiFi connection problem – NVS corrupted](http://desire.giesecke.tk/index.php/2018/02/26/wifi-connection-problem-nvs-corrupted/)