espressif/arduino-esp32 #1149: SD memory leak

juanfrito, 2018-02-25:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I've written a data logger for the ESP32 that eventually crashes and restarts. I was able to determine that it was due to a memory leak. 

Every time SD.begin is used, heap memory decreases 20 bytes (1st time 120 bytes). Every minute I write some data to a log file, but I want to be able to extract or insert an SD card in between, so I do an SD.begin() and SD.end() every two seconds, which causes the ESP32 to run out of heap space after some two hours.

### Sketch:
The sketch for the data logger is quite long, so as a demonstration I include here just the simplest one to show the leak
```
#include <SD.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println(ESP.getFreeHeap());
  SD.begin();
  SD.end();
  delay(1000);
}
```

### Serial Monitor Output:
```
222768
222648
222628
222608
222588
222568
222548

```

cybercatnet, 2019-06-17:
same problem

lbernstone, 2019-06-17:
Looks to me like https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L669 allocates memory for card every time begin runs, but it is never freed.  Unfortunately, my C skills are not good enough to figure out how to release it without hammering s_cards[pdrv].

atanisoft, 2019-06-17:
@lbernstone at this line https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L686 check for the car being null or not. If not null free it before assignment. That could work to avoid leaking memory, but it would be better for end() to call https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L645 as well.

cybercatnet, 2019-06-17:
The function sdcard_uninit() has a free of card pointer.
https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/libraries/SD/src/sd_diskio.cpp#L657

SD.end() call sdcard_uninit().
https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/libraries/SD/src/SD.cpp#L55

closed 2019-08-30 by stale[bot]

cybercatnet, 2019-08-31:
The problem is not fixed yet