espressif/arduino-esp32 #1150: Question - Is it possible to use BLE and Bluetooth at the same time?

beegee-tokyo, 2018-02-26:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		Espressif32 (Stage) Version 29b3a81
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I want to use BLE and Bluetooth together in one project. But when I initialize both, depending on the initialization order, either none of them works and my app just hangs or BLE works and BTSerial doesn't.    
 
Is it possible to use BLE and Bluetooth at the same time? In the example I am using BluetoothSerial, which is not yet merged officially, but it seems to work. My problem could be related to this as well, I am not sure.    

Both BLE and BluetoothSerial work when used alone, only when I try to initialize both the code just hangs or only BLE works. Serial debug output below shows that when I try to initialize BluetoothSerial first, the first debug message comes from BLE????? If I initialize BLE first, BLE works, but BluetoothSerial fails to initialize with 
`[E][BluetoothSerial.cpp:108] _init_bt(): _init_bt initialize controller failed`

Anybody got BLE and Bluetooth working side by side yet? Or am I just running out of memory here, because at the end of second debug log I can see
```
[E][WString.cpp:185] changeBuffer(): realloc failed! Buffer unchanged
[D][HTTPClient.cpp:685] getString(): not enough memory to reserve a string! need: 2709

```

### Sketch:
```cpp
#include <Arduino.h>
#include <BLEDevice.h>
#include "BluetoothSerial.h"
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>
#include <BLEAdvertising.h>

BluetoothSerial SerialBT;

bool initBtSerial();
void initBLEserver();

void setup() {
	// Initialize BT serial interface
	initBtSerial();

	// Initialize BLE server
	initBLEserver();
}

void loop() {
}

bool initBtSerial() {
	if (!SerialBT.begin("ESP32 Weather")) {
		Serial.println(" Failed to start BTSerial");
		return false;
	}
	return true;
}

// List of Service and Characteristic UUIDs
#define SERVICE_UUID        0x181A
//
//... more UUIDs
//

/** Characteristic for client notification */
BLECharacteristic *pCharacteristicNotify;
//
// ... more characteristics
//

/** BLE Advertiser */
BLEAdvertising* pAdvertising;
/** BLE Service */
BLEService *pService;
/** BLE Server */
BLEServer *pServer;

/** Flag if a client is connected */
bool bleConnected = false;

/** Digital output value received from the client */
uint8_t digitalOut = 0;
/** Flag for change in digital output value */
bool digOutChanged = false;

/**
 * MyServerCallbacks
 * Callbacks for client connection and disconnection
 */
class MyServerCallbacks: public BLEServerCallbacks {
	// TODO this doesn't take into account several clients being connected
		void onConnect(BLEServer* pServer) {
			bleConnected = true;
			pAdvertising->start();
		};

		void onDisconnect(BLEServer* pServer) {
			if (pServer->getConnectedCount() == 0) {
				bleConnected = false;
			}
		}
};

/**
 * MyCallbackHandler
 * Callbacks for client write requests
 */
class MyCallbackHandler: public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pCharacteristic) {
//
// ... Handle the incoming value
//
    }
  }
};

void initBLEserver() {
	// Initialize BLE
	BLEDevice::init(apName);
	// BLEDevice::setPower(ESP_PWR_LVL_P7);

	// Create BLE Server
	pServer = BLEDevice::createServer();

	// Set server callbacks
	pServer->setCallbacks(new MyServerCallbacks());

	// Create BLE Service
	pService = pServer->createService(BLEUUID((uint16_t)SERVICE_UUID),20);
//
// ... initializing the services and characteristics
//
	// Start the service
	pService->start();

	// Start advertising
	pAdvertising = pServer->getAdvertising();

	pAdvertising->start();

	Serial.println(" BLE active now");
}
```

### Debug Messages: (Init BTserial first, then BLE)
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Build: Feb 26 2018 13:17:37
Build: Feb 26 2018 13:17:37
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 7 - STA_GOT_IP
Connected to MHC2 with IP address 192.168.0.111
[E][BLEDevice.cpp:325] init(): esp_bt_controller_init: rc=259 Invalid state
[D][BLEDevice.cpp:69] createServer(): >> createServer
[D][BLEetoothSerial.cpp:47] esp_spp_cb(): ESP_SPP_INIT_EVT[I][BluetoothSerial.cpp:47] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:66] esp_spp_cb(): ESP_SPP_START_EVT
```
No more debug output after that!

### Debug Messages: (Init BLE first, then BTserial )
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Build: Feb 26 2018 13:44:17
Build: Feb 26 2018 13:44:17
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 7 - STA_GOT_IP
Connected to MHC2 with IP address 192.168.0.111
[D][BLEDevice.cpp:69] createServer(): >> createServer
[D][BLEServer.cpp:304] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
...
Lot of BLE debug output here, but no errors until
...
[E][BluetoothSerial.cpp:108] _init_bt(): _init_bt initialize controller failed
...
Lot of BLE debug output again, but no errors until
...
[D][HTTPClient.cpp:181] beginInternal(): host: api.wunderground.com port: 80 url: /api/e062ec62675eb278/conditions/q/PH/Paranaque_City.json
[D][HTTPClient.cpp:852] connect():  connected to api.wunderground.com:80
[D][HTTPClient.cpp:970] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:973] handleHeaderResponse(): size: 2708
[E][WString.cpp:185] changeBuffer(): realloc failed! Buffer unchanged
[D][HTTPClient.cpp:685] getString(): not enough memory to reserve a string! need: 2709
[D][HTTPClient.cpp:234] end(): still data in buffer (2708), clean up.
[D][HTTPClient.cpp:242] end(): tcp stop
```

copercini, 2018-02-26:
Theoretically, it should work

The problem is that `esp_bt_controller_enable` [can't be called twice using different configs](https://github.com/espressif/esp-idf/blob/165e3edd4949586dc90fa682f907d971fee488fc/components/bt/include/esp_bt.h#L161) and both libraries call it when they start

For both work together, they should call `esp_bt_controller_enable(ESP_BT_MODE_BTDM)` once, and not the BLE lib call `esp_bt_controller_enable(ESP_BT_MODE_BLE)` and BluetoohSerial call `esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT)` [here](https://github.com/copercini/arduino-esp32/blob/90174189003e5f02ca06d0e05de15a94807e0257/libraries/BluetoothSerial/src/BluetoothSerial.cpp#L101)

beegee-tokyo, 2018-02-26:
Thanks for the guidance into the right path. For BluetoothSerial I know where to override this, will check tomorrow in BLE lib if I can override it there as well. Will feedback here. If it works, both libs might need an update.

copercini, 2018-02-26:
I have an idea that can solve it, will try tonight after my work =)

beegee-tokyo, 2018-02-26:
According to [here]( https://github.com/espressif/arduino-esp32/blob/600f4c4130f2411e8ff33396d0ef40a7ec257cd6/cores/esp32/esp32-hal-bt.c#L38) btStart uses ESP_BT_MODE_BTDM already. So it might be easier to change something [here](https://github.com/nkolban/ESP32_BLE_Arduino/blob/3f5005c383540d07dd16ac044c78d7be28d4aa8e/src/BLEDevice.cpp#L329)

copercini, 2018-02-26:
I made some changes, hope it will work now =)
You can get the last version here: https://github.com/copercini/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.cpp

beegee-tokyo, 2018-02-27:
Improvement, but still not possible to use BLE and Bluetooth together.
I patched in your changes then I call in `setup()`
```
	// Initialize BLE server
	initBLEserver();

	// Initialize BT serial interface
	initBtSerial();

```
There are no errors in the debug log, however the BLE is not advertising, but the Serial Bluetooth is available and working.

So I tried to only initialize BLE only and then initialize Serial Bluetooth later with a command. Then the BLE Server is advertising and can be connected to. Everything is fine on BLE until I initialize Serial Bluetooth, then the BLE server kills the connection and stops advertising. But the Serial Bluetooth is advertising and working.

Seems there is something hidden down in ESP-IDF or @nkolban BLE implementation that doesn't allow both BLE and Bluetooth to advertise there services together, even if the Bluetooth is initializes with **`errRc = esp_bt_controller_enable(ESP_BT_MODE_BTDM);`**.

Giving up for now.

beegee-tokyo, 2018-02-27:
Ok, I didn't give up. Got it to work. The trick seems to be to somehow initialize BLE and Bluetooth Serial together.
First I tried to init BLE, start the server add callbacks and characteristics then start the advertising of the BLE service. After that I tried to init Bluetooth Serial. This didn't work, as said in my last comment.

So I tried to mix the initialization of both together in the sequence
```cpp
// Initialize BLE
	BLEDevice::init(apName);
	BLEDevice::setPower(ESP_PWR_LVL_P7);

	if (!SerialBT.begin(apName)) {
		sendDebug(debugLabel, errorLabel + digitalTimeDisplaySec() + " Failed to start BTSerial", false);
	}

	// Create BLE Server
	pServer = BLEDevice::createServer();

	// Set server callbacks
	pServer->setCallbacks(new MyServerCallbacks());

	// Create BLE Service
	pService = pServer->createService(BLEUUID((uint16_t)SERVICE_UUID),20);

	// Create BLE Characteristic for Alert
	pCharacteristicNotify = pService->createCharacteristic(
		BLEUUID((uint16_t)NOTIFICATION_UUID),
		BLECharacteristic::PROPERTY_READ	 |
		// BLECharacteristic::PROPERTY_INDICATE |
		BLECharacteristic::PROPERTY_NOTIFY
	);
//
// add other characteristics
//

	// Start the service
	pService->start();

	// Start advertising
	pAdvertising = pServer->getAdvertising();

	pAdvertising->start();
```
And now it works :-)

The working sequence seems to be 
```
Init BLE device => BLEDevice::init(apName);
Init Bluetooth Serial => SerialBT.begin(apName);
Then do the rest of the BLE initialisation like
- create server
- add callbacks to server
- add characteristics
- start server
- start advertising
```

beegee-tokyo, 2018-02-27:
Put my findings into the [Wiki](http://desire.giesecke.tk/index.php/2018/02/27/initialize-bluetooth-serial-and-ble-server/) and my [blog](http://desire.giesecke.tk/index.php/2018/02/27/esp32-bluetooth-serial-and-ble-server-running-together/)

Hope your library will be merged soon.

closed 2018-02-27 by beegee-tokyo

Akshay2804, 2020-04-17:
@beegee-tokyo were you able to merge it finally? 
I followed the entire discussion but still facing the same problem as you did, like either only BLE is working or only BluetoothSerial.

beegee-tokyo, 2020-04-17:
I gave up and I don't think the required changes got ever merged.

Akshay2804, 2020-04-19:
@beegee-tokyo okay! I am trying to reach to espressif to gain some information on the source file of ESP32, will post it here if some positive results are obtained.

BrushlessPower, 2020-05-27:
Hello,

i try to get BLE and Serial BT running.

but i cant notify with BLE any more.

[E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error

if i erase the SerialBT.begin() everything is working

matthew798, 2020-12-29:
This is also an issue for me, and the proposed order of initialization doesn't seem to work. Does anyone have more insight?

aguaviva, 2021-04-09:
I am having the same issue, starting BLE and BT cause a crash

this is my log:

```
** setup()
   * bleGamepad.begin()    <- this is from the ESP32-BLE-Gamepad lib.
   * InitBT()                          
```
where InitBT() is 
```
    esp_bluedroid_status_t bt_state = esp_bluedroid_get_status();
    if(bt_state == ESP_BLUEDROID_STATUS_UNINITIALIZED){
        if (esp_bluedroid_init()) {
            log_e("esp_bluedroid_init failed");
            return false;
        }
    }

    if(bt_state != ESP_BLUEDROID_STATUS_ENABLED){
        if (esp_bluedroid_enable()) {                                                     <-------------------- here is where the crash happens
            log_e("esp_bluedroid_enable failed");
            return false;
        }
    }
```

And this is the call stack

```
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:719 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008f8d6 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ecac:0x3ffd31c0 0x4008ef29:0x3ffd31e0 0x4008f8d6:0x3ffd3200 0x400f0762:0x3ffd3240 0x400f077a:0x3ffd3260 0x40168342:0x3ffd3280 0x4008ff9e:0x3ffd32b0
```

Any pointers on how to debug this? Are there any samples that show how to get BT and BLE cohexisting?

Cheers

anukul1392001, 2021-09-24:
@beegee-tokyo dude did the ble and classic bluetooth worked together finally ?

fpavlic, 2022-02-07:
@beegee-tokyo seems your solution is extended, however I can see in your repo that it is all commented. Does it still work?

https://github.com/beegee-tokyo/ESP32-Weatherstation/blob/master/src/ble_bt.cpp

beegee-tokyo, 2022-02-08:
Cannot say, I didn't touch ESP32 or the BLE/BT combination since years.