espressif/arduino-esp32 #1151: How can I program the ESP-WROOM-32 with LabVIEW ??

Abdulaziz95, 2018-02-26:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
How can I program the ESP-WROOM-32 with LabVIEW ??

lbernstone, 2018-02-26:
You need to explain what you are hoping to achieve. ESP32 has a standard serial and UART interface, so anything designed to collect data from another Arduino device and visualize it in Labview should be fairly interoperable (not all arduino libraries will work with ESP32).  If you want LabView to control an ESP32, then you will need to figure out a whole lot of integrations which are way beyond the scope of a one sentence question.

XYZVector, 2018-03-02:
You cannot do this.. Sorry but this is a C/C++ environment you are in. Look in google to see if this can be done.

closed 2018-04-19 by everslick